espressif/esp-iot-solution #232: Button cb needs 2 parameter to compile, doc says 1

sukesh-ak, 2022-11-10:
## Environment

- Development Kit:  UM-ESP32-S3
- Module or chip used:  ESP32-S3 with 8MB Flash / 8MB PSRAM
- IDF version IDF Version   : v4.4.2
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
Added `iot_button` using `idf.py add-dependency espressif/button^2.2.0`
As per documentation [here](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/input_device/button.html) iot_button cb has single parameter but fails to compile.

This fails
```c
static void print_button_event(button_handle_t btn)
```

This works
```c
static void print_button_event(button_handle_t btn, void * data)
```

Also Doc says
```c
iot_button_register_cb(gpio_btn, BUTTON_SINGLE_CLICK, button_single_click_cb);
```

But for it to work we need extra parameter
```c
iot_button_register_cb(gpio_btn, BUTTON_SINGLE_CLICK, button_single_click_cb,NULL);
```

### Expected Behavior

Should work with just one parameter. I guess the IDF Component Registry is not updated with latest version or has the wrong version [here](https://components.espressif.com/components/espressif/button/versions/2.2.0)

### Actual Behavior
Compile failure

### Steps to repropduce

```shell
idf.py add-dependency espressif/button^2.2.0
```

Add button and cb
```c
static void print_button_event(button_handle_t btn, void * data)
{
    button_event_t evt = iot_button_get_event(btn);
    switch (evt) {
    case BUTTON_PRESS_DOWN:
        ESP_LOGI(TAG, "BUTTON_PRESS_DOWN");
        break;
    case BUTTON_PRESS_UP:
        ESP_LOGI(TAG, "BUTTON_PRESS_UP");
        break;
    case BUTTON_PRESS_REPEAT:
        ESP_LOGI(TAG, "BUTTON_PRESS_REPEAT");
        break;
    case BUTTON_SINGLE_CLICK:
        ESP_LOGI(TAG, "BUTTON_SINGLE_CLICK");
        break;
    case BUTTON_DOUBLE_CLICK:
        ESP_LOGI(TAG, "BUTTON_DOUBLE_CLICK");
        break;
    case BUTTON_LONG_PRESS_START:
        ESP_LOGI(TAG, "BUTTON_LONG_PRESS_START");
        break;
    case BUTTON_LONG_PRESS_HOLD:
        ESP_LOGI(TAG, "BUTTON_LONG_PRESS_HOLD");
        break;

    default:
        break;
    }
}

void app_main(void)
{
    button_config_t btn_cfg = {
        .type = BUTTON_TYPE_GPIO,
        .gpio_button_config = {
            .gpio_num = BUTTON_IO_NUM,
            .active_level = BUTTON_ACTIVE_LEVEL,
        },
    };
    g_btn = iot_button_create(&btn_cfg);
    iot_button_register_cb(g_btn, BUTTON_PRESS_DOWN, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_PRESS_UP, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_PRESS_REPEAT, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_SINGLE_CLICK, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_DOUBLE_CLICK, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_LONG_PRESS_START, print_button_event,NULL);
    iot_button_register_cb(g_btn, BUTTON_LONG_PRESS_HOLD, print_button_event,NULL);
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(1000));
    }

    iot_button_delete(g_btn);
}

```

lijunru-hub, 2022-11-18:
Thanks for you report. The code in the package manager is behind our documentation and we will push the latest code up.

sukesh-ak, 2022-11-19:
> Thanks for you report. The code in the package manager is behind our documentation and we will push the latest code up.

Thank you for the update. You guys have valuable samples here. 
Hope to showcase it to the world sometime soon.

closed 2022-11-24 by zhanzhaocheng