espressif/esp-iot-solution #234: Migrate ADC to NG driver

espzav, 2022-11-24:
Now, it is working with old ADC driver in IDF 5.0 and graeter, but when user want to use ADC, he cannot use NG driver with combination this component which is using old ADC driver.

franz-ms-muc, 2022-12-06:
totally agree: 
esp-idf-v5.0/components/driver/deprecated/driver/adc.h

adc.h is deprecated now. 

must move to the new driver Thanks !

franz-ms-muc, 2022-12-06:
also we get Warning: 

```
/home/franz/esp-idf-v5.0/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
```

AshUK, 2022-12-06:
+1 would like to see this

franz-ms-muc, 2022-12-06:
+10000 from me

leeebo, 2022-12-06:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

leeebo, 2022-12-12:
There will be a new version to fit IDF 5.0 adc driver

lijunru-hub, 2022-12-21:
The new adc-driven oneshot mode is already supported in the new component.
https://github.com/espressif/esp-iot-solution/tree/master/components/button

Closes #234

menghang, 2022-12-22:
It seems that the new driver have one build error and some warnings. I think line 185 and line 212 should be fixed at least. @lijunru-hub  would you please have a check on it? Thanks a lot.

```
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c: In function 'button_adc_init':
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:185:27: warning: assignment to 'adc_oneshot_unit_handle_t' {aka 'struct adc_oneshot_unit_ctx_t *'} from incompatible pointer type 'struct adc_oneshot_unit_ctx_t 
** const*' [-Wincompatible-pointer-types]
  185 |             s_adc1_handle = &config->adc_handle ;
      |                           ^
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:212:9: error: 'ret' undeclared (first use in this 
function)
  212 |         ret = adc_oneshot_config_channel(s_adc1_handle, config->adc_channel, &oneshot_config);
      |         ^~~
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:212:9: note: each undeclared identifier is reported only once for each function it appears in
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c: In function 'button_adc_deinit':
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:267:15: warning: unused variable 'ret' [-Wunused-variable]
  267 |     esp_err_t ret = adc_oneshot_del_unit(s_adc1_handle);
      |               ^~~
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c: In function 'get_adc_volatge':
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:280:50: warning: passing argument 3 of 'adc_oneshot_read' from incompatible pointer type [-Wincompatible-pointer-types]
  280 |         adc_oneshot_read(s_adc1_handle, channel, &adc_raw);
      |                                                  ^~~~~~~~
      |                                                  |
      |                                                  uint32_t * {aka long unsigned int *}
In file included from D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:21:
D:/ESP/frameworks/esp-idf/components/esp_adc/include/esp_adc/adc_oneshot.h:85:87: note: expected 'int *' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
   85 | esp_err_t adc_oneshot_read(adc_oneshot_unit_handle_t handle, adc_channel_t chan, int *out_raw);        
      |                                                                                  ~~~~~^~~~~~~
D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:286:69: warning: passing argument 3 of 'adc_cali_raw_to_voltage' from incompatible pointer type [-Wincompatible-pointer-types]
  286 |     adc_cali_raw_to_voltage(g_button.adc1_cali_handle, adc_reading, &voltage);
      |                                                                     ^~~~~~~~
      |                                                                     |
      |                                                                     uint32_t * {aka long unsigned int *}
In file included from D:/SRC/C/ESP32/esp32c3/esprog/components/button/button_adc.c:22:
D:/ESP/frameworks/esp-idf/components/esp_adc/include/esp_adc/adc_cali.h:54:75: note: expected 'int *' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
   54 | esp_err_t adc_cali_raw_to_voltage(adc_cali_handle_t handle, int raw, int *voltage);
      |                                                                      ~~~~~^~~~~~~
```

closed 2022-12-23 by zhanzhaocheng