espressif/esp-iot-solution #239: usb_stream doesn't work for devices which send EOF in header-only frames (e.g. MS2109 based boards)

squix78, 2022-12-18:
## Environment

- Development Kit:      Custom board
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v5.1-dev-2356-gedd815af2e
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1) 11.2.0
    // 1.22.0-80-g6c4433a
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description
The usb_camera_mic_spk example doesn't work with USB UVC sources which send header-only frames. I'm using a VGA-Capture device based on the MS2109 chip and this chip seems to send frames that only contain the header but no payload. The headers of these frames contain the EOF bit set to true to signal the end of a frame. 

The code explicitly checks that the length of the body has to be not empty:
https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_stream/usb_stream.c#L1412

### Expected Behavior
The UVC driver detects frames also when the EOF bit is sent in an header-only frame.

### Actual Behavior
The UVC driver ignores header-only frames and thus can't detect the EOF bit. The condition on line 1385 https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_stream/usb_stream.c#L1385 
detects left-over frames and drops the buffer. As such the driver doesn't detect any frames.

### Steps to repropduce

1. Use a UVC devices which sends header-only frames to signal the end of a frame
2. No output generated, if logging in debug mode many "SWAP NO EOF" messages will printed

### Code to reproduce this issue

Use this example together with an MS2109 based device: examples/usb/host/usb_camera_mic_spk

## Suggested solution
Move the following block out of the enclosing condition to the end of the method. 
```
    if (header_info & (1 << 1)) {
        ///ESP_LOGI(TAG, "New Frame");
        /* The EOF bit is set, so publish the complete frame */
        _uvc_swap_buffers(strmh);
    }
```
This makes the driver also check for header-only frames and fixes the problem. I can create a PR if it has a chance to be accepted

leeebo, 2022-12-21:
Hi @squix78 thanks for you bug report, you are right this is a bug which will cause `no EOF` problem. Please feel free to add your MR !

closed 2022-12-23 by zhanzhaocheng

squix78, 2022-12-24:
Thank you @zhanzhaocheng for fixing!

leeebo, 2022-12-24:
Thanks @squix78, and we suggest you try with new version [0.4.0](https://components.espressif.com/components/espressif/usb_stream), which is a more complete uvc+uac driver, users DO NOT need to specify interface manually.