espressif/esp-iot-solution #240: [USB Dongle] AP mode does not assign the RNDIS interface IP

wuyuanyi135, 2022-12-19:
## Environment

- Development Kit:     None
- Kit version (for WroverKit/PicoKit/DevKitC): None
- Module or chip used:  ESP32S3
- IDF version (run ``git describe --tags`` to find it):  v4.4.3
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
I flashed the USB Dongle example with CDC-ACM and RNDIS enabled. The sta mode was tested fine (I can get the network access from the connected AP). However, when I run in AP mode, the RNDIS adapter was not assigned with IP (169.254.135.55). The device connects to my dongle's AP without issue but it cannot talk to my PC because it is not assigned with IP.

I tried to assign the IP manually (according to my phone WiFi information, which is connected to my dongle AP) to 192.168.4.1 and the mask 255.255.255.0. The gateway was left empty. However, I still could not make any communication between the PC to the connected device via the dongle.

I wonder, if the example does not cover this use case? (Dongle as AP) Because I don't even see the `NET_CONNECT` macro being called in AP mode. 
My intention is to use this dongle as the communication interface to the other ESP32 devices. Is this possible with the USB-networking option?

### Expected Behavior
The PC's RNDIS interface should be assigned with IP and it should be able to communicate with the devices connected to the dongle AP.

### Actual Behavior
The PC does not have an IP when in AP mode, hence not possible to communicate with the devices.

### Steps to repropduce

1. Flash the firmware
2. send command to the console like `ap test1234 test1234`
3. connect phone to the ssid (`test1234`)
4. ping the gateway from phone or ping the phone from PC
5. No response is received.

### Code to reproduce this issue

Untouched code in `examples/usb/device/usb_dongle`

## Debug Logs
No debug logs unfortunately because my board does not have UART exposed. If this is important I could wire it out and get the logs.

## Other items if possible
- [*] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/10261537/sdkconfig.txt)
- [*] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[usb_dongle.elf.txt](https://github.com/espressif/esp-iot-solution/files/10261538/usb_dongle.elf.txt)

bluestar714, 2023-04-04:
Hi, 
Why did you close this issue?
The code does not seem to assign an IP address when in AP mode, but I am interested in this usage.
Did you work around it with a different implementation?

wuyuanyi135, 2023-04-04:
@bluestar714 Because this is not the proper usage of USB RNDIS. Check https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/User_Guide.md

bluestar714, 2023-04-04:
@wuyuanyi135 

Thanks for the reply.
Well, I am aware that this is an unsupported usage.
I have determined that you have not implemented any additional implementation.
I'm also using the iot-bridge code, but this cdc-ecm seems to work as expected. However, I think I need RNDIS support for this.

wuyuanyi135, 2023-04-04:
Yes, iot-bridge has RNDIS support for this

bluestar714, 2023-04-04:
I thought iot-bridge itself does not support RNDIS yet.
Are you sure iot-bridge is already RNDIS-supported (implemented)?
If so, great!

wuyuanyi135, 2023-04-04:
It does support RNDIS but when I use the code the RNDIS netif L2 implementation was not public. I wrote my own. I heard they have decided to open that part. You can check their examples and see if it compiles.

bluestar714, 2023-04-04:
Thank you so much for your considerable tips.

I was able to verify the issue you are raising.
I also found that I need to make changes in and around bridge_usb.c. It looks like it has not been merged into master yet.
I will try to implement the relevant parts.

By the way, I am sorry to be so blunt, but if you don't mind, could you please provide me with the implementation code?
If you really don't mind, that's fine.
I really appreciate your replies so far.

wuyuanyi135, 2023-04-04:
@bluestar714 https://github.com/espressif/esp-iot-bridge/commit/1612e31208f6543fcb0241777ba1be74e4aeafdb#diff-a589039a3144caa3e14584f23da67816ed8b044e3c360e076d4a1e08d1cfad05 seems merged all proprietary sources. Do you have trouble compiling the examples?

wuyuanyi135, 2023-04-04:
My code was heavily coupled in my project so it is not easily shared. The official implementation is much organized than mine, I suggest you to stick with their implmentation :)

bluestar714, 2023-04-04:
ご対応ありがとうございます。

私はexampleのwifi_routerを使用しています。
sdkconfigにはTinyUSBのRNDISの項目が存在しません。
なので一旦ここにRNDIS対応の項目を追加したりUSB bridgeを編集することで対応しようとしています。

iot-bridgeにissueを上げてみます。

bluestar714, 2023-04-04:
Thank you for your response.

I am using example wifi_router.
There is no RNDIS entry for TinyUSB in sdkconfig.
So I am trying to support it by adding an item for RNDIS support here once and editing the USB bridge.

I will raise an issue to iot-bridge.