espressif/esp-iot-solution #241: Force usb stream if format not found

beckmx, 2023-01-05:
## Environment

- Development Kit:      [ESP32 S3]
- Kit version (ESP32-S3-DevKitC-1 v1.1)
- IDF version (run ``git describe --tags`` to find it): v5.0
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1) 11.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

Some cameras descriptor via UVC do not necessary export the full spec, I tested esp-iot-solution (master branch) and refused to extract the camera output because it didn't contain the full spec with the format indicated

//Detailed problem description goes here.

### Expected Behavior
I would expect a flag to force the usb_stream to try reading the uvc if the user knows the camera supports the format
### Actual Behavior
The code stops and doesn't continue loading the camera stream if it doesn't contain the description of the format.
### Steps to repropduce

1. Run the project ~/esp-iot-solution/examples/usb/host/usb_camera_mic_spk git:(master) 
2. Connect a camera with not all specs exported
3. The code stops with: 
W (57179) UVC_STREAM: MJPEG format NOT found
W (57179) UVC_STREAM: MJPEG width*heigh: 65535*65535, NOT found
E (57179) UVC_STREAM: /Users/andressantos/esp-iot-solution/components/usb/usb_stream/usb_stream.c:2828 (_usb_processing_task):Parse config descriptor, invalid device
I (57180) UVC_STREAM: Resetting USB Driver
I (57180) UVC_STREAM: Resetting default pipe// It helps if you attach a picture of your setup/wiring here.

## Debug Logs

```
I (305) wifi:Total power save buffer number: 16
I (305) wifi:Init max length of beacon: 752/752
I (306) wifi:Init max length of beacon: 752/752
I (307) camera_httpd: Starting web server on port: '80'
I (309) camera_httpd: Starting stream server on port: '81'
I (311) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.0
I (343) UVC_STREAM: Port=1 init succeed
I (343) UVC_STREAM: Waiting USB Connection
I (362) UVC_STREAM: USB Streaming Starting
I (362) uvc_mic_spk_demo: usb streaming start succeed
I (57060) UVC_STREAM: line 499 HCD_PORT_EVENT_CONNECTION
I (57110) UVC_STREAM: Resetting Port
I (57170) UVC_STREAM: Setting Port FIFO
I (57170) UVC_STREAM: Getting Port Speed
I (57170) UVC_STREAM: Port speed = 1
I (57171) UVC_STREAM: USB Speed: full-speed
I (57171) UVC_STREAM: Default pipe endpoint MPS update to 64
I (57172) UVC_STREAM: Set Device Addr = 1
I (57172) UVC_STREAM: Set Device Addr Done
I (57172) UVC_STREAM: get device desc
I (57172) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x54c
idProduct 0x1337
bNumConfigurations 1
I (57173) UVC_STREAM: get short config desc
I (57174) UVC_STREAM: get config desc, actual_num_bytes:16
I (57174) UVC_STREAM: get full config desc 289
I (57175) UVC_STREAM: get full config desc, actual_num_bytes:297
*** Configuration descriptor ***
wTotalLength 289
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
	*** Interface descriptor ***
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 0
	bInterfaceClass 0xe (Video)
	bInterfaceSubClass 0x1
	*** Interface descriptor ***
	bInterfaceNumber 1
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0xe (Video)
	bInterfaceSubClass 0x2
	*** Class-specific VS Interface Descriptor ***
	bNumFormats 1
		*** Endpoint descriptor ***
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
I (57178) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (57178) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
W (57179) UVC_STREAM: MJPEG format NOT found
W (57179) UVC_STREAM: MJPEG width*heigh: 65535*65535, NOT found
E (57179) UVC_STREAM: /Users/andressantos/esp-iot-solution/components/usb/usb_stream/usb_stream.c:2828 (_usb_processing_task):Parse config descriptor, invalid device
I (57180) UVC_STREAM: Resetting USB Driver
I (57180) UVC_STREAM: Resetting default pipe
W (57381) UVC_STREAM: Pipe wait event timeout
I (57382) UVC_STREAM: hcd recovering
I (57432) UVC_STREAM: hcd port state after recovering = 6
I (57632) UVC_STREAM: Waiting USB Connection
W (255078) UVC_STREAM: line 506 HCD_PORT_EVENT_DISCONNECTION
I (255079) UVC_STREAM: Resetting USB Driver
I (255079) UVC_STREAM: hcd recovering
I (255129) UVC_STREAM: hcd port state after recovering = 0
I (255329) UVC_STREAM: Waiting USB Connection
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/10355394/sdkconfig.txt)

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

leeebo, 2023-01-05:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

leeebo, 2023-01-06:
@beckmx Yes, forcing `format_index` to the user's configuration can be supported in next version. You can also test with the history version [v0.3.3](https://components.espressif.com/components/espressif/usb_stream/versions/0.3.3) , users must force every params there.

beckmx, 2023-01-06:
I just checked the source code of the camera, looks like it is sending DMA values through the wire as nv12 format, if I were to feed the esp32 with this format where could I start?

leeebo, 2023-01-06:
Hi @beckmx nv12 is a type of uncompressed format, from UVC side, the only different is the `Payload definitions, MPJEG and NV12 may have different payload headers, please refer [USB Video Payload Uncompressed v1.5](https://www.usb.org/document-library/video-class-v15-document-set) for details

closed 2023-01-11 by beckmx