espressif/esp-iot-solution #244: Data overflow on streaming

beckmx, 2023-01-15:
## Environment
Development Kit: [ESP32 S3]
Kit version (ESP32-S3-DevKitC-1 v1.1)
IDF version (run git describe --tags to find it): v5.0
Build System: [CMake]
Compiler version (run xtensa-esp32-elf-gcc --version to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1) 11.2.0
Operating System: [macOS]
Power Supply: [USB]
## Problem Description

//Detailed problem description goes here.
I am loading a screensize that is somehow irregular, I am wondering if is because I need more PSRAM or this is kinda big for the processor to handle.

### Expected Behavior
Memory allocation should be enough

### Actual Behavior
Overflow happens at usb_stream

### Steps to repropduce
Attach a camera with 960 by 544

## Debug Logs

```
I (8151) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (8151) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (8152) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (8152) UVC_STREAM: Actual MJPEG width*heigh: 960*544, frame index = 1
I (8153) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.0
I (8153) UVC_STREAM: _usb_get_parse_config_desc_2:0
I (8153) UVC_STREAM: urb free
I (8153) UVC_STREAM: Skipping camera verification?
I (8154) UVC_STREAM: urb alloced
I (8154) UVC_STREAM: Set Device Configuration = 1
I (8154) UVC_STREAM: Got expected pipe_handle and event
I (8155) UVC_STREAM: Set Device Configuration Done
I (8155) UVC_STREAM: urb free
I (8155) UVC_STREAM: Probe MJPEG payload size = 35840
I (8155) UVC_STREAM: urb alloced
I (8156) UVC_STREAM: SET_CUR Probe
I (8156) UVC_STREAM: Got expected pipe_handle and event
I (8156) UVC_STREAM: SET_CUR Probe Done
I (8157) UVC_STREAM: GET_CUR Probe
I (8157) UVC_STREAM: Got expected pipe_handle and event
**I (8157) UVC_STREAM: /Users/xxx/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1743 (_uvc_vs_commit_control):urb status: data overflow**
I (8158) UVC_STREAM: urb free
I (8158) UVC_STREAM: urb alloced
I (8158) UVC_STREAM: Set Device Interface = 1, Alt = 0
I (8159) UVC_STREAM: Got expected pipe_handle and event
I (8159) UVC_STREAM: Set Device Interface Done
I (8159) UVC_STREAM: urb free
I (8160) UVC_STREAM: usb stream task start
I (8160) UVC_STREAM: urb alloced
I (8160) UVC_STREAM: uvc stream urb ready
I (8160) UVC_STREAM: Sample processing task started
I (8161) UVC_STREAM: Creating uvc in(bulk) pipe itf = 1-0, ep = 0x81
I (8161) UVC_STREAM: pipe init succeed
I (8161) UVC_STREAM: urb context update
I (8162) UVC_STREAM: uvc streaming...
```

I checked the line of the overflow and apparently you guys know it could overflow in there and after that, sometimes the wifi doesnt server the html to the client. So I was wondering if there is a variable I could increase or is the physical memory that is not enough.

Greetings!

lijunru-hub, 2023-01-16:
Please try to use the latest usb_stream component, version 0.4.1

beckmx, 2023-01-16:
Hello @lijunru-hub I tried with now 0.4.1, same result, about the variable sizes, can you point me to some that can be increased?

```
I (25167) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (25168) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (25168) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (25168) UVC_STREAM: Actual MJPEG width*heigh: 960*544, frame index = 1
I (25169) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.1
I (25169) UVC_STREAM: _usb_get_parse_config_desc_2:0
I (25170) UVC_STREAM: urb free
I (25170) UVC_STREAM: Skipping camera verification?
I (25170) UVC_STREAM: urb alloced
I (25170) UVC_STREAM: Set Device Configuration = 1
I (25171) UVC_STREAM: Got expected pipe_handle and event
I (25171) UVC_STREAM: Set Device Configuration Done
I (25171) UVC_STREAM: urb free
I (25171) UVC_STREAM: Probe MJPEG payload size = 35840
I (25172) UVC_STREAM: urb alloced
I (25172) UVC_STREAM: SET_CUR Probe
I (25172) UVC_STREAM: Got expected pipe_handle and event
I (25173) UVC_STREAM: SET_CUR Probe Done
I (25173) UVC_STREAM: GET_CUR Probe
I (25173) UVC_STREAM: Got expected pipe_handle and event
I (25174) UVC_STREAM: /Users/xxxx/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1745 (_uvc_vs_commit_control):urb status: data overflow
I (25174) UVC_STREAM: urb free
I (25175) UVC_STREAM: urb alloced
I (25175) UVC_STREAM: Set Device Interface = 1, Alt = 0
I (25175) UVC_STREAM: Got expected pipe_handle and event
I (25175) UVC_STREAM: Set Device Interface Done
I (25176) UVC_STREAM: urb free
I (25176) UVC_STREAM: usb stream task start
I (25176) UVC_STREAM: urb alloced
I (25176) UVC_STREAM: uvc stream urb ready
I (25177) UVC_STREAM: Sample processing task started
I (25177) UVC_STREAM: Creating uvc in(bulk) pipe itf = 1-0, ep = 0x81
I (25178) UVC_STREAM: pipe init succeed
I (25178) UVC_STREAM: urb context update
I (25178) UVC_STREAM: uvc streaming...
```

lijunru-hub, 2023-01-17:
The reason for this error is that urb got the description length and the actual length is not the same, this has nothing to do with the image size, you have to make sure that your camera supports full speed mode, can you send the camera USB descriptor?

beckmx, 2023-01-17:
sure, this is the output from linux command v4l2, I am writing the codec for to support nv12, but the usb descriptor is this

```
v4l2-ctl --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : carduvc
	Bus info         : usb-3f980000.usb-1.5
	Driver version   : 5.10.63
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : uvcdevice
	Serial           : UDCD UVC
	Bus info         : usb-3f980000.usb-1.5
	Media version    : 5.10.63
	Hardware revision: 0x00000100 (256)
	Driver version   : 5.10.63
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : uvcdevice
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200000a: from remote pad 0x1000009 of entity 'Input 1': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
	Width/Height      : 960/544
	Pixel Format      : 'NV12' (Y/CbCr 4:2:0)
	Field             : None
	Bytes per Line    : 960
	Size Image        : 783360
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             :
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 960, Height 544
	Default     : Left 0, Top 0, Width 960, Height 544
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 960, Height 544, Flags:
Selection: crop_bounds, Left 0, Top 0, Width 960, Height 544, Flags:
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 60.000 (60/1)
	Read buffers     : 0
```

And the full descriptor I am outputting from the esp32S3 is this one, where all the frame formats are shown:

```
I (25132) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
[... 170 lines cut ...]
		bInterval 0
I (25167) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (25168) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (25168) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (25168) UVC_STREAM: Actual MJPEG width*heigh: 960*544, frame index = 1
I (25169) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.1
I (25169) UVC_STREAM: _usb_get_parse_config_desc_2:0
I (25170) UVC_STREAM: urb free
I (25170) UVC_STREAM: Skipping camera verification?
I (25170) UVC_STREAM: urb alloced
I (25170) UVC_STREAM: Set Device Configuration = 1
I (25171) UVC_STREAM: Got expected pipe_handle and event
I (25171) UVC_STREAM: Set Device Configuration Done
I (25171) UVC_STREAM: urb free
I (25171) UVC_STREAM: Probe MJPEG payload size = 35840
I (25172) UVC_STREAM: urb alloced
I (25172) UVC_STREAM: SET_CUR Probe
I (25172) UVC_STREAM: Got expected pipe_handle and event
I (25173) UVC_STREAM: SET_CUR Probe Done
I (25173) UVC_STREAM: GET_CUR Probe
I (25173) UVC_STREAM: Got expected pipe_handle and event
I (25174) UVC_STREAM: /Users/xyz/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1745 (_uvc_vs_commit_control):urb status: data overflow
I (25174) UVC_STREAM: urb free
I (25175) UVC_STREAM: urb alloced
```

lijunru-hub, 2023-01-18:
Please specify the image size as 480*272 and try to get a smaller image to see if it works properly.
[image]

beckmx, 2023-01-18:
Actually originally I changed the values to a "medium" size which is one of the frame configurations of the camera, I will try your suggestion later on
```
#define DEMO_UVC_FRAME_WIDTH        960
#define DEMO_UVC_FRAME_HEIGHT       544
```

beckmx, 2023-01-19:
Here is the output, that overflow seems to be there

```
I (13007) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (13008) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (13008) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (13008) UVC_STREAM: Actual MJPEG width*heigh: 480*272, frame index = 4
I (13009) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.1
I (13009) UVC_STREAM: _usb_get_parse_config_desc_2:0
I (13010) UVC_STREAM: urb free
I (13010) UVC_STREAM: Skipping camera verification?
I (13010) UVC_STREAM: urb alloced
I (13010) UVC_STREAM: Set Device Configuration = 1
I (13011) UVC_STREAM: Got expected pipe_handle and event
I (13011) UVC_STREAM: Set Device Configuration Done
I (13011) UVC_STREAM: urb free
I (13011) UVC_STREAM: Probe MJPEG payload size = 35840
I (13012) UVC_STREAM: urb alloced
I (13012) UVC_STREAM: SET_CUR Probe
I (13012) UVC_STREAM: Got expected pipe_handle and event
I (13013) UVC_STREAM: SET_CUR Probe Done
I (13013) UVC_STREAM: GET_CUR Probe
I (13013) UVC_STREAM: Got expected pipe_handle and event
I (13013) UVC_STREAM: /Users/xyz/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1745 (_uvc_vs_commit_control):urb status: data overflow
I (13014) UVC_STREAM: urb free
I (13015) UVC_STREAM: urb alloced
I (13015) UVC_STREAM: Set Device Interface = 1, Alt = 0
I (13015) UVC_STREAM: Got expected pipe_handle and event
I (13015) UVC_STREAM: Set Device Interface Done
I (13016) UVC_STREAM: urb free
I (13016) UVC_STREAM: usb stream task start
I (13016) UVC_STREAM: urb alloced
I (13016) UVC_STREAM: uvc stream urb ready
I (13017) UVC_STREAM: Sample processing task started
I (13017) UVC_STREAM: Creating uvc in(bulk) pipe itf = 1-0, ep = 0x81
I (13018) UVC_STREAM: pipe init succeed
I (13018) UVC_STREAM: urb context update
I (13018) UVC_STREAM: uvc streaming...
```

leeebo, 2023-01-30:
Hi, @beckmx thanks for your report, the issue happens because the Host driver requests a length = 26 Bytes package, but the device returns more. which will cause the [assert failed](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_stream/usb_stream.c#L1704).

Please printf the `urb_done->transfer.actual_num_bytes` value to check the actual package size, and then give a try with bigger transfer size.

You can try to modify `s_usb_dev.ep_mps` to `sizeof(usb_setup_packet_t) + usb_round_up_to_mps(CTRL_TRANSFER_DATA_MAX_BYTES, s_usb_dev.ep_mps)` in this line like below:

```
//urb_ctrl->transfer.num_bytes = s_usb_dev.ep_mps;
urb_ctrl->transfer.num_bytes = sizeof(usb_setup_packet_t) + usb_round_up_to_mps(CTRL_TRANSFER_DATA_MAX_BYTES, s_usb_dev.ep_mps);
```

beckmx, 2023-01-30:
Hello @leeebo thanx for the tip, I changed the suggestion and also printed the actual_num_bytes, I am getting 16 bytes first and then in other call 26
```
I (312) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.1
I (343) UVC_STREAM: Port=1 init succeed
I (343) UVC_STREAM: Waiting USB Connection
I (363) UVC_STREAM: USB Streaming Starting
I (363) uvc_mic_spk_demo: usb streaming start succeed
I (6565) UVC_STREAM: line 501 HCD_PORT_EVENT_CONNECTION
I (6615) UVC_STREAM: Resetting Port
I (6675) UVC_STREAM: Setting Port FIFO
I (6675) UVC_STREAM: Getting Port Speed
I (6675) UVC_STREAM: Port speed = 1
I (6676) UVC_STREAM: USB Speed: full-speed
I (6676) UVC_STREAM: pipe init succeed
I (6676) UVC_STREAM: urb alloced
I (6677) UVC_STREAM: Got expected pipe_handle and event
I (6677) UVC_STREAM: REPORTED BYTES==> 16
I (6677) UVC_STREAM: Default pipe endpoint MPS update to 64
I (6677) UVC_STREAM: urb free
I (6677) UVC_STREAM: urb alloced
I (6678) UVC_STREAM: Set Device Addr = 1
I (6678) UVC_STREAM: Got expected pipe_handle and event
I (6678) UVC_STREAM: Set Device Addr Done
I (6678) UVC_STREAM: urb free
I (6679) UVC_STREAM: urb alloced
I (6679) UVC_STREAM: get device desc
I (6679) UVC_STREAM: Got expected pipe_handle and event
I (6679) UVC_STREAM: REPORTED BYTES==> 26
I (6680) UVC_STREAM: get device desc, actual_num_bytes:26
```

I still see the issue of the overflow:

```
I (6719) UVC_STREAM: Probe MJPEG payload size = 35840
I (6719) UVC_STREAM: urb alloced
I (6719) UVC_STREAM: SET_CUR Probe
I (6720) UVC_STREAM: Got expected pipe_handle and event
I (6720) UVC_STREAM: SET_CUR Probe Done
I (6720) UVC_STREAM: GET_CUR Probe
I (6720) UVC_STREAM: Got expected pipe_handle and event
I (6721) UVC_STREAM: /Users/xxx/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1753 (_uvc_vs_commit_control):urb status: data overflow
I (6721) UVC_STREAM: urb free
I (6722) UVC_STREAM: urb alloced
I (6722) UVC_STREAM: Set Device Interface = 1, Alt = 0
I (6722) UVC_STREAM: Got expected pipe_handle and event
I (6723) UVC_STREAM: Set Device Interface Done
```

So, I guess I need to update a couple more lines with the num_bytes to match as your example, right?

leeebo, 2023-01-31:
Hi @beckmx please make sure change the right lines, you only need to modify the `urb_ctrl->transfer.num_bytes` in GET_CUR Probe stage

[image]

[image]

beckmx, 2023-01-31:
@leeebo sorry, as I have added some code for the codec I am building I lost track of the method, so yes, I am receiving a bigger reference in num_bytes, 34 in my case

```
I (47067) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (47068) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (47068) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (47068) UVC_STREAM: Actual MJPEG width*heigh: 480*272, frame index = 4
I (47069) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.1
I (47069) UVC_STREAM: _usb_get_parse_config_desc_2:0
I (47070) UVC_STREAM: urb free
I (47070) UVC_STREAM: Skipping camera verification?
I (47070) UVC_STREAM: urb alloced
I (47070) UVC_STREAM: Set Device Configuration = 1
I (47071) UVC_STREAM: Got expected pipe_handle and event
I (47071) UVC_STREAM: Set Device Configuration Done
I (47071) UVC_STREAM: urb free
I (47072) UVC_STREAM: Probe MJPEG payload size = 35840
I (47072) UVC_STREAM: urb alloced
I (47072) UVC_STREAM: SET_CUR Probe
I (47072) UVC_STREAM: Got expected pipe_handle and event
I (47073) UVC_STREAM: REPORTED BYTES 34
I (47073) UVC_STREAM: SET_CUR Probe Done
I (47073) UVC_STREAM: REPORTED BYTES 34
I (47073) UVC_STREAM: GET_CUR Probe
I (47074) UVC_STREAM: Got expected pipe_handle and event
I (47074) UVC_STREAM: /Users/xyz/esp-iot-solution/components/usb/usb_stream/usb_stream.c:1747 (_uvc_vs_commit_control):urb status: data overflow
I (47075) UVC_STREAM: urb free
```

leeebo, 2023-01-31:
@beckmx Just comment out the length check, keep going and see if it works

beckmx, 2023-01-31:
@leeebo I am starting to see _something_ new in the console :)

```
I (11102) UVC_STREAM: urb context update
I (11102) UVC_STREAM: uvc streaming...
I (11133) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 1, width = 480, height = 272, length = 35828, ptr = 0
I (11163) UVC_STREAM: bogus packet: actual_len=35840, header_len=55

I (11194) UVC_STREAM: bogus packet: actual_len=35840, header_len=83

I (11224) UVC_STREAM: bogus packet: actual_len=35840, header_len=231

I (11254) UVC_STREAM: bogus packet: actual_len=35840, header_len=188

I (11268) UVC_STREAM: bogus packet: actual_len=16652, header_len=181

I (11329) UVC_STREAM: bogus packet: actual_len=35840, header_len=55

I (11360) UVC_STREAM: bogus packet: actual_len=35840, header_len=83

I (11390) UVC_STREAM: bogus packet: actual_len=35840, header_len=231
```
I will continue on this and keep you posted

leeebo, 2023-02-01:
@beckmx `bogus packet: actual_len=35840, header_len=55` means the payload does not have a valid UVC MJPEG header. The first Byte should be `12` but we got `55`

[image]

We will release v0.4.2, from which  you can enable the VERBOSE log to print each payload for better debugging

closed 2023-02-01 by zhanzhaocheng

beckmx, 2023-02-03:
@leeebo I got too far on the project :p you can see it here, https://github.com/beckmx/esp32-ps-vita-streamer, I forked this repo and modified, but at the end I think the esp32 crashes for lack of memory?, but at least I am happy to see I almost got it working

leeebo, 2023-02-06:
` but at the end I think the esp32 crashes for lack of memory` 
Hi @beckmx you can just upload your full log here, i will help find the problem.

beckmx, 2023-02-06:
Sure, this is the part where it breaks:

```
I (4030) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 0
I (4030) UVC_STREAM: 	Endpoint(BULK) Addr = 0x81, MPS = 64
I (4031) UVC_STREAM: Actual MJPEG format index = 1, contains 5 frames
I (4031) UVC_STREAM: Actual MJPEG width*heigh: 960*544, frame index = 1
I (4032) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.2
I (4032) UVC_STREAM: Set Device Configuration = 1
I (4032) UVC_STREAM: Set Device Configuration Done
I (4033) UVC_STREAM: Probe MJPEG payload size = 2048
I (4033) UVC_STREAM: SET_CUR Probe
I (4033) UVC_STREAM: SET_CUR Probe Done
I (4033) UVC_STREAM: GET_CUR Probe
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 666666
dwMaxPayloadTransferSize: 1382412
bInterfaceNumber: 0
I (4034) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:42
I (4035) UVC_STREAM: SET_CUR COMMIT
I (4035) UVC_STREAM: SET_CUR COMMIT Done
I (4035) UVC_STREAM: Set Device Interface = 1, Alt = 0
I (4036) UVC_STREAM: Set Device Interface Done
I (4036) UVC_STREAM: usb stream task start
I (4036) UVC_STREAM: uvc stream urb ready
I (4036) UVC_STREAM: Sample processing task started
I (4037) UVC_STREAM: Creating uvc in(bulk) pipe itf = 1-0, ep = 0x81
I (4037) UVC_STREAM: uvc streaming...
I (4053) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 1, width = 960, height = 544, length = 2036, ptr = 0
I (4053) uvc_mic_spk_demo: size of DATA = 4 ptr = 2036
I (4053) uvc_mic_spk_demo: starting compression--->0
I (4055) uvc_mic_spk_demo: looping--->
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40383292  PS      : 0x00060833  A0      : 0x80382d2e  A1      : 0x3fcf28e0
0x40383292: remove_free_block at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:338
 (inlined by) block_locate_free at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:573
 (inlined by) tlsf_malloc at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:934

A2      : 0x3fca097c  A3      : 0x0000000c  A4      : 0x3fcc3de8  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x0000cdcd  A8      : 0x00000003  A9      : 0x00ffe3fb
A10     : 0x8f9bfe17  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x40379b54
0x40379b54: wifi_malloc at /Users/xx/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:64

A14     : 0x3fcef940  A15     : 0x0000cdcd  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
EXCVADDR: 0x8f9bfe23  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x4038328f:0x3fcf28e0 0x40382d2b:0x3fcf2900 0x40376245:0x3fcf2920 0x403762a3:0x3fcf2940 0x403762da:0x3fcf2960 0x40385045:0x3fcf2980 0x40379b59:0x3fcf29a0 0x4203f713:0x3fcf29c0 0x4202dffd:0x3fcf29e0 0x42007a62:0x3fcf2a00 0x42007ab5:0x3fcf2a30 0x40381321:0x3fcf2a50
0x4038328f: remove_free_block at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:337
 (inlined by) block_locate_free at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:573
 (inlined by) tlsf_malloc at /Users/xx/esp-idf/components/heap/tlsf/tlsf.c:934

0x40382d2b: multi_heap_malloc_impl at /Users/xx/esp-idf/components/heap/multi_heap.c:217

0x40376245: heap_caps_malloc_base at /Users/xx/esp-idf/components/heap/heap_caps.c:146

0x403762a3: heap_caps_malloc at /Users/xx/esp-idf/components/heap/heap_caps.c:166

0x403762da: heap_caps_malloc_default at /Users/xx/esp-idf/components/heap/heap_caps.c:191

0x40385045: malloc at /Users/xx/esp-idf/components/newlib/heap.c:24

0x40379b59: wifi_malloc at /Users/xx/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:68

0x4203f713: ieee80211_timer_process at ??:?

0x4202dffd: ftm_is_supported at ??:?

0x42007a62: timer_process_alarm at /Users/xx/esp-idf/components/esp_timer/src/esp_timer.c:361

0x42007ab5: timer_task at /Users/xx/esp-idf/components/esp_timer/src/esp_timer.c:387 (discriminator 1)

0x40381321: vPortTaskWrapper at /Users/xx/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 0f242bf93850dbf5

Rebooting...
```

leeebo, 2023-02-08:
Hi @beckmx, It's more like a memory stomp problem, the program trying to store data to an invalid address `EXCVADDR: 0x8f9bfe23`. 

BTW, you can first comment out the Wi-Fi related code, and check if UVC itself works.

For the memory, you can use [heap_caps_get_free_size](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv423heap_caps_get_free_size8uint32_t) to check the remaining memory space