espressif/esp-iot-solution #245: Knob component passed incorrect data in callbacks

cydergoth, 2023-01-17:
- [ X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ X] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      LILLYGO T-Embed
- Module or chip used: ESP32S3
- IDF version (run ``git describe --tags`` to find it): V5.0
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

In the "knob" component https://github.com/espressif/esp-iot-solution/tree/master/components/knob

`#define CALL_EVENT_CB(ev)   if(knob->cb[ev])knob->cb[ev](knob, knob->usr_data) `

This line is incorrect and results in passing the _base of the usr_data array_ instead of the correct usr_data value from registering the callback

It should be: 

`#define CALL_EVENT_CB(ev)   if(knob->cb[ev])knob->cb[ev](knob, knob->usr_data**[ev]**) `

### Expected Behavior

usr_data passed into this function VVV should be passed to the callback

```
esp_err_t iot_knob_register_cb(knob_handle_t knob_handle, knob_event_t event, knob_cb_t cb, **void *usr_data**)
{
    KNOB_CHECK(NULL != knob_handle, "Pointer of handle is invalid", ESP_ERR_INVALID_ARG);
    KNOB_CHECK(event < KNOB_EVENT_MAX, "event is invalid", ESP_ERR_INVALID_ARG);
    knob_dev_t *knob = (knob_dev_t *) knob_handle;
    knob->cb[event] = cb;
    **knob->usr_data[event] = usr_data; // <-------- Here the userdata is stored in the usr_data array**
    return ESP_OK;
}
```

### Actual Behavior

The parameter passed for usr_data to the callback is the base of the usr_data array in the knob structure

### Steps to repropduce

1. init a knob componet
2. register a callback with a usr_data pointer 
3. Trigger the callback
4. Note that the callback receives incorrect usr_data

### Code to reproduce this issue

See above

cydergoth, 2023-01-17:
Workaround: 

```
static void main_menu_knob_right_cb(void *arg, void *data)
{
    assert(data);
#ifdef BAD_KNOB_USR_DATA
    void** usr_data = (void**)data;
    main_scr_t main = (main_scr_t)usr_data[KNOB_RIGHT];
    assert(main->magic==0xF0F3);
#else
```

lijunru-hub, 2023-01-18:
Thank you for your contribution, this issue will be fixed in version 0.1.1. You can check the latest component code at this [link](https://components.espressif.com/components?q=knob) 

And we welcome your suggestions for features of this component.

closed 2023-02-07 by zhanzhaocheng