espressif/esp-iot-solution #246: TinyUSB USB BLE fail to compute shared secret

ImmineoII, 2023-01-24:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Kit version: 1.1
- Module or chip used: ESP32-S3-WROOM-2
- IDF version: 4.4.3
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System: Windows 10
- Power Supply: USB

## Problem Description

During computation of shared secret using ECDH between two devices that start a bonding procedure linux kernel hang with error.

### Expected Behavior
Use the devkit as a BLE USB dongle for a linux board

### Actual Behavior
Device is recognized as a BLE USB dongle, is possible to manage it and connect to other BLE device but it fail to bond.

### Steps to repropduce

1. Connect DevKit to linux PC and powerup
2. On linux PC run:
     1. bluetootctl 
     2. agent on
     3. power on
     4. advertise on
     5. discoverable on 
     6. pairable on
7. With another BLE device (I used my Motorola Edge 30 smartphone) try to connect to the PC, for this i used Nordic android app.
8. Connection willi be established succesfully
9. Try to bond devices, doesn't matter who start the procedure
10. ECDH shared secret computation fail
11. Disconnection triggered

## Debug Logs

ESP32 doesn't log anythings bad, simply ignore the problem and fail silently.

Linux kernel version: 4.14.78
BlueZ version: 5.50

I also tried to repeat the same test attaching the devboard to a Raspberri 4 withi Raspbian and the result is the same.

In both cases linux kernel log this:
[ 7270.029224] alg: ecdh: compute shared secret failed. err -22

Attached you can find sdkconfig, btmon and esp32 log.

[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/10488335/sdkconfig.txt)
[esp32.log](https://github.com/espressif/esp-iot-solution/files/10488347/esp32.log)
[btmon.log](https://github.com/espressif/esp-iot-solution/files/10488350/btmon.log)

xiewenxiang, 2023-01-30:
Does these two devices have been paired and bonding at some point before? And then one of the devices deleted the bonding information.

ImmineoII, 2023-01-30:
@xiewenxiang they never completed succesfully a bonding.

xiewenxiang, 2023-02-03:
[0001-bugfix-ble-ble-hci-layer-adapter-issues.patch](https://github.com/espressif/esp-iot-solution/files/10577113/0001-bugfix-ble-ble-hci-layer-adapter-issues.patch)
Hi @ImmineoII , please apply this patch and test again.

ImmineoII, 2023-02-06:
Hi @xiewenxiang, the patch solved the problem, i can now connect and perform bonding.
Are this patch goint to be included into a new commit of add_usb_solutions branch or i have to apply it locally?

ImmineoII, 2023-02-06:
i just see that patch is already merged, thanks for help :)

closed 2023-02-06 by ImmineoII