espressif/esp-iot-solution #248: 使用esp-iot-solution master分支的usb_stream摄像头颜色不对

ch-duan, 2023-02-08:
## Environment

- Development Kit:  ESP32-S3-DevKitC-1
- Kit version: 1.1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v5.0
- Build System:CMAKE
- Compiler version :11.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
master分支的usb_stream 连接usb摄像头时颜色偏红。这个摄像头在windows和ubuntu上的视频和图片都是正常的。我可以通过确认哪些信息去查找UVC的设置问题？
## Debug Logs

```
I (00:00:01.143) stream_s: Starting stream server on port: '80'
I (00:00:01.143) NET SENDER: create thread
I (00:00:01.155) NET SENDER: Socket created
I (00:00:01.159) NET SENDER: Socket bound, port 12345
I (00:00:01.166) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.2
I (00:00:01.203) UVC_STREAM: Port=1 init succeed
I (00:00:01.203) UVC_STREAM: Waiting USB Connection
I (00:00:01.223) UVC_STREAM: USB Streaming Starting
I (00:00:01.223) uvc_demo: Begin capture frame
I (00:00:01.454) UVC_STREAM: line 501 HCD_PORT_EVENT_CONNECTION
I (00:00:01.504) UVC_STREAM: Resetting Port
I (00:00:01.564) UVC_STREAM: Setting Port FIFO
I (00:00:01.564) UVC_STREAM: Getting Port Speed
I (00:00:01.564) UVC_STREAM: Port speed = 1
I (00:00:01.568) UVC_STREAM: USB Speed: full-speed
I (00:00:01.575) UVC_STREAM: Default pipe endpoint MPS update to 64
I (00:00:01.581) UVC_STREAM: Set Device Addr = 1
I (00:00:01.586) UVC_STREAM: Set Device Addr Done
I (00:00:01.591) UVC_STREAM: get device desc
I (00:00:01.597) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0xc45
idProduct 0x64ab
bcdDevice 6.10
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
I (00:00:01.626) UVC_STREAM: get short config desc
I (00:00:01.633) UVC_STREAM: get config desc, actual_num_bytes:16
I (00:00:01.639) UVC_STREAM: get full config desc 249
I (00:00:01.646) UVC_STREAM: get full config desc, actual_num_bytes:257
*** Configuration descriptor ***
bLength 9
bDescriptorType 0x2
wTotalLength 249
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
*** Interface Association Descriptor: Video ***
bLength 8
bDescriptorType 0xb
bFirstInterface 0
bInterfaceCount 2
bFunctionClass 0xe
bFunctionSubClass 0x3
bFunctionProtocol 0x0
iFunction 4
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 0x4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
        bInterfaceProtocol 0x0
        iInterface 4
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 0x5
                bEndpointAddress 0x87   EP 7 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 0x4
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        bInterfaceProtocol 0x0
        iInterface 0
        *** Class-specific VS Interface Descriptor ***
        bLength 0xe
        bDescriptorType 0x24
        bDescriptorSubType 1
        bNumFormats 1
        wTotalLength 61
        bEndpointAddress 129
        bFunctionProtocol 0
        bmInfo 0x5
        bTerminalLink 1
        bStillCaptureMethod 1
        bTriggerSupport 0
        bTriggerUsage 1
        bControlSize 4
        bmaControls 0xb
        *** VS Format MJPEG Descriptor ***
        bLength 0xb
        bDescriptorType 0x24
        bDescriptorSubType 0x6
        bFormatIndex 0x1
        bNumFrameDescriptors 1
        bmFlags 0x1
        bDefaultFrameIndex 1
        bAspectRatioX 0
        bAspectRatioY 0
        bmInterlaceFlags 0x0
        bCopyProtect 0
        *** VS MJPEG Frame Descriptor ***
        bLength 0x1e
        bDescriptorType 0x24
        bDescriptorSubType 0x7
        bFrameIndex 0x1
        bmCapabilities 0x0
        wWidth 640
        wHeigh 480
        dwMinBitRate 122880000
        dwMaxBitRate 122880000
        dwMaxVideoFrameBufSize 614400
        dwDefaultFrameInterval 400000
        bFrameIntervalType 1
        FrameInterval[0] 400000
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 0x4
        bInterfaceNumber 1
        bAlternateSetting 1
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 0x5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 512
                bInterval 1
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 0x4
        bInterfaceNumber 1
        bAlternateSetting 2
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 0x5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 944
                bInterval 1
I (00:00:01.881) UVC_STREAM: Actual VS Interface(MPS < 600) found, interface = 1, alt = 1
I (00:00:01.890) UVC_STREAM:    Endpoint(ISOC) Addr = 0x81, MPS = 512
I (00:00:01.897) UVC_STREAM: Actual MJPEG format index = 1, contains 1 frames
I (00:00:01.905) UVC_STREAM: Actual MJPEG width*heigh: 640*480, frame index = 1
I (00:00:01.913) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.2
I (00:00:01.920) UVC_STREAM: Set Device Configuration = 1
I (00:00:01.927) UVC_STREAM: Set Device Configuration Done
I (00:00:01.933) UVC_STREAM: uvc_dev vs ifc not found
I (00:00:01.938) UVC_STREAM: Probe MJPEG payload size = 512
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 666666
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 49152
dwMaxPayloadTransferSize: 512
dwClockFrequency: 0
bmFramingInfo: 0
bPreferredVersion: 0
bMinVersion: 0
bMaxVersion: 0
bInterfaceNumber: 0
I (00:00:01.973) UVC_STREAM: SET_CUR Probe
I (00:00:01.981) UVC_STREAM: SET_CUR Probe Done
I (00:00:01.983) UVC_STREAM: GET_CUR Probe
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 666666
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 944
dwClockFrequency: 0
bmFramingInfo: 0
bPreferredVersion: 0
bMinVersion: 0
bMaxVersion: 0
bInterfaceNumber: 0
I (00:00:02.016) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:34
I (00:00:02.023) UVC_STREAM: SET_CUR COMMIT
I (00:00:02.030) UVC_STREAM: SET_CUR COMMIT Done
I (00:00:02.033) UVC_STREAM: Set Device Interface = 1, Alt = 1
I (00:00:02.040) UVC_STREAM: Set Device Interface Done
I (00:00:02.046) UVC_STREAM: usb stream task start
I (00:00:02.051) UVC_STREAM: uvc stream urb ready
I (00:00:02.057) UVC_STREAM: Sample processing task started
I (00:00:02.063) UVC_STREAM: Creating uvc in(isoc) pipe itf = 1-1, ep = 0x81
I (00:00:02.071) UVC_STREAM: uvc streaming...
```

## Other items if possible

- [sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/10682230/sdkconfig.txt) sdkconfig file (attach the sdkconfig file from your project folder)
- [image]
这是错误的图片

leeebo, 2023-02-08:
@ch-duan Hii，usb_stream 组件不进行 video control 的操作，也不会对获取到的图像数据进行处理。应用层接收到的为摄像头在默认配置下的数据。

“_Windows 和 Linux 正常，ESP32S3 平台上颜色偏红_。”
这里的关键区别是，在 ESP32S3 上摄像头工作在 Fullspeed 模式，而在以上系统中摄像头工作在 Highspeed 模式。摄像头固件在 Fullspeed 和 Highspeed 模式下描述符和配置都可能不同。

所以这里可能是该摄像头在 Fullspeed 模式下，输出图像颜色异常。你可以通过一个 USB 隔离器，将摄像头速率限制在 Fullspeed 模式下接入电脑，验证以上猜想

ch-duan, 2023-02-08:
> 

我这个摄像头是dvp接口通过解码芯片转成USB输出的。如果是这个速度导致的图像问题，我要去找摄像头厂商将USB速度限制在Fullspeed对吧

ch-duan, 2023-02-08:
@leeebo 我这个摄像头是dvp接口通过解码芯片转成USB输出的。如果是这个速度导致的图像问题，我要去找摄像头厂商将USB速度限制在Fullspeed对吧

leeebo, 2023-02-08:
@ch-duan 和供应商确认摄像头 Fullspeed 模式下，输出正常的图像

closed 2023-03-07 by ch-duan