espressif/esp-iot-solution #250: usb_stream: ID 0c45:6366 Microdia Webcam support

Draghtnod, 2023-02-09:
I am trying to get this Webcam working. It seems to be 'smart' and outputs a different Configuration Descriptor on the PC than on the ESP. Unfortunately, only 2 uncompressed formats are supported in this "comparibility" mode. Does maybe someone have a hint, how this "compatibility" mode is triggered?

Check [usb_stream.c ](https://github.com/espressif/esp-iot-solution/blob/6cba9ceb60b56f2d33b3a049abf1ecfc6412f364/components/usb/usb_stream/usb_stream.c#L2885) here. The Descriptors are read right after reset and some basic usb port configuration.

Device/Configuration Descriptors on ESP:
<details>
<summary>Click to expand</summary>

```
I (1353) UVC_STREAM: UVC Streaming Config Succeed, Version: 0.4.3
I (1883) UVC_STREAM: Port=1 init succeed
I (1883) UVC_STREAM: Waiting USB Connection
I (1883) UVC_STREAM: USB Streaming Starting
I (1883) uvc_demo: usb streaming start succeed
I (2133) UVC_STREAM: line 506 HCD_PORT_EVENT_CONNECTION
I (2143) UVC_STREAM: Resetting Port
I (2203) UVC_STREAM: Setting Port FIFO
I (2203) UVC_STREAM: Getting Port Speed
I (2203) UVC_STREAM: Port speed = 1
I (2203) UVC_STREAM: USB Speed: full-speed
I (2213) UVC_STREAM: Default pipe endpoint MPS update to 64
I (2213) UVC_STREAM: Set Device Addr = 1
I (2223) UVC_STREAM: Set Device Addr Done
I (2223) UVC_STREAM: get device desc
I (2233) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0xc45
idProduct 0x6366
bcdDevice 0.02
iManufacturer 2
iProduct 1
iSerialNumber 3
bNumConfigurations 1
I (2263) UVC_STREAM: get short config desc
I (2263) UVC_STREAM: get config desc, actual_num_bytes:16
I (2273) UVC_STREAM: get full config desc 463
I (2303) UVC_STREAM: get full config desc, actual_num_bytes:471
*** Configuration descriptor ***
bLength 9
bDescriptorType 0x2
wTotalLength 463
bNumInterfaces 4
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 128mA
*** Interface Association Descriptor: Video ***
bLength 8
bDescriptorType 0xb
bFirstInterface 0
bInterfaceCount 2
bFunctionClass 0xe
bFunctionSubClass 0x3
bFunctionProtocol 0x0
iFunction 5
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 0x4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
[... 929 lines cut ...]
        tSamFreq[ 4]        24000
        tSamFreq[ 5]        44100
        tSamFreq[ 6]        48000
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0190  1x 400 bytes
        bInterval               4
        bRefresh                0
        bSynchAddress           0
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000
```
</details>

leeebo, 2023-02-10:
@Draghtnod That's because the camera vendors usually provide different configurations for USB **Fullspeed** and **Highspeed**, ESP32S3 supports only Fullspeed mode, so the descriptor for this mode is printed.

The descriptors got on PC are Highspeed configs. To get Fullspeed descriptors from PC, you can add a USB Fullspeed isolator before USB interface, force it not work in Highspeed mode.

Draghtnod, 2023-02-10:
@leeebo Thank you, this explains a lot. I guess my only option then is to buy a different camera. :(

closed 2023-02-10 by Draghtnod