espressif/esp-iot-solution #254: uvc camera get device descriptor failed

Gornerrrrr, 2023-03-31:
I used the ESP32-S2-Saola-1 board to connect a UVC camera and obtain images on a web page. 
Previously, I tested multiple UVC cameras purchased online and found one that was compatible and has been in use. However, after purchasing the same model of camera from the same vendor, it is now unable to be used. The vendor has replied that the module is the same as before, and upon inspection, the chips on the module are indeed the same. What could be the possible reasons for this issue?

log (old module):
`[   935][I][uvc_stream.c:1865] uvc_streaming_config(): UVC Streaming Config Succeed
[   942][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[   947][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[   952][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[   958][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[  1163][I][uvc_stream.c:1890] uvc_streaming_start(): UVC Streaming Starting
[  1163][I][cam.cpp:138] init(): uvc streaming successed!
[  1193][I][uvc_stream.c:481] _usb_port_init(): Port=1 init succeed
[  1193][I][uvc_stream.c:1545] _usb_processing_task(): Waitting USB Connection
[  1444][I][uvc_stream.c:382] _usb_port_event_dflt_process(): line 382 HCD_PORT_EVENT_CONNECTION
[  1444][I][uvc_stream.c:1562] _usb_processing_task(): Resetting Port
[  1507][I][uvc_stream.c:1566] _usb_processing_task(): Setting Port FIFO
[  1507][I][uvc_stream.c:1570] _usb_processing_task(): Getting Port Speed
[  1509][I][uvc_stream.c:535] _usb_port_get_speed(): Port speed = 1
[  1515][I][uvc_stream.c:1574] _usb_processing_task(): USB Speed: full-speed
[  1522][I][uvc_stream.c:1585] _usb_processing_task(): Camera Start Streaming
[  1528][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[  1534][I][uvc_stream.c:932] _usb_set_device_addr(): Set Device Addr = 1
[  1540][D][uvc_stream.c:826] _default_pipe_event_wait_until(): Got expected pipe_handle and event
[  1549][I][uvc_stream.c:944] _usb_set_device_addr(): Set Device Addr Done
[  1555][D][uvc_stream.c:591] _usb_urb_free(): urb free
[  1560][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[  1566][I][uvc_stream.c:844] _usb_get_dev_desc(): get device desc
[  1577][D][uvc_stream.c:826] _default_pipe_event_wait_until(): Got expected pipe_handle and event
[  1580][I][uvc_stream.c:855] _usb_get_dev_desc(): get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x38f
idProduct 0x6001
bcdDevice 5.30
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
[  1611][D][uvc_stream.c:591] _usb_urb_free(): urb free
[  1616][D][uvc_stream.c:573] _usb_urb_alloc(): urb alloced
[  1622][I][uvc_stream.c:879] _usb_get_config_desc(): get short config desc
[  1636][D][uvc_stream.c:826] _default_pipe_event_wait_until(): Got expected pipe_handle and event
[  1637][I][uvc_stream.c:890] _usb_get_config_desc(): get config desc, actual_num_bytes:16
[  1645][I][uvc_stream.c:897] _usb_get_config_desc(): get full config desc
[  1673][D][uvc_stream.c:826] _default_pipe_event_wait_until(): Got expected pipe_handle and event
[  1673][I][uvc_stream.c:908] _usb_get_config_desc(): get full config desc, actual_num_bytes:459
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 451
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x0
	iInterface 4
	bcdUVC 100
[... 638 lines cut ...]
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       4
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0 
      iInterface              4 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
`

leeebo, 2023-03-31:
@Gornerrrrr Hi, seems you are using the legacy version uvc_stream from usb/add_usb_solutions branch. We suggest you using the new component [usb_stream ](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/usb_stream)from the master branch, try with the example https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk

Gornerrrrr, 2023-04-07:
I've tried an updated version idf 4.4.4 and new iot solution, but it's still the same problem.

> [   866][I][uvc_stream.c:1747] uvc_streaming_config(): [UVC_STREAM] Isoc transfer mode
[   868][I][uvc_stream.c:1754] uvc_streaming_config(): [UVC_STREAM] UVC Streaming Config Succeed
[   927][I][uvc_stream.c:1780] uvc_streaming_start(): [UVC_STREAM] UVC Streaming Starting
[   927][I][cam.cpp:144] init(): [UVC_STREAM] uvc streaming successed!
[   956][I][uvc_stream.c:470] _usb_port_init(): [UVC_STREAM] Port=1 init succeed
[   956][I][uvc_stream.c:1438] _usb_processing_task(): [UVC_STREAM] Waiting USB Connection
[  1210][I][uvc_stream.c:370] _usb_port_event_dflt_process(): [UVC_STREAM] line 370 HCD_PORT_EVENT_CONNECTION
[  1260][I][uvc_stream.c:1457] _usb_processing_task(): [UVC_STREAM] Resetting Port
[  1320][I][uvc_stream.c:1460] _usb_processing_task(): [UVC_STREAM] Setting Port FIFO
[  1321][I][uvc_stream.c:1463] _usb_processing_task(): [UVC_STREAM] Getting Port Speed
[  1324][I][uvc_stream.c:524] _usb_port_get_speed(): [UVC_STREAM] Port speed = 1
[  1332][I][uvc_stream.c:1466] _usb_processing_task(): [UVC_STREAM] USB Speed: full-speed
[  1340][I][uvc_stream.c:930] _usb_set_device_addr(): [UVC_STREAM] Set Device Addr = 1
[  1347][I][uvc_stream.c:942] _usb_set_device_addr(): [UVC_STREAM] Set Device Addr Done
[  1355][I][uvc_stream.c:840] _usb_get_dev_desc(): [UVC_STREAM] get device desc
[  1363][W][uvc_stream.c:614] _default_pipe_event_dflt_process(): [UVC_STREAM] line 614 Pipe: default HCD_PIPE_EVENT_ERROR_XFER
[  1373][E][uvc_stream.c:846] _usb_get_dev_desc(): [UVC_STREAM] C:\Users\PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\uvc_stream\src\uvc_stream.c:846 (_usb_get_dev_desc):urb event error
[  1392][I][uvc_stream.c:761] _usb_pipe_flush(): [UVC_STREAM] pipe flushing: state = 1
[  1599][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  1599][E][uvc_stream.c:1481] _usb_processing_task(): [UVC_STREAM] C:\Users\PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\uvc_stream\src\uvc_stream.c:1481 (_usb_processing_task):Get device descriptor failed
[  1618][I][uvc_stream.c:1591] _usb_processing_task(): [UVC_STREAM] Resetting USB Driver
[  1625][I][uvc_stream.c:1593] _usb_processing_task(): [UVC_STREAM] Resetting stream pipe
[  1633][I][uvc_stream.c:737] _usb_pipe_deinit(): [UVC_STREAM] pipe state = 0
[  1839][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  1839][I][uvc_stream.c:1598] _usb_processing_task(): [UVC_STREAM] Resetting default pipe
[  1845][I][uvc_stream.c:737] _usb_pipe_deinit(): [UVC_STREAM] pipe state = 0
[  2051][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  2051][I][uvc_stream.c:1606] _usb_processing_task(): [UVC_STREAM] hcd recovering
[  2106][I][uvc_stream.c:1609] _usb_processing_task(): [UVC_STREAM] hcd port state = 6
[  2306][I][uvc_stream.c:1438] _usb_processing_task(): [UVC_STREAM] Waiting USB Connection

leeebo, 2023-04-07:
@Gornerrrrr Not sure what exactly caused the failure to get the device descriptor, can you add some delay before `_usbh_get_dev_desc` is called and try again?

```c
#ifdef CONFIG_USBH_GET_DEVICE_DESC
                    vTaskDelay(pdMS_TO_TICKS(100)); // add some delay time
                    ret = _usbh_get_dev_desc(pipe_instance_dflt, NULL);
                    ERR_CHECK_GOTO(ESP_OK == ret, "Get device descriptor failed", usb_driver_reset_);
#endif
```

Gornerrrrr, 2023-04-07:
@leeebo It's the same problem with add 1000ms delay. And I tried to set it without getting the descriptor. Get HCD_PIPE_EVENT_ERROR_XFER too.
`[   902][I][uvc_stream.c:1748] uvc_streaming_config(): [UVC_STREAM] Isoc transfer mode
[   904][I][uvc_stream.c:1755] uvc_streaming_config(): [UVC_STREAM] UVC Streaming Config Succeed
[   963][I][uvc_stream.c:1781] uvc_streaming_start(): [UVC_STREAM] UVC Streaming Starting
[   963][I][cam.cpp:144] init(): [UVC_STREAM] uvc streaming successed!
[   992][I][uvc_stream.c:470] _usb_port_init(): [UVC_STREAM] Port=1 init succeed
[   993][I][uvc_stream.c:1438] _usb_processing_task(): [UVC_STREAM] Waiting USB Connection
[  1246][I][uvc_stream.c:370] _usb_port_event_dflt_process(): [UVC_STREAM] line 370 HCD_PORT_EVENT_CONNECTION
[  1296][I][uvc_stream.c:1457] _usb_processing_task(): [UVC_STREAM] Resetting Port
[  1356][I][uvc_stream.c:1460] _usb_processing_task(): [UVC_STREAM] Setting Port FIFO
[  1357][I][uvc_stream.c:1463] _usb_processing_task(): [UVC_STREAM] Getting Port Speed
[  1360][I][uvc_stream.c:524] _usb_port_get_speed(): [UVC_STREAM] Port speed = 1
[  1368][I][uvc_stream.c:1466] _usb_processing_task(): [UVC_STREAM] USB Speed: full-speed
[  1376][I][uvc_stream.c:930] _usb_set_device_addr(): [UVC_STREAM] Set Device Addr = 1
[  1384][I][uvc_stream.c:942] _usb_set_device_addr(): [UVC_STREAM] Set Device Addr Done
[  1391][I][uvc_stream.c:963] _usb_set_device_config(): [UVC_STREAM] Set Device Configuration = 1
[  1400][W][uvc_stream.c:614] _default_pipe_event_dflt_process(): [UVC_STREAM] line 614 Pipe: default HCD_PIPE_EVENT_ERROR_XFER
[  1411][E][uvc_stream.c:967] _usb_set_device_config(): [UVC_STREAM] C:\Users\PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\uvc_stream\src\uvc_stream.c:967 (_usb_set_device_config):urb event error
[  1430][I][uvc_stream.c:761] _usb_pipe_flush(): [UVC_STREAM] pipe flushing: state = 1
[  1637][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  1638][E][uvc_stream.c:1489] _usb_processing_task(): [UVC_STREAM] C:\Users\PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\uvc_stream\src\uvc_stream.c:1489 (_usb_processing_task):Set device configuration failed
[  1656][I][uvc_stream.c:1592] _usb_processing_task(): [UVC_STREAM] Resetting USB Driver
[  1664][I][uvc_stream.c:1594] _usb_processing_task(): [UVC_STREAM] Resetting stream pipe
[  1672][I][uvc_stream.c:737] _usb_pipe_deinit(): [UVC_STREAM] pipe state = 0
[  1878][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  1879][I][uvc_stream.c:1599] _usb_processing_task(): [UVC_STREAM] Resetting default pipe
[  1884][I][uvc_stream.c:737] _usb_pipe_deinit(): [UVC_STREAM] pipe state = 0
[  2090][W][uvc_stream.c:800] _default_pipe_event_wait_until(): [UVC_STREAM] Pipe wait event timeout
[  2091][I][uvc_stream.c:1607] _usb_processing_task(): [UVC_STREAM] hcd recovering
[  2145][I][uvc_stream.c:1610] _usb_processing_task(): [UVC_STREAM] hcd port state = 6
[  2345][I][uvc_stream.c:1438] _usb_processing_task(): [UVC_STREAM] Waiting USB Connection`

I'm sure the camera in PC can be work.

leeebo, 2023-04-07:
Hi @Gornerrrrr , please try with the basic USB host driver to find if the device can be correctly enum https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib

Gornerrrrr, 2023-04-07:
你好, 我尝试复制烧写了这个驱动示例.
可用的摄模块:
> [   715][I][uvc_test.ino:43] host_lib_daemon_task(): [DAEMON] Installing USB Host Library
> [   744][I][class_driver.cpp:144] class_driver_task(): [CLASS] Registering Client
> E (1469) HUB: Configuration descriptor larger than control transfer max length
> E (1470) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC

不可用的模块:
> [   715][I][uvc_test.ino:43] host_lib_daemon_task(): [DAEMON] Installing USB Host Library
> [   744][I][class_driver.cpp:144] class_driver_task(): [CLASS] Registering Client
> E (1436) HUB: Bad transfer status: CHECK_FULL_DEV_DESC
> E (1436) HUB: Stage failed: CHECK_FULL_DEV_DESC

补充下我用的是arduino平台编译, 不知道如何才能设置ENUM_CTRL_TRANSFER_MAX_DATA_LEN参数?

leeebo, 2023-04-07:
不可用的模块，在这个代码下，枚举也是失败的。

@Gornerrrrr 请在[这一行](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/usb_host_lib/main/usb_host_lib_main.c#L39)之前加 vTaskDelay(pdMS_TO_TICKS(100)); 再测试一下 不可用的模块，看是否可以枚举成功

Gornerrrrr, 2023-04-07:
@leeebo 我添加了延时
```
    bool has_clients = true;
    bool has_devices = true;
    while (has_clients || has_devices ) {
        vTaskDelay(pdMS_TO_TICKS(100)); //添加延时测试是否可以成功枚举
        uint32_t event_flags;
        ESP_ERROR_CHECK(usb_host_lib_handle_events(portMAX_DELAY, &event_flags));
        if (event_flags & USB_HOST_LIB_EVENT_FLAGS_NO_CLIENTS) {
            has_clients = false;
        }
        if (event_flags & USB_HOST_LIB_EVENT_FLAGS_ALL_FREE) {
            has_devices = false;
        }
    }
```
报错还是一样的
> [   744][I][class_driver.cpp:144] class_driver_task(): [CLASS] Registering Client
> E (2140) HUB: Bad transfer status: CHECK_FULL_DEV_DESC
> E (2140) HUB: Stage failed: CHECK_FULL_DEV_DESC

可能是协议什么部分出了问题? 
会不会和硬件供电等有关系? 但是更换以前的模块(厂商说都是一样的) 就可以正常运行.

leeebo, 2023-04-07:
@Gornerrrrr 应该是摄像头 `全速模式` 兼容性问题，请联系供应商 确认新的模块 `全速模式` 是可用的。（你在 PC 测试，默认是 `高速模式`，可以加一个隔离器，让摄像头工作在全速模式去测）

Gornerrrrr, 2023-04-07:
@leeebo 这个新模块已经联系供应商确认过了，刷了固件后软硬件和旧模块一样的，但是就是通讯失败...

Gornerrrrr, 2023-04-10:
@leeebo 今天从供应商那边又拿了三个一模一样的模块测试,可以正常工作了. 但是还是会出现之前那个枚举时获取设备描述符失败的问题, 这时直接断电重启一下(软重启也可以)就可以解决了. 请问有可能是什么原因导致的呢? 
和硬件是否有关? 我在测试用主板上直接接成功率非常高, 只有一两次失败过, 但是接到设备上(USB线更长但是电压也有5V) 基本每次长时间关机后重启都会出现这个问题.

leeebo, 2023-04-10:
@Gornerrrrr 如果是做产品，可以[联系乐鑫商务](https://www.espressif.com/en/contact-us/technical-inquiries/software-bug-report)，寄送摄像头申请协助调试

Gornerrrrr, 2023-04-12:
@leeebo 好的感谢这几天你的帮助! 已经发现了决定性的通讯失败原因, 接在设备上连接线比较长就会不行, 接在测试板上线短就没问题, 大概率供电问题.

closed 2023-04-14 by Gornerrrrr