espressif/esp-iot-solution #255: esp32s3 usb_cdc_4g_module with EC200U-EU

sjleesj, 2023-03-31:
Test esp32s3+EC200U-EU get below error，I haved try change the end Endpoint with  0x81-0x01, 0x82-0x01, 0x83-0x02,  0x84-0x02, 0x85-0x03, 0x86-0x04, 0x87-0x05, 0x88-0x06, 0x89-0x07, 0x8a-0x07, 0x8b-0x08, all get error; Please help analyze the reason!

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17c4
load:0x403c9700,len:0xe78
load:0x403cc700,len:0x310c
entry 0x403c9980
I (25) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (25) boot: compile time 23:49:09
I (25) boot: chip revision: v0.1
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (104) boot_comm: chip revision: 1, min. application chip revision: 0
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=21268h (135784) map
I (140) esp_image: segment 1: paddr=00031290 vaddr=3fc99a20 size=0443ch ( 17468) load
I (144) esp_image: segment 2: paddr=000356d4 vaddr=40374000 size=0a944h ( 43332) load
I (155) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=987b4h (624564) map
I (250) esp_image: segment 4: paddr=000d87dc vaddr=4037e944 size=0b0d4h ( 45268) load
I (267) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (279) cpu_start: Pro cpu up.
I (279) cpu_start: Starting app cpu, entry point is 0x40375678
0x40375678: call_start_cpu1 at D:/idf/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (293) cpu_start: Pro cpu start user code
I (293) cpu_start: cpu freq: 240000000
I (293) cpu_start: Application information:
I (295) cpu_start: Project name:     cdc_4g
I (300) cpu_start: App version:      1
I (305) cpu_start: Compile time:     Mar 31 2023 15:17:13
I (311) cpu_start: ELF file SHA256:  d1cef5d9f5f8e444...
I (317) cpu_start: ESP-IDF:          v4.4.4-dirty
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (329) heap_init: At 3FCA1820 len 00047EF0 (287 KiB): D/IRAM
I (336) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (343) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (349) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (355) spi_flash: detected chip: gd
I (359) spi_flash: flash io: qio
W (363) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (377) sleep: Configure to isolate all GPIO pins in sleep state
I (383) sleep: Enable automatic switching of GPIO sleep configuration
I (390) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (3431) 4g_main: ====================================
I (3431) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3431) 4g_main: ====================================
W (3437) modem_board: Force reset modem
I (3441) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
W (3451) modem_board: Resetting modem using io=0, level=0
W (3657) modem_board: Waiting for modem initialize ready
I (8687) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (8737) USB_HCDC: CDC task start
I (8737) ESP_USBH: usb port start succeed
[... 382 lines cut ...]
I (15313) modem_board: reconnect after 3s...
I (16313) modem_board: reconnect after 2s...
I (17313) modem_board: reconnect after 1s...
I (17313) modem_board: Modem state STAGE_SYNC, Start
W (17359) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (17359) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (17364) modem_board: Modem state STAGE_SYNC, Start
W (17415) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (17415) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (17420) modem_board: Modem state STAGE_SYNC, Start
W (17471) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (17471) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (17476) modem_board: Modem state STAGE_SYNC, Start
W (17527) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (17527) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (17532) modem_board: Modem state STAGE_SYNC, Start
W (17583) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (17583) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (17588) modem_board: Modem state STAGE_SYNC, Start
W (17639) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0

sjleesj, 2023-03-31:
the Endpoint is  0x7 OUT, 0x8a  IN,    and then also got error:
I (25086) ESP_USBH: Set Device Configuration = 1
I (25091) ESP_USBH: Set Device Configuration Done
I (25096) ESP_USBH: Pipe init succeed, addr: 8A
I (25102) ESP_USBH: Pipe init succeed, addr: 07
I (25107) modem_board: DTE reconnect, reconnecting ...

I (26113) modem_board: reconnect after 5s...
I (27113) modem_board: reconnect after 4s...
I (28113) modem_board: reconnect after 3s...
I (29113) modem_board: reconnect after 2s...
I (30113) modem_board: reconnect after 1s...
I (30113) modem_board: Modem state STAGE_SYNC, Start
W (30159) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (30159) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (30164) modem_board: Modem state STAGE_SYNC, Start
W (30215) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (30215) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (30220) modem_board: Modem state STAGE_SYNC, Start
W (30271) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+IFC=0,0
...failed
E (30271) modem_board: modem_board_start_up(245): set_flow_ctrl failed
I (30276) modem_board: Modem state STAGE_SYNC, Start

but when I Block this code，it works;
[image]

[image]

leeebo, 2023-04-07:
@sjleesj Thanks for your report. Set `echo behavior` is a very common AT command, but seems the module EC200U does not support it (or they use a custom format?), would you please help check the modem's specification to confirm it?

sjleesj, 2023-04-08:
The module manufacturer provided these specifications

[EC200U Mini PCle.zip](https://github.com/espressif/esp-iot-solution/files/11183804/EC200U.Mini.PCle.zip)
[Quectel_ECx00U&EGx00U系列_Linux_USB驱动_用户指导_V1.0.0_Preliminary_20210310.pdf](https://github.com/espressif/esp-iot-solution/files/11183805/Quectel_ECx00U.EGx00U._Linux_USB._._V1.0.0_Preliminary_20210310.pdf)

leeebo, 2023-04-10:
@sjleesj The module may not support flow control command, according to the module's AT manual, we should check whether it can be configured before issuing the configuration command... Will fix it, thanks for report

[image]

closed 2023-04-18 by zhanzhaocheng