espressif/esp-iot-solution #259: 调用了uac相关的代码，但是电脑上看不到uac设备

weiqifa0, 2023-04-06:
按照例子程序写了代码
运行日志有看到log
I (381) lvgl_helpers: Display hor size: 240, ver size: 240
I (382) lvgl_helpers: Display buffer size: 2400
I (386) lvgl_helpers: Initializing SPI master for display
I (393) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (399) lvgl_helpers: MISO pin: -1, MOSI pin: 11, SCLK pin: 12, IO2/WP pin: -1, IO3/HD pin: -1
I (408) lvgl_helpers: Max transfer size: 4800 (bytes)
I (414) lvgl_helpers: Initializing SPI bus...
I (419) disp_spi: Adding SPI device
I (423) disp_spi: Clock speed: 60000000Hz, mode: 2, CS pin: 10
ST7789 initialization.
Enabling backlight.
I (632) st7789: Display orientation: LANDSCAPE
I (632) st7789: 0x36 command value: 0x60
 page_fft_start    esp_get_free_heap_size : 335460  
 page_fft_start    esp_get_minimum_free_heap_size : 331012  
FFT_Setup !Dram: 335460 bytes
I (650) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=2
I (652) I2S: I2S0, MCLK output by GPIO0
I (657) UVC_STREAM: UAC Streaming Config Succeed, Version: 0.4.11
I (694) UVC_STREAM: Port=1 init succeed
I (694) UVC_STREAM: Waiting USB Connection
I (694) UVC_STREAM: USB Streaming Starting
I (697) uac: usb streaming start succeed

但是这电脑上看不到uac设备，是怎么回事？

leeebo, 2023-04-07:
@weiqifa0 因为这个 USB Host UAC+UVC，作为主机连接其它摄像头和音频。而非作为 Device 接入电脑。USB Device UAC 这里有一个 [example](https://github.com/espressif/esp-box/tree/master/examples/usb_headset)，可以满足你的需求

weiqifa0, 2023-04-07:
感谢感谢

closed 2023-08-04 by leeebo