espressif/esp-iot-solution #261: usb_camera_wifi_transfer - Get device descriptor failed

ScootchDragon, 2023-04-28:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4.4
    // v3.2-dev-1148-g96cd3b75c
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

When running the "usb_camera_wifi_transfer" example in the "add_usb_solutions" branch, trying to get the device descriptor gives me an error.

The camera I am using is "ELP-USBFHD06H-L21"

### Expected Behavior

I am expecting it to return a device descriptor.

### Actual Behavior

Returns a "urb event error" and "get device descriptor failed".

### Steps to repropduce

Here is how my camera is wired:
[image]

### Code to reproduce this issue

Run the "usb_camera_wifi_transfer" example in the "add_usb_solutions" branch.

## Debug Logs

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbcc
load:0x403cc700,len:0x2fac
SHA-256 comparison failed:
Calculated: 347f6a07e62c605f73e139bcf5609c35ff9e975b3ff61ccd00b14275ceb51e58
Expected: d132bda8c1107965847794a220d3591a47e6f90b864e557009f0ff43bc724f74
Attempting to boot anyway...
entry 0x403c9954
I (43) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (43) boot: compile time 11:11:03
I (43) boot: chip revision: v0.1
I (45) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (52) boot.esp32s3: Boot SPI Speed : 80MHz
I (57) boot.esp32s3: SPI Mode       : DIO
I (61) boot.esp32s3: SPI Flash Size : 8MB
I (66) boot: Enabling RNG early entropy source...
I (72) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=18ae0h (101088) map
I (143) esp_image: segment 1: paddr=00028b08 vaddr=3fc96650 size=04654h ( 18004) load
I (147) esp_image: segment 2: paddr=0002d164 vaddr=40374000 size=02eb4h ( 11956) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=780fch (491772) map
I (245) esp_image: segment 4: paddr=000a8124 vaddr=40376eb4 size=0f794h ( 63380) load
I (267) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (279) opi psram: vendor id : 0x0d (AP)
I (279) opi psram: dev id    : 0x02 (generation 3)
I (279) opi psram: density   : 0x03 (64 Mbit)
I (283) opi psram: good-die  : 0x01 (Pass)
I (287) opi psram: Latency   : 0x01 (Fixed)
I (292) opi psram: VCC       : 0x01 (3V)
I (297) opi psram: SRF       : 0x01 (Fast Refresh)
I (302) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (308) opi psram: BurstLen  : 0x01 (32 Byte)
I (313) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (319) opi psram: DriveStrength: 0x00 (1/1)
I (324) spiram: Found 64MBit SPI RAM device
I (328) spiram: SPI RAM mode: sram 40m
I (333) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (340) cpu_start: Pro cpu up.
I (344) cpu_start: Starting app cpu, entry point is 0x40375388
0x40375388: call_start_cpu1 at C:/Users/Kevin/vs-code-esp-idf-4-4/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (1081) spiram: SPI SRAM memory test OK
I (1090) cpu_start: Pro cpu start user code
I (1090) cpu_start: cpu freq: 240000000
I (1090) cpu_start: Application information:
I (1093) cpu_start: Project name:     uvc_mjpeg_wifi
I (1099) cpu_start: App version:      642e3c0-dirty
I (1104) cpu_start: Compile time:     Apr 28 2023 11:10:42
I (1110) cpu_start: ELF file SHA256:  f03d67917e7e9db9...
I (1116) cpu_start: ESP-IDF:          v4.4.4
I (1122) heap_init: Initializing. RAM available for dynamic allocation:
I (1129) heap_init: At 3FC9E558 len 0004B1B8 (300 KiB): D/IRAM
I (1135) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1142) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1148) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (1155) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1163) spi_flash: detected chip: generic
I (1167) spi_flash: flash io: dio
I (1172) sleep: Configure to isolate all GPIO pins in sleep state
I (1178) sleep: Enable automatic switching of GPIO sleep configuration
I (1186) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1201) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1229) pp: pp rom version: e7ae62f
I (1230) net80211: net80211 rom version: e7ae62f
I (1231) wifi:wifi driver task: 3fcf4d58, prio:23, stack:6656, core=0
I (1234) system_api: Base MAC address is not set
I (1239) system_api: read default base MAC address from EFUSE
I (1255) wifi:wifi firmware version: 6567a16
I (1256) wifi:wifi certification version: v7.0
I (1256) wifi:config NVS flash: enabled
I (1257) wifi:config nano formating: disabled
I (1261) wifi:Init data frame dynamic rx buffer num: 32
I (1266) wifi:Init management frame dynamic rx buffer num: 32
I (1272) wifi:Init management short buffer num: 32
I (1277) wifi:Init static tx buffer num: 16
I (1281) wifi:Init tx cache buffer num: 32
I (1284) wifi:Init static tx FG buffer num: 2
I (1288) wifi:Init static rx buffer size: 1600
I (1292) wifi:Init static rx buffer num: 10
I (1296) wifi:Init dynamic rx buffer num: 32
I (1301) wifi_init: rx ba win: 6
I (1304) wifi_init: tcpip mbox: 32
I (1308) wifi_init: udp mbox: 6
I (1312) wifi_init: tcp mbox: 6
I (1316) wifi_init: tcp tx win: 5744
I (1320) wifi_init: tcp rx win: 5744
I (1324) wifi_init: tcp mss: 1440
I (1328) wifi_init: WiFi IRAM OP enabled
I (1333) wifi_init: WiFi RX IRAM OP enabled
I (1341) camera wifi: wifi_init_softap finished.SSID:ESP32S2-UVC password:
I (1346) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (1386) wifi:mode : softAP (f4:12:fa:f9:79:9d)
I (1387) wifi:Total power save buffer number: 8
I (1387) wifi:Init max length of beacon: 752/752
I (1388) wifi:Init max length of beacon: 752/752
I (1393) camera_httpd: Starting web server on port: '80'
I (1399) camera_httpd: Starting stream server on port: '81'
I (1406) UVC_STREAM: Bulk transfer mode
I (1409) UVC_STREAM: UVC Streaming Config Succeed
I (1464) UVC_STREAM: UVC Streaming Starting
I (1494) UVC_STREAM: Port=1 init succeed
I (1494) UVC_STREAM: Waiting USB Connection
I (1745) UVC_STREAM: line 367 HCD_PORT_EVENT_CONNECTION
I (1795) UVC_STREAM: Resetting Port
I (1855) UVC_STREAM: Setting Port FIFO
I (1855) UVC_STREAM: Getting Port Speed
I (1855) UVC_STREAM: Port speed = 1
I (1857) UVC_STREAM: USB Speed: full-speed
I (1862) UVC_STREAM: Set Device Addr = 1
I (1867) UVC_STREAM: Set Device Addr Done
I (1871) UVC_STREAM: get device desc
W (6875) UVC_STREAM: Pipe wait event timeout
E (6875) UVC_STREAM: C:/Users/Kevin/esp-iot-solution/components/usb/uvc_stream/uvc_stream.c:843 (_usb_get_dev_desc):urb event error
I (6882) UVC_STREAM: pipe flushing: state = 0
E (6887) UVC_STREAM: C:/Users/Kevin/esp-iot-solution/components/usb/uvc_stream/uvc_stream.c:1478 (_usb_processing_task):Get device descriptor failed
I (6901) UVC_STREAM: Resetting USB Driver
I (6905) UVC_STREAM: Resetting stream pipe
I (6910) UVC_STREAM: pipe state = 0
W (7114) UVC_STREAM: Pipe wait event timeout
I (7114) UVC_STREAM: Resetting default pipe
I (7114) UVC_STREAM: pipe state = 0
W (7317) UVC_STREAM: Pipe wait event timeout
I (7317) UVC_STREAM: hcd recovering
I (7367) UVC_STREAM: hcd port state = 6
I (7567) UVC_STREAM: Waiting USB Connection
```

leeebo, 2023-05-10:
Hi @ScootchDragon the examples in `add_usb_solutions` branch is archived. Please using new [usb_stream ](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/usb_stream) component and [example ](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk) to have a try !

Gornerrrrr, 2023-06-05:
Same problem. I'm try updating new [usb_stream ](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/usb_stream)component and it didn't work.
Get this:
> E (2369) USB_STREAM: S:\Work\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.9\libraries\usb-stream\src\usb_stream.c:2734 (_uvc_uac_device_enum):Enum abort
E (2373) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Failed

leeebo, 2023-06-05:
@Gornerrrrr Please check the same issue if you got the descriptor failed https://github.com/espressif/esp-iot-solution/issues/267

Gornerrrrr, 2023-06-06:
@leeebo Hi thanks help!
I'm try add 200ms delay get "CHECK_FULL_DEV_DESC " error again.
> static esp_err_t _uvc_uac_device_enum(bool abort_process, bool *waiting_urb_done)
{
    vTaskDelay(pdMS_TO_TICKS(200)); // 添加延时, 尝试修复 CHECK_FULL_DEV_DESC 错误
    UVC_CHECK_GOTO(!abort_process, "Enum abort", stage_failed_);
    _usb_device_t *usb_dev = &s_usb_dev;

And then try run v4.4 [usb_host_lib](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib).

The successed log:
> ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x177c
load:0x4004c000,len:0xa48
load:0x40050000,len:0x2e58
entry 0x4004c204
I (21) boot: ESP-IDF v4.4-367-gc29343eb94-dirty 2nd stage bootloader
I (21) boot: compile time 09:18:49
I (21) boot: chip revision: 0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (30) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=09d14h ( 40212) map
I (99) esp_image: segment 1: paddr=00019d3c vaddr=3ffbd760 size=01da0h (  7584) load
I (101) esp_image: segment 2: paddr=0001bae4 vaddr=40022000 size=04534h ( 17716) load
I (112) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=193ech (103404) map
I (137) esp_image: segment 4: paddr=00039414 vaddr=40026534 size=07228h ( 29224) load
I (144) esp_image: segment 5: paddr=00040644 vaddr=50000000 size=00010h (    16) load
I (150) boot: Loaded app from partition at offset 0x10000
I (150) boot: Disabling RNG early entropy source...
I (165) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (166) cpu_start: Pro cpu up.
I (177) cpu_start: Pro cpu start user code
I (177) cpu_start: cpu freq: 160000000
I (177) cpu_start: Application information:
I (182) cpu_start: Project name:     usb_host_lib_example
I (188) cpu_start: App version:      v4.4-367-gc29343eb94-dirty
I (195) cpu_start: Compile time:     Jun  6 2023 09:33:06
I (201) cpu_start: ELF file SHA256:  0d9e0ef26c1c05e1...
I (207) cpu_start: ESP-IDF:          v4.4-367-gc29343eb94-dirty
I (213) heap_init: Initializing. RAM available for dynamic allocation:
I (221) heap_init: At 3FFBFD70 len 0003C290 (240 KiB): DRAM
I (227) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (233) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (240) spi_flash: detected chip: generic
I (244) spi_flash: flash io: dio
W (248) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (261) cpu_start: Starting scheduler on PRO CPU.
I (267) DAEMON: Installing USB Host Library
I (297) CLASS: Registering Client
I (867) CLASS: Opening device at address 1
I (867) CLASS: Getting device information
I (867) CLASS:  Full speed
I (867) CLASS:  bConfigurationValue 1
I (867) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
[... 140 lines cut ...]
I (150) boot: Disabling RNG early entropy source...
I (165) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (166) cpu_start: Pro cpu up.
I (177) cpu_start: Pro cpu start user code
I (177) cpu_start: cpu freq: 160000000
I (177) cpu_start: Application information:
I (182) cpu_start: Project name:     usb_host_lib_example
I (188) cpu_start: App version:      v4.4-367-gc29343eb94-dirty
I (195) cpu_start: Compile time:     Jun  6 2023 09:33:06
I (201) cpu_start: ELF file SHA256:  0d9e0ef26c1c05e1...
I (207) cpu_start: ESP-IDF:          v4.4-367-gc29343eb94-dirty
I (213) heap_init: Initializing. RAM available for dynamic allocation:
I (221) heap_init: At 3FFBFD70 len 0003C290 (240 KiB): DRAM
I (227) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (233) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (240) spi_flash: detected chip: generic
I (244) spi_flash: flash io: dio
W (248) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (261) cpu_start: Starting scheduler on PRO CPU.
I (267) DAEMON: Installing USB Host Library
I (297) CLASS: Registering Client
**E (807) HUB: Bad transfer status: CHECK_FULL_DEV_DESC
E (807) HUB: Stage failed: CHECK_FULL_DEV_DESC**

I had the same issue before https://github.com/espressif/esp-iot-solution/issues/254, The camera is the same model and firmware from the same manufacturer, how to fix this problem?

leeebo, 2023-06-06:
@Gornerrrrr What's the difference when you got `The successed log` and `The failed log`? Do you mean that the same hardware and software sometimes succeed and sometimes fail?

Gornerrrrr, 2023-06-06:
@leeebo I have 20 cameras. These cameras are the same model, firmware and batch. And then 15 cameras succeessed, 5 cameras failed. I don't know why.

leeebo, 2023-06-06:
@Gornerrrrr It seems to be more related to the hardware, how long is your USB data cable? you can try to use the logic analyzer to capture the USB packages to find out what happens

Gornerrrrr, 2023-06-06:
@leeebo The USB data cable just 50mm. If no more recommendation, I will buy analyzer.

Gornerrrrr, 2023-06-07:
@leeebo I solved the problem, adding 33pf to ground capacitor to the D+ D- signal line will no longer cause enumeration failures : )

leeebo, 2023-06-08:
@Gornerrrrr Congratulations! and please help close all related issues if which has been solved

Gornerrrrr, 2023-06-19:
@leeebo Sorry late reply.
I just referred to the ** _ESP32-S2 Hardware Design Guide _** tried to add two ground capacitors, did not expect to solve. Could be some kind of signal interference?
[image]

leeebo, 2023-07-10:
@Gornerrrrr As the error was fixed by hardware, we will close the issue

closed 2023-07-10 by leeebo