espressif/esp-iot-solution #262: Error "Pipe(uvc): ERROR_OVERFLOW" when testing USB camera with esp32s3

qsjhyy, 2023-05-10:
Development board: esp32s3
Development environment: Win10 + IDFv5.0.1 + esp iot solution
Camera: USB2.0 UVC MJPEG camera with plug-and-play support

I configured the usb_camera_mic_spk example according to the camera descriptor and burned it to test, but after communication, it reported an error "Pipe(uvc): ERROR_OVERFLOW".

```
        .format_index = 1, // Corresponding bFormatIndex for MJPEG, bFormatIndex: 0x02 (2)
        .frame_width = 160, // Horizontal pixel of MJPEG, such as 320
        .frame_height = 120, // Vertical pixel of MJPEG, such as 240
        .frame_index = 3, // Corresponding bFrameIndex for MJPEG 320*240, bFrameIndex: 0x05
        .frame_interval = 500000, // Optional frame rate dwFrameInterval, such as 30fps
        .interface = 1, // Optional video streaming interface bInterfaceNumber, usually 1
        .interface_alt = 3, // Interface option bAlternateSetting, bAlternateSetting: 0x04
        .ep_addr = 0x82, // Corresponding bEndpointAddress for interface option, bEndpointAddresS: 0x82 (Direction=IN EndpointID=2)
        .ep_mps = 512, // Determined MPS for interface option, such as 512
```

#157 Meanwhile, I also tried to modify the descriptions here, but failed. I am very confused. Is it due to the camera itself?

Log:
[log.txt](https://github.com/espressif/esp-iot-solution/files/11439093/log.txt)

leeebo, 2023-05-10:
Hi @qsjhyy the esp32s3 host is trying to `Probe MJPEG payload size = 512` but the camera returns 800, then the camera give packets with size 800, that's why the UVC pipe overflow (can only accept packet size <=512 Bytes).

[image]

BTW, below params is optional now, In most cases, you can leave it as 0

```
        .format_index = 1, // Corresponding bFormatIndex for MJPEG, bFormatIndex: 0x02 (2)
        .frame_index = 3, // Corresponding bFrameIndex for MJPEG 320*240, bFrameIndex: 0x05
        .interface = 1, // Optional video streaming interface bInterfaceNumber, usually 1
        .interface_alt = 3, // Interface option bAlternateSetting, bAlternateSetting: 0x04
        .ep_addr = 0x82, // Corresponding bEndpointAddress for interface option, bEndpointAddresS: 0x82 (Direction=IN EndpointID=2)
        .ep_mps = 512, // Determined MPS for interface option, such as 512
```

qsjhyy, 2023-05-11:
> Hi @qsjhyy the esp32s3 host is trying to `Probe MJPEG payload size = 512` but the camera returns 800, then the camera give packets with size 800, that's why the UVC pipe overflow (can only accept [...]

When I asked ChatGPT, it told me:
"Also, in the UVC protocol, MaxPayloadTransferSize refers to the maximum number of bytes transferred in a single data transfer, not the maximum packet size (MPS). Therefore, although the MPS is set to 512, the MaxPayloadTransferSize can be larger than 512, depending on the negotiation between the device and the host. This explains why the MaxPayloadTransferSize obtained in "GET_CUR Probe" is 800."

Of course, I am not sure if what it said is correct. Here is the attached conversation for inquiry. I look forward to your opinion and thank you in advance for your help.
[ask_chatgpt.txt](https://github.com/espressif/esp-iot-solution/files/11447504/ask_chatgpt.txt)
[ask_chatgpt(English).pdf](https://github.com/espressif/esp-iot-solution/files/11447528/ask_chatgpt.English.pdf)

leeebo, 2023-05-12:
@qsjhyy For ESP32-S3, due to the limited FIFO size, when setting MPS to 512 and if we received a packet exceeding the MPS size, the overflow error will be triggered

Gornerrrrr, 2023-06-06:
@leeebo I have same issue too. I'm use idfv4.4.4 and example USB_CAMERA_MIC_SPK test.

The camera descriptor:
```
*** Configuration descriptor ***
wTotalLength 451
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 2
        *** VS Format MJPEG Descriptor ***
        bFormatIndex 0x1
        bNumFrameDescriptors 3
        bDefaultFrameIndex 1
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x1
        wWidth 320
        wHeigh 240
        FrameInterval[0] 500000
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x2
        wWidth 176
        wHeigh 144
        FrameInterval[0] 500000
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x3
        wWidth 160
        wHeigh 120
        FrameInterval[0] 500000
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 1
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
                *** Endpoint descriptor ***
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 800
                bInterval 1
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 2
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
                *** Endpoint descriptor ***
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 700
                bInterval 1
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 3
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
                *** Endpoint descriptor ***
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 512
                bInterval 1
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 4
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
                *** Endpoint descriptor ***
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x5        ISOC
                wMaxPacketSize 192
                bInterval 1
```

The config auto set:
```
I (1505) USB_STREAM: Actual VS Interface(MPS <= 512) found, interface = 1, alt = 3
I (1513) USB_STREAM:    Endpoint(ISOC) Addr = 0x82, MPS = 512
I (1520) USB_STREAM: Actual MJPEG format index = 1, contains 3 frames
I (1527) USB_STREAM: Actual MJPEG width*heigh: 65535*65535, frame index = 1
I (1534) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Succeed
I (1542) USB_STREAM: ENUM Stage SET_CONFIG, Succeed
I (1548) uvc_mic_spk_demo: UVC: get frame list size = 3, current = 0
I (1554) uvc_mic_spk_demo:      frame[0] = 320x240
I (1559) uvc_mic_spk_demo:      frame[1] = 176x144
I (1564) uvc_mic_spk_demo:      frame[2] = 160x120
I (1569) uvc_mic_spk_demo: Device connected
I (1574) USB_STREAM: Creating UVC pipe: ifc=1-3, ep=0x82, mps=512
I (1581) USB_STREAM: Probe Format(1) MJPEG, Frame(1) 320*240, interval(666666)
I (1588) USB_STREAM: Probe payload size = 512
```

When alt=3 mps=512 is no problem. But get:
```
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 500000
dwMaxPayloadTransferSize: 800
bInterfaceNumber: 0
I (3781) USB_STREAM: Sample processing task created
I (3781) USB_STREAM: Set Device Interface = 1, Alt = 3
I (3782) USB_STREAM: Sample processing task start
I (3792) USB_STREAM: Set Device Interface Done
W (4920) USB_STREAM: Pipe(UVC): ERROR_OVERFLOW
I (4920) USB_STREAM: usb stream task wait reset
W (5920) USB_STREAM: usb stream task recover, reason: stream error
I (5920) USB_STREAM: Resetting UVC pipe
```

So it's set failed? Or do I need to set it up manually?
Please provide some suggestions, thanks!

qsjhyy, 2023-06-07:
> Now I'm manually set alt=4, It works fine. [image] [image]

However, when I turn on Wi-Fi and connect the camera, it enters a continuous rebooting state.

leeebo, 2023-07-10:
Link #254

closed 2023-07-10 by leeebo

Hard-bitten, 2024-06-05:
@qsjhyy 后来怎么样了？解决了吗？

qsjhyy, 2024-06-06:
> @qsjhyy 后来怎么样了？解决了吗？

[Gornerrrrr](https://github.com/Gornerrrrr)根据这位老兄上面说的方法，也许算是解决了。我后面的问题可能是其他原因导致的。

dzz10, 2024-10-14:
@qsjhyy 按照上面的方法改了之后，我这里不报错了，但是画面只有一小部分，日志显示一帧只有5000byte
你这边有什么好的建议吗
I (459246) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6862, width = 640, height = 480, length = 5088, ptr = 0
I (459311) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6863, width = 640, height = 480, length = 5072, ptr = 0
I (459378) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6864, width = 640, height = 480, length = 5088, ptr = 0
I (459446) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6865, width = 640, height = 480, length = 5088, ptr = 0
[image]