espressif/esp-iot-solution #265: ESP32 WITH USB CAMERA WIFI TRANSFER

Nadeemranm, 2023-05-22:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      |ESP32-DevKitC|]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         |CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    //
- Operating System:     [Windows|Linux|]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior 
I have two  usb cameras 
1)
Backtrace: 0x40379292:0x3FC989F0 0x403773E5:0x3FC98A10 0x4200C16A:0x3FCDA0E0 0x4200ECA5:0x3FCDA170 0x4200ADCA:0x3FCDA190

E (266724) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (266724) task_wdt:  - IDLE (CPU 1)
E (266724) task_wdt: Tasks currently running:
E (266724) task_wdt: CPU 0: IDLE
E (266724) task_wdt: CPU 1: usb_proc
E (266724) task_wdt: Print CPU 1 backtrace
2)
I (6957) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:34
I (6963) UVC_STREAM: SET_CUR COMMIT
I (6967) UVC_STREAM: SET_CUR COMMIT Done
I (6972) UVC_STREAM: Set Device Interface = 1, Alt = 1
I (6978) UVC_STREAM: Set Device Interface Done
I (6983) UVC_STREAM: Sample processing task started
I (6988) UVC_STREAM: Camera Start Streaming
### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

[image]

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
 /* create eventgroup for task sync */
    s_evt_handle = xEventGroupCreate();
    if (s_evt_handle == NULL) {
        ESP_LOGE(TAG, "line-%u event group create failed", __LINE__);
        assert(0);
    }

    /* init wifi */
    app_wifi_main();

    /* start http server*/
    app_httpd_main();

    /* malloc double buffer for usb payload, xfer_buffer_size >= frame_buffer_size*/
    uint8_t *xfer_buffer_a = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(xfer_buffer_a != NULL);
    uint8_t *xfer_buffer_b = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(xfer_buffer_b != NULL);

    /* malloc frame buffer for a jpeg frame*/
    uint8_t *frame_buffer = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(frame_buffer != NULL);

    /* the quick demo skip the standard get descriptors process,
    users need to get params from camera descriptors,
    run the example first to printf the log with descriptors */
    uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL,
        .xfer_type = DEMO_XFER_MODE,
        .configuration = DESCRIPTOR_CONFIGURATION_INDEX,
        .format_index = DESCRIPTOR_FORMAT_MJPEG_INDEX,
        .frame_width = DEMO_FRAME_WIDTH,
        .frame_height = DEMO_FRAME_HEIGHT,
        .frame_index = DEMO_FRAME_INDEX,
        .frame_interval = DEMO_FRAME_INTERVAL,
        .interface = DESCRIPTOR_STREAM_INTERFACE_INDEX,
        .interface_alt = DEMO_INTERFACE_ALT,
        .isoc_ep_addr = DESCRIPTOR_STREAM_ENDPOINT_ADDR,
        .isoc_ep_mps = DEMO_EP_MPS,
        .xfer_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .xfer_buffer_a = xfer_buffer_a,
        .xfer_buffer_b = xfer_buffer_b,
        .frame_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .frame_buffer = frame_buffer,
    };

    /* pre-config UVC driver with params from known USB Camera Descriptors*/
    esp_err_t ret = uvc_streaming_config(&uvc_config);

    /* Start camera IN stream with pre-configs, uvc driver will create multi-tasks internal
    to handle usb data from different pipes, and user's callback will be called after new frame ready. */
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "uvc streaming config failed");
    } else {
        uvc_streaming_start(frame_cb, NULL);
    }

    while (1) {
        /* task monitor code if necessary */
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

}

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Nadeemranm, 2023-05-23:
help

leeebo, 2023-05-23:
Hi @Nadeemranm please try the latest examples in the master branch https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk

Nadeemranm, 2023-05-23:
when i run this code got random data on terminal
i have also set the baud rate 115200
still getting random data
i have atteched

usb_camera_mic_spk\build\usb_camera_mic_spk.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM15 instead...
--- idf_monitor on \\.\COM15 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1684
load:0x403c9700,len:0xbf8
load:0x403cc700,len:0x2e98
entry 0x403c990c
�A�,0�KJ�%
          3<�=(��)�d/�2�e���Ƃ����cCւtԳ:rւ�u�1��`Ƃ u�1�� �� m2p�p ]'��}#�}#�� Ƃ}#�~�$% �"ݣ�tb��}#A��3�$�`ע!m�)��!���3�$�|#�� �� m�(�� J2!䴃d��e���a�t��)�%�    �)��#�-���#�
                                                                                                                                                                          `��me�4��k0~
I$�
   ���
      `C$�C��        ���[�1ܢd�� �)(<!�d�%��r��қ��#�趒��9��h#�
                                                             1�\
                                                                ��gmeQ�b�$�t   ���u��f��5��� �e��g���F��u��e�d)�ؖ#��U�u�$21K�%     ��-��#�!
                                                                                                                                          ���$�3-�-K`0��k�#�5���K����
                                                                                                                                                                     ��
                                                                                                                                                                       8���z
!�փ��+��!
         �!
           ��˨�
C�}�t�1��!KY��i
               \��!
                  ؂���
d�s�|x�t4�
��%c3a�`��T4�
�!��`a��h�䬚t�t��h��T4�
���a�1\�@:*��ƪA�1|�`#�d2��p�|�`�Ad2��a�1|�@:#
�m�p�|�`#�d2��a�1|#K��#
v�M���@��X�� R�
               X�C=     �-�v�`�x��x�)���#2%15 Q�#�m��!˭��E��0�y�

qsjhyy, 2023-05-26:
> when i run this code got random data on terminal i have also set the baud rate 115200 still getting random data i have atteched
> usb_camera_mic_spk\build\usb_camera_mic_spk.elf --- WARNING: GDB cannot open serial ports accessed as COMx --- Using \\.\COM15 instead... --- idf_monitor on \\.\COM15 115200 --- --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --- ESP-ROM:esp32s3-20210327 Build:Mar 27 2021 rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT) SPIWP:0xee mode:DIO, clock div:1 load:0x3fce3810,len:0x1684 load:0x403c9700,len:0xbf8 load:0x403cc700,len:0x2e98 entry 0x403c990c �A�,0�KJ�% 3<�=(��)�d/�2�e���Ƃ����cCւtԳ:rւ�u�1��`Ƃ u�1�� �� m2p�p ]'��}#�}#�� Ƃ}#�~�$% �"ݣ�tb��}#A��3�$�`ע!m�)��!���3�$�|#�� �� m�(�� J2!䴃d��e���a�t��)�%� �)��#�-���#� `��me�4��k0~ I$� ��� `C$�C�� ���[�1ܢd�� �)(<!�d�%��r��қ��#�趒��9��h#� 1�\ ��gmeQ�b�$�t ���u��f��5��� �e��g���F��u��e�d)�ؖ#��U�u�$21K�% ��-��#�! ���$�3-�-K`0��k�#�5���K���� �� 8���z !�փ��+��! �! ��˨� C�}�t�1��!KY��i \��! ؂��� d�s�|x�t4� ��%c3a�`��T4� �!��`a��h�䬚t�t��h��T4� ���a�1\�@:*��ƪA�1|�`#�d2��p�|�`�Ad2��a�1|�@:# �m�p�|�`#�d2��a�1|#K��# v�M���@��X�� R� X�C= �-�v�`�x��x�)���#2%15 Q�#�m��!˭��E��0�y�

I'm not sure why the default print baud rate is set to 2000000. However, I have encountered this issue before, and it seems to be set in the "sdkconfig" file, possibly to keep up with the camera data.

As for finding a camera with MPS 512, like in #262, it might require some research and exploration.

Nadeemranm, 2023-05-26:
Sir  now i faced this type error
i have attached camera descriptor file and terminal output msg

qsjhyy, 2023-05-27:
[image]

I still cannot set the MPS of the camera, even if the camera itself has an interface with MPS of 512.

leeebo, 2023-05-29:
@qsjhyy @Nadeemranm The example UART baud rate is relatively high, because the USB synchronous has the requirement of latency, in order to prevent the interference of print when debugging.

The ESP32-S3 does have some restrictions on camera selection due to [MPS limitations](https://docs.espressif.com/projects/espressif-esp-iot-solution/zh_CN/latest/usb/usb_stream.html#id1). If you can not found the suitable camera, It is recommended that purchase the [ESP32-S3-LCD-EV-Board (with a 720P USB Camera)](https://m.tb.cn/h.UDepUJP?tk=TpmvdLBsNKX) to a quick start.

If you are a commercial user and need to purchase related cameras in bulk, please contact sales@espressif.com Business

lijunru-hub, 2023-10-08:
Update camera link [ESP32-S3-LCD-EV-Board (with a 720P USB Camera)](https://item.taobao.com/item.htm?spm=a21n57.1.0.0.76a3523c3CXp5F&id=680580609719&ns=1&abbucket=17#detail)

closed 2024-05-21 by leeebo