espressif/esp-iot-solution #266: usb camera with esp32s3 +wifi transfer not working

nadeem268, 2023-05-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32-DevKitC
- Kit version : DevKitC v1
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version : xtensa-esp32-elf-gcc 
- Operating System:     Linux
- Power Supply:         [USB|

## Problem Description

  I tried to use camera +wifi  transfer example from the esp-iot-solution-usb-add_usb_solution_libarary code build properly no error  but i don,t get result properly  
i do not have recived ip address 
[image]

### Expected Behavior
 i want live streaming through the usb camera + wifi transfer
### Actual Behavior
code run properly but result not matched not  mcu connected to the wifi but not generate the  ip address 
### Steps to repropduce

1. step1
2. 
[image]
 

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
#define DESCRIPTOR_CONFIGURATION_INDEX 1
#define DESCRIPTOR_FORMAT_MJPEG_INDEX  1
#define DESCRIPTOR_STREAM_INTERFACE_INDEX 1
#define DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR 0x87

#define DEMO_FRAME_WIDTH      640
#define DEMO_FRAME_HEIGHT     480
#define DEMO_XFER_BUFFER_SIZE (64*1024) //Double buffer
#define DEMO_FRAME_INDEX 3
[... 136 lines cut ...]

0x403773e5: _xt_lowint1 at /home/nadeem/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

0x4200eafe: usb_print_config_descriptor at /home/nadeem/esp/esp-idf/components/usb/usb_helpers.c:282

0x4200adc6: _usb_get_config_desc at /home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions/components/usb/uvc_stream/uvc_stream.c:905
 (inlined by) _usb_processing_task at /home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions/components/usb/uvc_stream/uvc_stream.c:1481

E (11814) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (11814) task_wdt:  - IDLE (CPU 1)
E (11814) task_wdt: Tasks currently running:
E (11814) task_wdt: CPU 0: IDLE
E (11814) task_wdt: CPU 1: usb_proc
E (11814) task_wdt: Print CPU 1 backtrace

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Nadeemranm, 2023-05-25:
sir???

nadeem268, 2023-05-25:
sir now faced this type of error
I (33814) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:34
I (33821) UVC_STREAM: SET_CUR COMMIT
I (33825) UVC_STREAM: SET_CUR COMMIT Done
I (33830) UVC_STREAM: Set Device Interface = 131, Alt = 4
I (33836) UVC_STREAM: Set Device Interface Done
I (33841) UVC_STREAM: Sample processing task started
I (33846) UVC_STREAM: Camera Start Streaming
W (34416) UVC_STREAM: line 693 Pipe: iso HCD_PIPE_EVENT_ERROR_OVERFLOW

leeebo, 2023-05-29:
Same issue with #255

Nadeemranm, 2023-05-29:
39503) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512 (0c
8c...a4 7d)
W (39599) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...cc 81)
W (39695) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...83 7f)
W (39791) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...c7 79)
W (39891) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...b7 80)
W (39987) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...c7 80)
W (40083) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...c2 7f)
W (40179) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...c2 80)
W (40279) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...98 81)
W (40375) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...85 86)
W (40471) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...b7 81)

 what is this problem sir?
pls help sir

Nadeemranm, 2023-05-29:
sir i want to interface only usb camera not MIC and not SPK
 i am using "USB_CAMERA_MIC_SPK " code
when  set the manually camera all parameters  and then a upload the in my
MCU
By defult "uvc_mic_spk_demo  function run itself
so that i have already comment this function

Nadeemranm, 2023-05-29:
I (25) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (25) boot: compile time 17:46:05
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0be7ch (
48764) map
I (101) esp_image: segment 1: paddr=0001bea4 vaddr=3fc91c00 size=02378h (
 9080) load
I (104) esp_image: segment 2: paddr=0001e224 vaddr=40374000 size=01df4h (
 7668) load
I (111) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=20f94h
(135060) map
I (142) esp_image: segment 4: paddr=00040fbc vaddr=40375df4 size=0bdf4h (
48628) load
I (159) boot: Loaded app from partition at offset 0x10000
I (159) boot: Disabling RNG early entropy source...
I (170) cpu_start: Pro cpu up.
I (171) cpu_start: Starting app cpu, entry point is 0x403751b8
I (0) cpu_start: App cpu up.
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 240000000 Hz
I (185) cpu_start: Application information:
I (188) cpu_start: Project name:     usb_host_lib_example
I (194) cpu_start: App version:      1
I (199) cpu_start: Compile time:     May 29 2023 18:56:48
I (205) cpu_start: ELF file SHA256:  6ee9875a326e81e7...
ELF file not found. You need to build & flash the project before running
'monitor', and the binary on the device must match the one in the build
directory exactly.
I (211) cpu_start: ESP-IDF:          v5.0.1-dirty
I (216) cpu_start: Min chip rev:     v0.0
I (221) cpu_start: Max chip rev:     v0.99
I (226) cpu_start: Chip rev:         v0.1
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (238) heap_init: At 3FC949B0 len 00054D60 (339 KiB): D/IRAM
I (244) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (251) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (257) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (264) spi_flash: detected chip: generic
I (268) spi_flash: flash io: dio
W (272) spi_flash: Detected size(8192k) larger than the size in the binary
image header(2048k). Using the size in the binary image header.
I (285) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (306) DAEMON: Installing USB Host Library
I (336) CLASS: Registering Client
E (816) HUB: Incorrect number of bytes returned: CHECK_FULL_CONFIG_DESC
E (816) HUB: Stage failed: CHECK_FULL_CONFIG_DESC
  what is the solution of this error
when i plagin my usb camera  popup this types of error  in terminal

leeebo, 2023-05-30:
```
W (40375) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...85 86)
W (40471) UVC_STREAM: Transfer buffer overflow, got data=49664 B, last=512
(0c 8c...b7 81)
```
@Nadeemranm 
This means you have successfully started a frame stream, but the frame is larger than the buffer size you config. which causes the Transfer buffer overflow, please give a larger `xfer_buffer` and `frame_buffer` to make it work. or chose a small resolution with a  smaller frame size

Nadeemranm, 2023-05-30:
Sir when i increase the "DEMO_UVC_XFER_BUFFER_SIZE   (83*1024)
this types error occur
E (9435) UVC_STREAM:
/home/nadeem/Downloads/esp-iot-solution-master/components/usb/usb_stream/usb_stream.c:826
(_usb_pipe_init):pipe alloc failed
E (9448) UVC_STREAM:
/home/nadeem/Downloads/esp-iot-solution-master/components/usb/usb_stream/usb_stream.c:2645
(_usb_stream_handle_task):pipe init failed
I (9463) UVC_STREAM: usb stream task reset
I (9468) UVC_STREAM: Resetting stream Driver
I (9473) UVC_STREAM: usb stream task reset done
I (9479) UVC_STREAM: Creating uvc in(isoc) pipe itf = 1-3, ep = 0x83

Nadeemranm, 2023-05-30:
My usb camera is support the USB 2.0
but the uvc 1.1 campaatble only

Nadeemranm, 2023-06-05:
> @Nadeemranm <https://github.com/Nadeemranm> Congratulations, For wifi
> transfer test, have you enabled the WIFI Transfer macro?
> https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk#usb-stream-example

usb camera lcd display code not  are the campile
when i do the run usb camera lcd display
I have faced this type of error
error msg is no version of idf match >=4.4.1,<4.5.0
and project depends on idf (=4.4),project requires idf(>=4.4.0<4.4.1>
so because project depends on the idf (>=4.4.1) ,version solving failed
pls help me to resolve this error
Thank you!!

nopnop2002, 2023-07-06:
@Nadeemranm 

Try this.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc

Cameras that can be used with esp-idf are limited.

leeebo, 2023-07-10:
Close reason: duplicate issue #265

closed 2023-07-10 by leeebo

Nadeemranm, 2023-08-10:
Hello sir good morning
i am using this"
https://github.com/0015/ESP32-S2-S3-Projects/tree/main/usb_camera_lcd_display
"
i  am using ESP32-S3-DevkitC-1 and usb camera and ili9488 16 bit parallel
line interface lcd
my usb camera working properly  and lcd little bit working not properly i
means  usb camera live streaming not displaying on lcd
I (0) cpu_start: App cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 240000000 Hz
I (210) cpu_start: Application information:
I (213) cpu_start: Project name:     uvc_mjpeg_lcd_lovyangfx
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Aug  9 2023 18:17:54
I (230) cpu_start: ELF file SHA256:  0926a7a732204901...
I (236) cpu_start: ESP-IDF:          v5.0-541-g885e501d99-dirty
I (243) cpu_start: Min chip rev:     v0.0
I (248) cpu_start: Max chip rev:     v0.99
I (252) cpu_start: Chip rev:         v0.1
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FC958D0 len 00053E40 (335 KiB): D/IRAM
I (271) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (277) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (284) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (290) spi_flash: detected chip: generic
I (295) spi_flash: flash io: qio
W (299) spi_flash: Detected size(8192k) larger than the size in the binary
image header(4096k). Using the size in the binary image header.
I (312) sleep: Configure to isolate all GPIO pins in sleep state
I (318) sleep: Enable automatic switching of GPIO sleep configuration
I (326) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (341) gpio: GPIO[47]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
Pulldown: 0| Intr:0
I (590) UVC_STREAM: Isoc transfer mode
I (590) UVC_STREAM: UVC Streaming Config Succeed
I (640) UVC_STREAM: UVC Streaming Starting
I (670) UVC_STREAM: Port=1 init succeed
I (670) UVC_STREAM: Waiting USB Connection
I (920) UVC_STREAM: line 367 HCD_PORT_EVENT_CONNECTION
I (970) UVC_STREAM: Resetting Port
I (1030) UVC_STREAM: Setting Port FIFO
I (1030) UVC_STREAM: Getting Port Speed
I (1030) UVC_STREAM: Port speed = 1
I (1032) UVC_STREAM: USB Speed: full-speed
I (1037) UVC_STREAM: Set Device Addr = 1
I (1042) UVC_STREAM: Set Device Addr Done
I (1046) UVC_STREAM: get device desc
I (1053) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0xab20
idProduct 0x1276
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (1080) UVC_STREAM: get short config desc
I (1086) UVC_STREAM: get config desc, actual_num_bytes:16
W (1091) UVC_STREAM: Configuration descriptor larger than control transfer
max length
I (1099) UVC_STREAM: Set Device Configuration = 1
[... 79 lines cut ...]
height = 240, length = 6340, ptr = 1070162196
E (3104) uvc_demo: callback! frame_format = 7, seq = 27, width = 320,
height = 240, length = 6324, ptr = 1070162196
E (3172) uvc_demo: callback! frame_format = 7, seq = 28, width = 320,
height = 240, length = 6329, ptr = 1070162196
E (3236) uvc_demo: callback! frame_format = 7, seq = 29, width = 320,
height = 240, length = 6320, ptr = 1070162196
E (3304) uvc_demo: callback! frame_format = 7, seq = 30, width = 320,
height = 240, length = 6322, ptr = 1070162196
E (3372) uvc_demo: callback! frame_format = 7, seq = 31, width = 320,
height = 240, length = 6344, ptr = 1070162196
E (3436) uvc_demo: callback! frame_format = 7, seq = 32, width = 320,
height = 240, length = 6355, ptr = 1070162196
E (3504) uvc_demo: callback! frame_format = 7, seq = 33, width = 320,
height = 240, length = 6369, ptr = 1070162196
E (3572) uvc_demo: callback! frame_format = 7, seq = 34, width = 320,
height = 240, length = 6365, ptr = 1070162196
E (3636) uvc_demo: callback! frame_format = 7, seq = 35, width = 320,
height = 240, length = 6359, ptr = 1070162196
E (3704) uvc_demo: callback! frame_format = 7, seq = 36, width = 320,
height = 240, length = 6366, ptr = 1070162196
E (3772) uvc_demo: callback! frame_format = 7, seq = 37, width = 320,
height = 240, length = 6350, ptr = 1070162196
E (3836) uvc_demo: callback! frame_format = 7, seq = 38, width = 320,
height = 240, length = 6368, ptr = 1070162196

Nadeemranm, 2023-08-14:
Live streaming working fine !
but i have an problem how to set resolution like 320*240 and 320*480
tft.pushImage(0,0 ,320,240 (uint16_t *)data);
What is the right resolution ?