espressif/esp-iot-solution #267: usb_camera_mic_spk and usb_cdc not working : get_short_dev_desc not working

sebbrz, 2023-05-29:
## Environment

- Development Kit: ESP32-S3-DevKitM N16R8
- Module or chip used:  ESP32S3
- IDF version : v5.2-dev-350-g56123c52aa
- Build System:         CMake
- Compiler version : esp-12.2.0_20230208
- Operating System:     Linux
- Power Supply:         USB
- USB camera : D- on GPIO19, D+ on GPIO20, external USB power supply.

## Problem Description
Hello,
I'm trying to use the usb_camera_mic_spk (with mic disabled) in order to stream a UVC USB2.0 camera.
When I connect the camera (powered by an external USB connection), I can see it's detected, but right after the GET_SHORT_DEV_DESC, the program enters a loop, and does not proceed to the CHECK_SHORT_DEV_DESC step, see snipet below :
I (141) USB_STREAM: UVC Streaming Config Succeed, Version: 1.0.2
I (141) uvc_mic_spk_demo: uvc streaming config OK
I (141) USB_STREAM: USB streaming callback register succeed
I (142) USB_STREAM: Pre-alloc ctrl urb succeed, size = 1024
D (142) USB_STREAM: Camera instance created
I (142) USB_STREAM: USB task start
D (143) intr_alloc: Connected src 38 to int 19 (cpu 1)
I (143) USB_STREAM: USB stream task start
I (173) USB_STREAM: Port=1 init succeed, context = 0x3fcbfffc
I (173) USB_STREAM: Waiting USB Connection
I (173) USB_STREAM: USB Streaming Start Succeed
I (174) USB_STREAM: Waiting USB Device Connection
I (3525) USB_STREAM: line 152 HCD_PORT_EVENT_CONNECTION
D (3525) USB_STREAM: port update: action_bits = 0x0008
I (3525) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (3576) USB_STREAM: Resetting Port
I (3636) USB_STREAM: Setting Port FIFO, 1
D (3636) USB_STREAM: Action: ACTION_DEVICE_CONNECT, Done! (1024)
D (3636) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage NONE (0)
E (3637) USB_STREAM: Control URB NULL
I (3637) USB_STREAM: USB Speed: full-speed
D (3637) USB_STREAM: pipe(0x3fcc0138) init succeed
I (3637) USB_STREAM: ENUM Stage START, Succeed
D (3638) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (0), Action Bits (1024)
D (3639) USB_STREAM: Action: ACTION_DEVICE_ENUM Stage GET_SHORT_DEV_DESC (2)
E (3639) USB_STREAM: Control URB Not NULL
I (3639) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
D (3640) USB_STREAM: Action: ACTION_DEVICE_ENUM, Waiting URB Done, Action Bits (0)
D (3640) USB_STREAM: Action: ACTION_DEVICE_ENUM, Done (2), Action Bits (0)
D (3828) USB_STREAM: USB task running 300
D (7128) USB_STREAM: USB task running 600
D (10428) USB_STREAM: USB task running 900
D (13728) USB_STREAM: USB task running 1200
D (17028) USB_STREAM: USB task running 1500
D (20328) USB_STREAM: USB task running 1800
D (23628) USB_STREAM: USB task running 2100

I modified the usb_stream.c file code, since the action_bits variable goes to zero right after the GET_SHORT_DEV_DESC stage.
I tried to force CHECK_SHORT_DEV_DESC stage, but got the program to crash :/.

When I try the usb_cdc_4g_module, I get the following error : usbh_get_dev_desc):urb event error
 
I (3611) 4g_main: ====================================
I (3611) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3611) 4g_main: ====================================
I (3617) modem_board: iot_usbh_modem, version: 0.1.5
I (3622) modem_board: Force reset modem board....
I (3628) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (3637) modem_board: Resetting modem using io=13, level=0
I (3843) modem_board: Waiting for modem initialize ready
I (8843) USB_HCDC: iot_usbh_cdc, version: 0.1.3
I (8843) IOT_USBH: iot_usbh, version: 0.1.2
I (8873) IOT_USBH: USB Port=1 init succeed, fifo strategy=0
I (8873) IOT_USBH: usb port start succeed
I (8873) USB_HCDC: CDC task start
I (8873) USB_HCDC: usbh cdc driver install succeed
I (8882) USB_HCDC: Waiting CDC Device Connection
I (8873) IOT_USBH: USB Processing Start
I (8892) IOT_USBH: Waiting USB Connection
I (8896) IOT_USBH: Port power: ON Succeed
I (12089) IOT_USBH: line 131 PORT_EVENT_CONNECTION
I (12189) IOT_USBH: Resetting Port
I (12249) IOT_USBH: Port reset succeed
I (12249) IOT_USBH: Getting Port Speed
I (12249) IOT_USBH: USB Speed: full-speed
I (12252) IOT_USBH: Pipe init succeed, addr: 00
I (12257) IOT_USBH: Set Device Addr = 1
I (12261) IOT_USBH: Set Device Addr Done
I (12266) IOT_USBH: get device desc
**E (13270) IOT_USBH: /home/sebastien/esp/esp-iot-solution/components/usb/iot_usbh/iot_usbh.c:775 (_usbh_get_dev_desc):urb event error
E (13272) IOT_USBH: /home/sebastien/esp/esp-iot-solution/components/usb/iot_usbh/iot_usbh.c:1009 (_usbh_processing_task):Get device descriptor failed**
W (13286) IOT_USBH: Resetting USB Driver
W (13290) IOT_USBH: Resetting default pipe
I (13395) IOT_USBH: iot_usbh_pipe_deinit: pipe state = 0
I (13395) IOT_USBH: Pipe deinit succeed, addr: 00
I (13395) IOT_USBH: hcd recovering
I (13399) IOT_USBH: Port power: ON Failed
I (13504) IOT_USBH: Waiting USB Connection

I've tested the camera with v4l2 without any problem, see below : 
Driver Info:
	Driver name      : uvcvideo
	Card type        : USB Camera: USB Camera
	Bus info         : usb-0000:07:00.3-2
	Driver version   : 5.19.17
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : USB Camera: USB Camera
	Serial           : SN0001
	Bus info         : usb-0000:07:00.3-2
	Media version    : 5.19.17
	Hardware revision: 0x00000100 (256)
	Driver version   : 5.19.17
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : USB Camera: USB Camera
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x0100000d   : 0: Sink
	  Link 0x0200001a: from remote pad 0x1000010 of entity 'Extension 4' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1843789
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 10000000.000 (10000000/1)
	Read buffers     : 0

I don't understand what is going wrong with this camera. 
I chose it so that it would be fully compatible with the usb_camera HTTP transfer.

I downloaded the latest version of IDF and usb_stream (1.0.2) module without success.

Would you have any clue of what's going on ? It seems the short device description is neither accessible nor existant.
Do you know if there is a USB device testing program for ESPS3 ?

Thank you in advance for your kind help !

Sebastien

leeebo, 2023-05-30:
@sebbrz Hi
1. Please try to add a delay time between each enum stage to give it a try (eg. add a `vTaskDelay(200 / portTICK_PERIOD_MS);` in the begin `_uvc_uac_device_enum`)
2. If that does not work, please try with the [basic USB example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) to see if we can be enumerated properly

If both do not work, probably because your device doesn't support working in full-speed mode. It is recommended that purchase the [ESP32-S3-LCD-EV-Board (with a 720P USB Camera)](https://m.tb.cn/h.UDepUJP?tk=TpmvdLBsNKX) for a quick start.

If you are a commercial user and need to purchase related cameras in bulk, please contact [sales@espressif.com](mailto:sales@espressif.com)

sebbrz, 2023-05-30:
@leeebo Thanks a lot for your quick help !
It's a personal project, I'm not a commercial user :) 
I'm trying to build an Allsky Wifi Camera, using a very sensitive IMX290 sensor, so that I can have timelapse of the sky, even at night. 
I already use another old version of ESP32 to collect all the weather condition data, and it works wonderfully.
Looking at the ESP32-S3-LCD-EV Board, it seems that the S3 specs on this board are quite the same as mine (16MB Flash and 8-line 8MB PSRAM memory). I've already tried to improve my board speed by disabling Wifi and Mic on the example with no success at that time.
I'll give a try to the basic example, and to adding delay between each enum step, and get back to you, this evening.
Thanks again for you help :)
Sebastien

sebbrz, 2023-05-30:
@leeebo Hi again
I have the exact same result with the basic USB example :
the program stops right after the enum stage GET_SHORT_DEV_DESC, even if I had the delay in the while (1) loop :

I (782) main_task: Calling app_main()
I (792) DAEMON: Installing USB Host Library
V (792) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (802) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x802
D (812) intr_alloc: Connected src 38 to int 17 (cpu 0)
I (842) CLASS: Registering Client
D (5782) HUB: Root port reset
D (5782) HUB: Stage done: START
D (5782) HUB: Stage done: GET_SHORT_DEV_DESC
D (12702) HUB: Stage done: CLEANUP_FAILED
D (12702) HUB: Recovering root port
D (12702) HUB: Stage done: NONE
D (16552) HUB: Root port reset
D (16552) HUB: Stage done: START
D (16552) HUB: Stage done: GET_SHORT_DEV_DESC
D (21842) HUB: Stage done: CLEANUP_FAILED
D (21842) HUB: Recovering root port
D (21842) HUB: Stage done: NONE
D (24092) HUB: Root port reset
D (24092) HUB: Stage done: START
D (24092) HUB: Stage done: GET_SHORT_DEV_DESC

I've contacted the board reseller in order to confirm the board is able to handle USB full speed mode.
I'll keep you posted :)

Best regards

Sebastien

closed 2023-07-10 by leeebo