espressif/esp-iot-solution #271: Migrate ADC driver to New driver in bus component

nopnop2002, 2023-06-03:
Migrate ADC driver to New driver in bus component

I want to migrate adc1_get_raw() to adc_oneshot_read().

But this is using driver/i2s.h here so it can't be migrated.

https://github.com/espressif/esp-iot-solution/blob/master/components/bus/include/i2s_lcd_driver.h#L17

```
[908/915] Building C object esp-idf/tft_library/CMakeFiles/__idf_tft_library.dir/i2s_lcd_esp32_driver.c.objIn file included from /home/nop/esp-idf/components/driver/deprecated/driver/i2s.h:23,
                 from ../components/tft_library/i2s_lcd_driver.h:17,
                 from ../components/tft_library/i2s_lcd_esp32_driver.c:35:
/home/nop/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
```

leeebo, 2023-06-06:
Hi @nopnop2002 IDF 5.0 dropped some APIs, which caused this problem. If you are using IDF 5.0 for a new project and want to use the I2S interface to drive the screen, we suggest you use the new [esp_lcd ](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) component instead

leeebo, 2023-06-06:
If you just want to use I2C/SPI in the bus component, you can just move i2s_lcd_esp32s2_driver.c i2s_lcd_esp32_driver.c out from the CMakeLists.

nopnop2002, 2023-06-06:
>we suggest you use the new [esp_lcd ](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd)component instead

Thank you for your comment.

I'll look into the esp_lcd component over the weekend.