espressif/esp-iot-solution #272: esp32s3 usb camera lcd display code not campile

nadeem268, 2023-06-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC: v1
- Module or chip used:  [ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it): 
    // v5.0-dev-1148-g96cd3b75c
- Build System:         |CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     |Linux|
- Power Supply:         [USB|

When i run usb camera with lcd code  
code not compailing properly i got some error msg 
olving dependencies requirements

..CMake Error at /home/nadeem/esp/esp-idf/tools/cmake/build.cmake:519 (message):
  ERROR: Because no versions of idf match >=4.4.1,<4.5.0

   and project depends on idf (=4.4), project requires idf (>=4.4.0,<4.4.1).

  So, because project depends on idf (>=4.4.1), version solving failed.

Call Stack (most recent call first):
  /home/nadeem/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!

//Detailed problem description goes here

### Expected Behavior

### Actual Behavior
code not campiling error msg 
and project depends on idf (=4.4), project requires idf (>=4.4.0,<4.4.1).

  So, because project depends on idf (>=4.4.1), version solving failed.
And i am using the v5.0 version of esp-idf

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

leeebo, 2023-06-05:
@nadeem268 ESP32 does Not support USB function. Please choose the ESP32S3 board, and set the target using `idf.py set-target esp32s3` first.

lijunru-hub, 2023-06-07:
best use ESP32-S3-LCD-EV-Board and a usb-camera

closed 2023-07-10 by leeebo