espressif/esp-iot-solution #273: Support of the capacitive touch in iot_button

espzav, 2023-06-05:
Hi,
please, do you plan to add support for capacitive touch buttons into the iot_button component? We are using this component in ESP-BSP and we would like to unify code for GPIO/ADC/Touch buttons. 

Thank you.

leeebo, 2023-06-05:
Hi @espzav yes I think officially support Touch sensor in button compoment can be good, However, compared with GPIO/ADC, Touch sensor is much more complicated. Users must adjust multiple parameters according to their own hardware, which may greatly increase the complexity of iot_button component.  We'll think about it, but it might take some time to think about it

lijunru-hub, 2024-07-08:
If integrating touch buttons temporarily within regular buttons is not feasible, you can use the custom_button type within buttons. Users will handle a series of touch button detections.

lijunru-hub, 2025-03-20:
We have developed a series of touch software state machine components and encapsulated the touch detection mechanism.

- [touch_sensor_fsm](https://components.espressif.com/components/espressif/touch_sensor_fsm/versions/0.5.2~2)
- [touch_sensor_lowlevel](https://components.espressif.com/components/espressif/touch_sensor_lowlevel/versions/0.7.0) 
- [touch_proximity_sensor](https://components.espressif.com/components/espressif/touch_proximity_sensor/versions/0.2.0)
- [touch_button_sensor](https://components.espressif.com/components/espressif/touch_button_sensor/versions/0.1.0)
- touch_button: Coming soon.