espressif/esp-iot-solution #275: 烧写官方代码示例，出现报错怎么解决

efffx, 2023-06-12:
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader-prefix/src/bootloader-stamp/bootloader-build D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/bootloader.elf D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/bootloader.bin D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/bootloader.map 

cmd.exe /C "cd /D D:\ESPIDF5.0\Espressif\frameworks\esp-idf-v5.0\examples\esp-dev-kits-master\esp32-s3-lcd-ev-board\examples\lvgl_demos\build\bootloader && D:\ESPIDF5.0\Espressif\tools\cmake\3.24.0\bin\cmake.exe --build ."

[0/1] Re-running CMake...

-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/sdkconfig
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- App "bootloader" version: v5.0-dirty
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/efuse D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_app_format D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_common D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_rom D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esp_system D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/esptool_py D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/freertos D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/hal D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/log D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader/subproject/main D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader/subproject/components/micro-ecc D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/newlib D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/partition_table D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/soc D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/spi_flash D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/xtensa
-- Configuring done
CMake Warning in D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support/CMakeLists.txt:
  The object file directory

    D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/./

  has 204 characters.  The maximum full path to an object file is 250
  characters (see CMAKE_OBJECT_PATH_MAX).  Object file

    bootloader_flash/src/bootloader_flash_config_esp32s3.c.obj

  cannot be safely placed under this directory.  The build may not work
  correctly.

-- Generating done
-- Build files have been written to: D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader
[1/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/usb_otg_periph.c.obj

[2/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/lcd_periph.c.obj

[3/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj

[4/103] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj

[5/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/dport_access_common.c.obj

[6/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/gdma_periph.c.obj

[7/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/sdm_periph.c.obj

[8/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/adc_periph.c.obj

[9/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/dedic_gpio_periph.c.obj

[10/103] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xt_trax.c.obj

[11/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/gpio_periph.c.obj

[12/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/rmt_periph.c.obj

[13/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/i2s_periph.c.obj

[14/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/i2c_periph.c.obj

[15/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/interrupts.c.obj

[16/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/sdio_slave_periph.c.obj

[17/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/rtc_io_periph.c.obj

[18/103] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/temperature_sensor_periph.c.obj

[... 105 lines cut ...]
[156/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_chip_th.c.obj

[157/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/flash_mmap.c.obj

[158/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_os_func_noos.c.obj

[159/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/flash_ops.c.obj

[160/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/memspi_host_driver.c.obj

[161/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/cache_utils.c.obj

[162/1418] Building C object esp-idf/pthread/CMakeFiles/__idf_pthread.dir/pthread_local_storage.c.obj

[163/1418] Building C object esp-idf/pthread/CMakeFiles/__idf_pthread.dir/pthread_cond_var.c.obj

[164/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_os_func_app.c.obj

[165/1418] Building C object esp-idf/pthread/CMakeFiles/__idf_pthread.dir/pthread.c.obj

[166/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp_flash_spi_init.c.obj

[167/1418] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp_flash_api.c.obj

ninja: build stopped: subcommand failed.

Lzw655, 2023-06-12:
> D:/ESPIDF5.0/Espressif/frameworks/esp-idf-v5.0/examples/esp-dev-kits-master/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.di [...]

@efffx 这段提示的信息指出，工程的编译路径过长，可能会导致编译失败。请将 esp-dev-kits 仓库或工程移出 ESP-IDF，并放置在较浅的文件路径，而且建议尽量保证 ESP-IDF 仓库干净。

efffx, 2023-06-12:
已经成功烧写，感谢解答

closed 2023-06-12 by efffx