espressif/esp-iot-solution #276: usb cdc 4g module connection

mayankbytebeam, 2023-06-12:
How to set up SIM7600 modem via ESP32S3 USB Port. 

IDF Version: 5.0.0
ESP IoT Solution Version: Master Branch

Just selected `user defined` modem board in the component config -> usb host modm.

Serial Logs: 
```
I (3581) 4g_main: ====================================
I (3581) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3581) 4g_main: ====================================
I (3587) modem_board: iot_usbh_modem, version: 0.1.5
I (3592) modem_board: Force reset modem board....
I (3598) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (3607) modem_board: Resetting modem using io=13, level=0
I (3813) modem_board: Waiting for modem initialize ready
I (8813) USB_HCDC: iot_usbh_cdc, version: 0.1.3
I (8813) IOT_USBH: iot_usbh, version: 0.1.2
I (8843) IOT_USBH: USB Port=1 init succeed, fifo strategy=0
I (8843) USB_HCDC: CDC task start
I (8843) IOT_USBH: usb port start succeed
I (8846) USB_HCDC: usbh cdc driver install succeed
I (8852) USB_HCDC: Waiting CDC Device Connection
I (8857) IOT_USBH: USB Processing Start
I (8862) IOT_USBH: Waiting USB Connection
I (8866) IOT_USBH: Port power: ON Succeed
```

Ya that sit, it is hanged there. I guess it is wating for usb device.

Any help will be appreciated :)

leeebo, 2023-06-13:
@mayankbytebeam Which development board do you use? Please make sure the device is powered

mayankbytebeam, 2023-06-13:
@leeebo I used ESP32S3 and SIM7600 module in this setup.

leeebo, 2023-06-13:
@mayankbytebeam ESP32S3 DevKitC1 USB Port can not output 5V power.

mayankbytebeam, 2023-06-14:
@leeebo ok now I've powered the modem board exrternally but still some error is coming 

```
I (9702) IOT_USBH: Set Device Configuration = 1
I (9708) IOT_USBH: Set Device Configuration Done
I (9713) IOT_USBH: Pipe init succeed, addr: 81
I (9718) IOT_USBH: Pipe init succeed, addr: 01
I (9723) USB_HCDC: CDC Device Connected
I (9727) esp-modem: --------- Modem PreDefined Info ------------------
I (9734) esp-modem: Model: User Defined
I (9739) esp-modem: Modem itf: IN Addr:0x81, OUT Addr:0x01
I (9745) esp-modem: ----------------------------------------------------
I (9753) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (9762) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (9772) USB_HCDC: rx0 flush -0 = 0
I (9776) 4g_main: Modem Board Event: USB connected
I (9776) modem_board: DTE reconnect, reconnecting ...

I (10787) modem_board: reconnect after 5s...
I (11787) modem_board: reconnect after 4s...
I (12787) modem_board: reconnect after 3s...
I (13787) modem_board: reconnect after 2s...
I (14787) modem_board: reconnect after 1s...
I (14787) modem_board: Modem state STAGE_SYNC, Start
E (16797) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (16812) modem_board: Current timeouts: 1 and errors: 0
E (46812) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
W (46812) esp_modem_dce: esp_modem_dce_generic_command(22): Command:+++ response timeout
E (48817) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
E (50822) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (50837) modem_board: Current timeouts: 2 and errors: 0
W (50837) modem_board: Reset modem through reset pin........
W (50839) 4g_main: Modem Board Event: Hardware restart
I (50840) modem_board: modem_board_reset!
I (56050) 4g_main: Modem Board Event: Hardware restart done
E (58050) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
```

I've selected the modem baord as `user defined`.
can you tell me if any additional configurations is required for setting this example for SIM7600 modem?

leeebo, 2023-06-14:
@mayankbytebeam 4G modules generally have multiple interfaces, of which only one is the Modem interface, it seems `0x01 0x81` corresponding interface is not, you need to try other interfaces. _In my experience, it's often the second-to-last interface_

leeebo, 2023-06-14:
@mayankbytebeam If you don't know how to find the endpoint address of the interface, please attach your full log here, i will help

mayankbytebeam, 2023-07-31:
@leeebo sorry for the delay, we have changed the design a bit. Now we are using the N706 modem instead of SIM7600.

Serial logs,

```
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x2949
idProduct 0x7401
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
I (10134) IOT_USBH: get short config desc
I (10139) IOT_USBH: get config desc, actual_num_bytes:16
I (10145) IOT_USBH: get full config desc
I (10150) IOT_USBH: get full config desc, actual_num_bytes:244
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 236
bNumInterfaces 9
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 400mA
*** Interface Association Descriptor ***
bLength 8
bDescriptorType 11
bFirstInterface 0
bInterfaceCount 2
bFunctionClass 0xe0
bFunctionSubClass 0x1
bFunctionProtocol 0x3
iFunction 5
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe0
        bInterfaceSubClass 0x1
        bInterfaceProtocol 0x3
        iInterface 3
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 8
                bInterval 16
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xa
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 4
                *** Endpoint descriptor ***
[... 157 lines cut ...]
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 8
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xff
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 6
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x89   EP 9 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x8    EP 8 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
```

mayankbytebeam, 2023-08-01:
@leeebo Can you help to find out the correct cdc endpoint addresses,

Here are are serial logs,
```
I (25) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (25) boot: compile time 15:11:09
I (25) boot: chip revision: v0.1
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (104) boot_comm: chip revision: 1, min. application chip revision: 0
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2efech (192492) map
I (149) esp_image: segment 1: paddr=0003f014 vaddr=3fc9a400 size=01004h (  4100) load
I (150) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a77b0h (686000) map
I (259) esp_image: segment 3: paddr=000e77d8 vaddr=3fc9b404 size=03654h ( 13908) load
I (261) esp_image: segment 4: paddr=000eae34 vaddr=40374000 size=163a0h ( 91040) load
I (282) esp_image: segment 5: paddr=001011dc vaddr=50000000 size=00010h (    16) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (303) cpu_start: Pro cpu up.
I (303) cpu_start: Starting app cpu, entry point is 0x403757c8
0x403757c8: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (317) cpu_start: Pro cpu start user code
I (317) cpu_start: cpu freq: 240000000 Hz
I (317) cpu_start: Application information:
I (320) cpu_start: Project name:     usb_cdc_4g_module
I (326) cpu_start: App version:      7f6be2fb-dirty
I (332) cpu_start: Compile time:     Aug  1 2023 15:58:40
I (338) cpu_start: ELF file SHA256:  4cc902f5cf4cede6...
I (344) cpu_start: ESP-IDF:          v5.0-dirty
I (349) heap_init: Initializing. RAM available for dynamic allocation:
I (356) heap_init: At 3FCA26C8 len 00047048 (284 KiB): D/IRAM
I (363) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (369) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (375) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (382) spi_flash: detected chip: generic
I (386) spi_flash: flash io: qio
W (390) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (404) sleep: Configure to isolate all GPIO pins in sleep state
I (410) sleep: Enable automatic switching of GPIO sleep configuration
I (417) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (433) led_indicator: LED Indicator Version: 0.3.0
I (439) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (448) led_indicator: blink_lists is null, use default blink list
W (455) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (472) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default
I (485) led_indicator: LED Indicator Version: 0.3.0
I (491) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (500) led_indicator: blink_lists is null, use default blink list
W (507) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (524) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x11, is_active_level_high:1, blink_lists:default
I (537) led_indicator: LED Indicator Version: 0.3.0
I (543) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (552) led_indicator: blink_lists is null, use default blink list
W (559) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (576) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x10, is_active_level_high:1, blink_lists:default
I (3604) 4g_main: ====================================
[... 307 lines cut ...]
I (11665) modem_board: reconnect after 5s...
I (12665) modem_board: reconnect after 4s...
I (13665) modem_board: reconnect after 3s...
I (14665) modem_board: reconnect after 2s...
I (15665) modem_board: reconnect after 1s...
I (15665) modem_board: Modem state STAGE_SYNC, Start
E (17675) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (17690) modem_board: Current timeouts: 1 and errors: 0
E (47690) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
W (47690) esp_modem_dce: esp_modem_dce_generic_command(22): Command:+++ response timeout
E (49695) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
[... cut ...]
I (51715) modem_board: Current timeouts: 2 and errors: 0
W (51715) modem_board: Reset modem through reset pin........
W (51717) 4g_main: Modem Board Event: Hardware restart
I (51718) modem_board: modem_board_reset!
I (51728) 4g_main: Modem Board Event: Hardware restart done
E (53728) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (53743) modem_board: Current timeouts: 3 and errors: 0
I (53743) modem_board: modem_board_power_down!
```

leeebo, 2023-08-02:
@mayankbytebeam From the module's descriptor, which has eight CDC-like interfaces, please first test with the penultimate one (`IN 0x88, OUT 0x07`). If that does not work, please try the eight interfaces one by one

[image]

lovepreethitech, 2023-08-04:
Hi,

I m facing the same issue.
Hardware - ESP32 S2 WROOM 1
Modem - User Defined ( NEOWAY N58 )

@leeebo  which interface class's endpoints should i use from below -

sudo lsusb -v -d 2949:7401
[sudo] password for lovepreet: 

Bus 001 Device 008: ID 2949:7401 NEOWAY NEOWAY-N58
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2949 
  idProduct          0x7401 
  bcdDevice            0.00
  iManufacturer           1 NEOWAY
  iProduct                2 NEOWAY-N58
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00ec
    bNumInterfaces          9
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              400mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       3 RNDIS
      iFunction               5 RNDIS
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      3 RNDIS
      iInterface              3 RNDIS Communications Control
      ** UNRECOGNIZED:  05 24 00 10 01
      ** UNRECOGNIZED:  05 24 01 00 01
      ** UNRECOGNIZED:  04 24 02 00
      ** UNRECOGNIZED:  05 24 06 00 01
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
[... 628 lines cut ...]
E (46831) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
W (46831) esp_modem_dce: esp_modem_dce_generic_command(22): Command:+++ response timeout
E (48836) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
E (50841) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (50856) modem_board: Current timeouts: 2 and errors: 0
W (50856) modem_board: Reset modem through reset pin........
W (50858) 4g_main: Modem Board Event: Hardware restart
I (50864) modem_board: modem_board_reset!
I (56069) 4g_main: Modem Board Event: Hardware restart done
E (58069) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (58084) modem_board: Current timeouts: 3 and errors: 0
I (58084) modem_board: modem_board_power_down!
E (68585) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
E (70590) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (70605) modem_board: Current timeouts: 4 and errors: 0
I (70605) modem_board: modem_board_power_down!

-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*--*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*--*-**-

@mayankbytebeam - Any Solution You Found ?

mayankbytebeam, 2023-08-04:
@leeebo Thanks for your input. 0x85 and 0x04 worked for me.  
@lovepreethitech  Try each end point one by one.

lovepreethitech, 2023-08-04:
Hi @mayankbytebeam , 

I guess you are using NEOWAY module.
I am using NEOWAY N58.

I have tried all the EPs
Majority interfaces gives - AT command Timeout
and 2 interfaces gives - AT&W failed error
----------------------------------------------------------------------

EP IN | EP OUT | ERROR
81 | 1 | AT COMMAND – TIMEOUT
83 | 2 | AT&W COMMAND FAILED – ERROR
84 | 3 | AT COMMAND – TIMEOUT
85 | 4 | AT&W COMMAND FAILED – ERROR
86 | 5 | AT COMMAND – TIMEOUT
87 | 6 | AT COMMAND – TIMEOUT
88 | 7 | AT COMMAND – TIMEOUT
89 | 8 | AT COMMAND – TIMEOUT

Have you faced anything like this while working ?

leeebo, 2023-08-04:
@lovepreethitech Hi please link `Reset Control`  IO to your modem `Reset` Pin, try it again

[image]

leeebo, 2023-08-11:
@lovepreethitech Hi, please set the ESP_LOG level to verbose to find the detailed Interaction process

mayankbytebeam, 2023-08-11:
@lovepreethitech Can you try commenting the command that is stucking?

lovepreethitech, 2023-08-11:
@mayankbytebeam In which file you comment that ?

mayankbytebeam, 2023-08-11:
@lovepreethitech 

It's in `iot_usbh_modem` component, 

https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/src/esp_modem_dce_common_commands.c#L268

lovepreethitech, 2023-08-14:
hi @leeebo @mayankbytebeam 

After commenting AT&W command related lines in 3 files its no more giving AT&W error but core panic ed at last and reboot.

[image]
[image]
[image]

------------------------------------------------------------------------------------------------------------------------------------------------------------------------

express8081999, 2023-09-01:
> @mayankbytebeam Which development board do you use? Please make sure the device is powered

esp32s3 （合宙 luatos.com） and air780e 4G

weekroom, 2023-09-19:
Hello, I used Lexin's esp32s3 DevKitC1 development board and Heju's AIR780E for USB host communication, but there were errors and timeout issues related to the ESP32-S3 modem module. What is the reason for this? I cannot find a solution。Here is my output log
I (43) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (43) boot: compile time 12:18:19
I (43) boot: chip revision: v0.1
I (45) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (52) qio_mode: Enabling default flash chip QIO
I (58) boot.esp32s3: Boot SPI Speed : 80MHz
I (63) boot.esp32s3: SPI Mode       : QIO
I (67) boot.esp32s3: SPI Flash Size : 8MB
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (96) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (103) boot:  2 factory          factory app      00 00 00010000 00100000
I (111) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (118) boot: End of partition table
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1b664h (112228) map
I (155) esp_image: segment 1: paddr=0002b68c vaddr=3fc99d10 size=0443ch ( 17468) load
I (159) esp_image: segment 2: paddr=0002fad0 vaddr=40374000 size=00548h (  1352) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=7fbb4h (523188) map
I (249) esp_image: segment 4: paddr=000afbdc vaddr=40374548 size=157bch ( 87996) load
I (274) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (286) cpu_start: Pro cpu up.
I (286) cpu_start: Starting app cpu, entry point is 0x403756b0
0x403756b0: call_start_cpu1 at C:/Users/huang/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (300) cpu_start: Pro cpu start user code
I (300) cpu_start: cpu freq: 240000000
I (300) cpu_start: Application information:
I (303) cpu_start: Project name:     usb_cdc_4g_module
I (309) cpu_start: App version:      1
I (313) cpu_start: Compile time:     Sep 19 2023 17:51:12
I (319) cpu_start: ELF file SHA256:  d01d46f315f7c67d...
I (325) cpu_start: ESP-IDF:          v4.4.4-dirty
I (331) heap_init: Initializing. RAM available for dynamic allocation:
I (338) heap_init: At 3FCA15A8 len 00048168 (288 KiB): D/IRAM
I (344) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (351) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (357) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (364) spi_flash: detected chip: generic
I (368) spi_flash: flash io: qio
I (372) sleep: Configure to isolate all GPIO pins in sleep state
I (379) sleep: Enable automatic switching of GPIO sleep configuration
I (386) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (401) led_indicator: LED Indicator Version: 0.3.0
I (407) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (416) led_indicator: blink_lists is null, use default blink list
W (423) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (440) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default
I (453) led_indicator: LED Indicator Version: 0.3.0
I (459) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (468) led_indicator: blink_lists is null, use default blink list
W (475) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (492) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x11, is_active_level_high:1, blink_lists:default
I (505) led_indicator: LED Indicator Version: 0.3.0
I (511) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (520) led_indicator: blink_lists is null, use default blink list
W (527) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (544) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x10, is_active_level_high:1, blink_lists:default
I (3576) 4g_main: ====================================
I (3576) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3576) 4g_main: ====================================
I (3582) modem_board: iot_usbh_modem, version: 0.1.6
I (3587) modem_board: Force reset modem board....
I (3593) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[... 270 lines cut ...]
I (95597) modem_board: Modem state STAGE_SYNC, Start
E (97613) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (97628) modem_board: Current timeouts: 1 and errors: 0
E (127628) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
W (127628) esp_modem_dce: esp_modem_dce_generic_command(22): Command:+++ response timeout
E (129633) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
[... cut ...]
I (138881) modem_board: Current timeouts: 3 and errors: 0
I (138881) modem_board: modem_board_power_down!
E (149383) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutdem_dce: esp_modem_dce_generic_command(22): Command:AT
E (151388) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutdem_dce: esp_modem_dce_generic_command(22): Command:AT
I (151403) modem_board: Current timeouts: 4 and errors: 0
I (151403) modem_board: modem_board_power_down!

Y1hsiaochunnn, 2023-09-19:
Maybe some of the environment dependencies weren't introduced correctly, I've solved my problem and it looks similar to yours

weekroom, 2023-09-20:
@Y1hsiaochunnn  I can use the code from the Gitee library normally and have added the AIR780E option, which is convenient for me to apply to projects。https://gitee.com/dog_who_loves_cat_mint/esp32_-cdc_4-g

leeebo, 2023-10-09:
@Y1hsiaochunnn 
```
I (15163) modem_board: Modem state STAGE_CHECK_REGIST, Start
W (15170) moGuru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.
```

It's a know bug, log in #295. will fix soon