espressif/esp-iot-solution #277: ILLI9488 display is not working.

nikhil-robinson, 2023-06-17:
## Environment

- Module or chip used: esp32s3 wroom-1 N8R8
- IDF version (run ``git describe --tags`` to find it): v4.4.4-303-g90d6e45d9f
- Build System:        cmake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):8.4.0
- Operating System:     Linux
- Power Supply:         [USB]

## Problem Description

Trying to test illi9488 display  with the example screen not able to get any display out.  my connections are correct was able to test the wiring with different display ili9431 with same example and it works. The display also works tested with an Arduino board.

### Expected Behavior

generate mandelbrot

### Actual Behavior

Gray screen

### Steps to repropduce

1. Compile the example from esp-iot-solution/examples/screen for esp32s3
2. 

### Code to reproduce this issue

```cpp

void app_main(void)
{
    esp_err_t ret = ESP_OK;
    spi_config_t bus2_conf = {
        .miso_io_num = -1,
        .mosi_io_num = 40,
        .sclk_io_num = 39,
        .max_transfer_sz = 8 * 480 * 320 + 10,
    };
    spi_bus_handle_t spi_bus = spi_bus_create(SPI2_HOST, &bus2_conf);

    scr_interface_spi_config_t spi_lcd_cfg = {
        .spi_bus = spi_bus,
        .pin_num_cs = 41,
        .pin_num_dc = 42,
        .clk_freq = 40000000,
        .swap_data = true,
    };

    scr_interface_driver_t *iface_drv;
    scr_interface_create(SCREEN_IFACE_SPI, &spi_lcd_cfg, &iface_drv);
    ret = scr_find_driver(SCREEN_CONTROLLER_ILI9488, &g_lcd);
    if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen find failed");
    }

    scr_controller_config_t lcd_cfg = {
        .interface_drv = iface_drv,
        .pin_num_rst = 48,
        .pin_num_bckl = -1,
        .rst_active_level = 0,
        .bckl_active_level = 1,
        .offset_hor = 0,
        .offset_ver = 0,
        .width = 320,
        .height = 480,
        .rotate = 0,
    };
    ret = g_lcd.init(&lcd_cfg);

    if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen initialize failed");
    }

    g_lcd.get_info(&g_lcd_info);
    ESP_LOGI(TAG, "Screen name:%s | width:%d | height:%d", g_lcd_info.name, g_lcd_info.width, g_lcd_info.height);

    screen_clear(&g_lcd, COLOR_ESP_BKGD);
    vTaskDelay(pdMS_TO_TICKS(500));

    /**  Run test */
    lcd_bitmap_test(&g_lcd);
    vTaskDelay(pdMS_TO_TICKS(2000));
    lcd_speed_test(&g_lcd);
    vTaskDelay(pdMS_TO_TICKS(2000));

    init_CLUT(g_color_table); /** Initialize color look up table */

    /** Define an interesting point on the complex plane */
    float real = -0.68481 + (g_lcd_info.width / 2 / (float)MAX_ZOOM);
    float img = 0.380584 + (g_lcd_info.height / 2 / (float)MAX_ZOOM);

    uint16_t *pixels = heap_caps_malloc(g_lcd_info.width * sizeof(uint16_t), MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL);
    if (NULL == pixels) {
        ESP_LOGE(TAG, "Memory for bitmap is not enough");
        return;
    }

    float zoom;

    while (1) {
        for (zoom = 50; zoom <= MAX_ZOOM; zoom *= 1.1f) {
            int32_t off_x = (real) * zoom;
            int32_t off_y = (img) * zoom;
            generate_mandelbrot(&g_lcd, g_lcd_info.width, g_lcd_info.height, g_lcd_info.width / 2 - off_x,  g_lcd_info.height / 2 - off_y, zoom, pixels);
            vTaskDelay(1); /** Delay one tick for feed task watchdog */
        }
        for (; zoom > 50; zoom /= 1.1f) {
            int32_t off_x = (real) * zoom;
            int32_t off_y = (img) * zoom;
            generate_mandelbrot(&g_lcd, g_lcd_info.width, g_lcd_info.height, g_lcd_info.width / 2 - off_x,  g_lcd_info.height / 2 - off_y, zoom, pixels);
            vTaskDelay(1);
        }
    }
}

```

## Debug Logs

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1640
load:0x403c9700,len:0xbcc
load:0x403cc700,len:0x301c
SHA-256 comparison failed:
Calculated: 5a8ac49ac8d65ca91207c1bcd70fe06326891da92eec6456cac17ac2579ee54f
Expected: 00b7fc3a65d1633d183948d7f38c742fd2083a4438f9b58880ee4538d45f7540
Attempting to boot anyway...
entry 0x403c9954
I (43) boot: ESP-IDF v4.4.4-303-g90d6e45d9f-dirty 2nd stage bootloader
I (43) boot: compile time 15:59:33
I (44) boot: chip revision: v0.1
I (47) boot.esp32s3: Boot SPI Speed : 80MHz
I (52) boot.esp32s3: SPI Mode       : DIO
I (56) boot.esp32s3: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=1acd8h (109784) map
I (132) esp_image: segment 1: paddr=0002ad00 vaddr=3fc93f50 size=0224ch (  8780) load
I (135) esp_image: segment 2: paddr=0002cf54 vaddr=40374000 size=030c4h ( 12484) load
I (141) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=20400h (132096) map
I (170) esp_image: segment 4: paddr=00050428 vaddr=403770c4 size=0ce8ch ( 52876) load
I (188) boot: Loaded app from partition at offset 0x10000
I (189) boot: Disabling RNG early entropy source...
I (200) cpu_start: Pro cpu up.
I (200) cpu_start: Starting app cpu, entry point is 0x403752f0
0x403752f0: call_start_cpu1 at /home/nikhi/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (214) cpu_start: Pro cpu start user code
I (214) cpu_start: cpu freq: 160000000
I (215) cpu_start: Application information:
I (217) cpu_start: Project name:     screen_demo
I (222) cpu_start: App version:      dcb2082-dirty
I (228) cpu_start: Compile time:     Jun 17 2023 15:59:24
I (234) cpu_start: ELF file SHA256:  3458caa6a34c19c5...
I (240) cpu_start: ESP-IDF:          v4.4.4-303-g90d6e45d9f-dirty
I (246) cpu_start: Min chip rev:     v0.0
I (251) cpu_start: Max chip rev:     v0.99 
I (256) cpu_start: Chip rev:         v0.1
I (261) heap_init: Initializing. RAM available for dynamic allocation:
I (268) heap_init: At 3FC96DC8 len 00052948 (330 KiB): D/IRAM
I (274) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (281) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (287) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (294) spi_flash: detected chip: generic
I (298) spi_flash: flash io: dio
I (303) sleep: Configure to isolate all GPIO pins in sleep state
I (309) sleep: Enable automatic switching of GPIO sleep configuration
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) spi_bus: SPI2 bus created
I (337) spi_bus: SPI2 bus device added, CS=41 Mode=0 Speed=40000000
I (764) ILI9488: MADCTL=0x8
I (764) screen example: Screen name:ILI9488 | width:320 | height:480
I (8840) screen example: -------ILI9488 Test Result------
I (8840) screen example: @resolution 240x240          [time per frame=13.88MS, fps=72.05, speed=7.92MB/s]
I (8846) screen example: @resolution 320x480 infer to [time per frame=37.01MS, fps=27.02]
I (8855) screen example: -------------------------------------
I (10861) screen example: zoom = 50
I (10861) screen example: start(-3.820000, -4.340000)
I (11420) screen example: end  (2.560000, 5.240000)
I (11422) screen example: zoom = 55
I (11422) screen example: start(-3.527273, -3.890909)
I (12009) screen example: end  (2.272727, 4.818182)
I (12011) screen example: zoom = 60
I (12011) screen example: start(-3.283333, -3.533333)
I (12623) screen example: end  (2.033333, 4.450000)
I (12625) screen example: zoom = 66
I (12625) screen example: start(-3.045455, -3.166667)
I (13270) screen example: end  (1.787879, 4.090909)
I (13272) screen example: zoom = 73

I (13272) screen example: start(-2.808219, -2.821918)
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/11778646/sdkconfig.txt)

Lzw655, 2023-06-19:
Hi @nikhil-robinson, the "Display" component is not maintained anymore. We strongly recommend you use "esp_lcd" component to develop the LCD. 

Please refer to the [example](https://github.com/espressif/esp-idf/tree/release/v5.0/examples/peripherals/lcd/spi_lcd_touch) and [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html).

Lzw655, 2023-06-28:
@nikhil-robinson If your problem has been resolved, please close this issue. Thank you!

closed 2023-07-01 by nikhil-robinson