espressif/esp-iot-solution #278: Get device descriptor success but no image or frame from USB camera.

MrNash97, 2023-06-18:
## Environment

- Development Kit:      [[YD-ESP32-S3](https://github.com/vcc-gnd/YD-ESP32-S3)]
- Module or chip used:  [ESP32-S3-WROOM-1]
- IDF version (run ``git describe --tags`` to find it): v5.0.2
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2022r1) 11.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
I'm using the usb_camera_mic_spk sample project to test my Hikvision DS-U02 camera. I have built and flashed the project and able to get the device config. However, I cannot see any image/frame on the wifi camera IP web view. The camera itself is turned ON and able to enter Active mode indicated by the Blue LED active on the webcam.

### Expected Behavior
Can get a still image or stream frames from the camera displayed to the web view.

### Actual Behavior
Cannot see any image/frame on the wifi camera IP web view.

### Steps to reproduce
1. Plug the camera into the ESP-S3 board and power the board via USB
2. Build and flash project via VS Code (Ctrl+E+D)
3. Open the camera IP address acquired from the terminal
4. Clik Get Still & Start Stream

[image]

### Code to reproduce this issue
<details>
  <summary>Show/Hide</summary>

```cpp
/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>
#include <string.h>
#include <assert.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "freertos/task.h"
#include "esp_err.h"
#include "esp_log.h"
#include "usb_stream.h"

static const char *TAG = "uvc_mic_spk_demo";
/****************** configure the example working mode *******************************/
#define ENABLE_UVC_CAMERA_FUNCTION        1        /* enable uvc function */
#define ENABLE_UAC_MIC_SPK_FUNCTION       0        /* enable uac mic+spk function */
#if (ENABLE_UVC_CAMERA_FUNCTION)
#define ENABLE_UVC_FRAME_RESOLUTION_ANY   1        /* Using any resolution found from the camera */
#define ENABLE_UVC_WIFI_XFER              1        /* transfer uvc frame to wifi http */
#endif
#if (ENABLE_UAC_MIC_SPK_FUNCTION)
#define ENABLE_UAC_MIC_SPK_LOOPBACK       0        /* transfer mic data to speaker */
static uint32_t s_mic_samples_frequence = 0;
static uint32_t s_mic_ch_num = 0;
static uint32_t s_mic_bit_resolution = 0;
static uint32_t s_spk_samples_frequence = 0;
static uint32_t s_spk_ch_num = 0;
static uint32_t s_spk_bit_resolution = 0;
#endif

#define BIT0_FRAME_START     (0x01 << 0)
#define BIT1_NEW_FRAME_START (0x01 << 1)
#define BIT2_NEW_FRAME_END   (0x01 << 2)
[... 726 lines cut ...]
I (1544) uvc_mic_spk_demo:      frame[4] = 640x480
I (1545) uvc_mic_spk_demo:      frame[5] = 640x480
I (1545) uvc_mic_spk_demo: Device connected
I (1545) USB_STREAM: Creating UVC pipe: ifc=1-2, ep=0x85, mps=512
I (1546) USB_STREAM: Probe Format(2) MJPEG, Frame(1) 640*360, interval(400000)
I (1546) USB_STREAM: Probe payload size = 512
bFormatIndex: 2
bFrameIndex: 1
dwFrameInterval: 400000
dwMaxPayloadTransferSize: 3060
bInterfaceNumber: 0
I (1565) USB_STREAM: Sample processing task created
I (1565) USB_STREAM: Sample processing task start
I (1566) USB_STREAM: Set Device Interface = 1, Alt = 2
I (1576) USB_STREAM: Set Device Interface Done
I (1576) USB_STREAM: USB Device Connected
I (2734) esp_netif_handlers: sta ip: 192.168.1.101, mask: 255.255.255.0, gw: 192.168.1.1
I (2734) camera wifi: got ip:192.168.1.101

```
</details>

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/11780893/sdkconfig.txt)

leeebo, 2023-06-19:
@MrNash97 From the camera descriptors and parsed result, the host has already probed `format 2 (MJPEG) and frame 1(640*360)`  successfully, it should work 🤔 .

Would you please change to verbose mode to get more logs?

MrNash97, 2023-06-20:
Yeah, I hope that my camera can work with this example.

Sure, here is the verbose log. I removed the long-repetitive log otherwise it will be too long.

<details>
  <summary>Show/Hide Verbose Log</summary>

  ```cpp
I (25) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (25) boot: compile time 22:35:00
I (25) boot: chip revision: v0.1
I (25) boot.esp32s3: Boot SPI Speed : 80MHz
I (25) boot.esp32s3: SPI Mode       : DIO  
I (25) boot.esp32s3: SPI Flash Size : 8MB
I (26) boot: Enabling RNG early entropy source...
I (26) boot: Partition Table:
I (26) boot: ## Label            Usage          Type ST Offset   Length
I (26) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (27) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (27) boot:  2 factory          factory app      00 00 00010000 00100000
I (28) boot: End of partition table
I (28) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=2cd24h (183588) map
I (61) esp_image: segment 1: paddr=0003cd4c vaddr=3fc98d00 size=032cch ( 13004) load
I (65) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=8dbd8h (580568) map
I (169) esp_image: segment 3: paddr=000cdc00 vaddr=3fc9bfcc size=00d70h (  3440) load
I (170) esp_image: segment 4: paddr=000ce978 vaddr=40374000 size=14cf0h ( 85232) load
I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
D (198) flash HPM: HPM with dummy, status is 3
I (199) cpu_start: Pro cpu up.
I (199) cpu_start: Starting app cpu, entry point is 0x40375604
0x40375604: call_start_cpu1 at C:/Users/FXMEDIA/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: Dc o(re 11 9a9c)c esesf uesrer:o r Iintr clrBL& e_a mask is:EG0xs us
d 2 bits starting with 0 bit
D (200) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 13 bit
D (200) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (200) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (201) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 11 bit
D (201) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (202) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (202) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (203) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (203) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (203) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (204) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (204) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (205) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (205) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (214) clk: RTC_SLOW_CLK calibration value: 4012864
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
I (215) cpu_start: Pro cpu start user code
I (215) cpu_start: cpu freq: 240000000 Hz
I (215) cpu_start: Application information:
I (215) cpu_start: Project name:     usb_camera_mic_spk
I (215) cpu_start: App version:      1
I (216) cpu_start: Compile time:     Jun 18 2023 14:32:13
I (216) cpu_start: ELF file SHA256:  bf5135e4f7ef584b...
I (216) cpu_start: ESP-IDF:          v5.0.2-dirty
I (217) cpu_start: Min chip rev:     v0.0
I (217) cpu_start: Max chip rev:     v0.99 
I (217) cpu_start: Chip rev:         v0.1
V (217) memory_layout: reserved range is 0x3c0bcd14 - 0x3c0bcd34
D (218) memory_layout: Checking 5 reserved memory ranges:
D (218) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc98d00
D (218) memory_layout: Reserved memory range 0x3fc98d00 - 0x3fca08f8
D (219) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
[... 1504 lines cut ...]
V (22142) USB_STREAM: process payload=0, len = 0
V (22142) USB_STREAM: process payload=1, len = 0
V (22142) USB_STREAM: process payload=2, len = 0
V (22143) USB_STREAM: process payload=3, len = 0
V (22146) USB_STREAM: Pipe(UVC): XFER_DONE
V (22146) USB_STREAM: process payload=0, len = 0
V (22146) USB_STREAM: process payload=1, len = 0
V (22146) USB_STREAM: process payload=2, len = 0
V (22147) USB_STREAM: process payload=3, len = 0
V (22150) USB_STREAM: Pipe(UVC): XFER_DONE
V (22150) USB_STREAM: process payload=0, len = 0
V (22150) USB_STREAM: process payload=1, len = 0
V (22150) USB_STREAM: process payload=2, len = 0
V (22151) USB_STREAM: process payload=3, len = 0
V (22154) USB_STREAM: Pipe(UVC): XFER_DONE
V (22154) USB_STREAM: process payload=0, len = 0
V (22154) USB_STREAM: process payload=1, len = 0
V (22154) USB_STREAM: process payload=2, len = 0
V (22155) USB_STREAM: process payload=3, len = 0
V (22158) USB_STREAM: Pipe(UVC): XFER_DONE

// Disconnect the board
  ```

</details>

leeebo, 2023-06-21:
```
V (22154) USB_STREAM: Pipe(UVC): XFER_DONE
V (22154) USB_STREAM: process payload=0, len = 0
V (22154) USB_STREAM: process payload=1, len = 0
V (22154) USB_STREAM: process payload=2, len = 0
V (22155) USB_STREAM: process payload=3, len = 0
```

@MrNash97 The Host (esp32s3) Initiate an IN transfer to the Camera, the Camera feedback ACK but did not give us any data. The camera sensors don't seem ready yet. You can wait longer to take a look.

And I found you have the following resolution, you can choose another one through [config](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/host/usb_camera_mic_spk/main/main.c#L238) `uvc_config.frame_width = 640` `uvc_config.frame_height= 480` and give it a try, 

```
I (1589) uvc_mic_spk_demo:      frame[0] = 640x360
I (1589) uvc_mic_spk_demo:      frame[1] = 1280x720
I (1589) uvc_mic_spk_demo:      frame[2] = 1280x960
I (1590) uvc_mic_spk_demo:      frame[3] = 1920x1080
I (1590) uvc_mic_spk_demo:      frame[4] = 640x480
I (1590) uvc_mic_spk_demo:      frame[5] = 640x480
```

MrNash97, 2023-06-25:
I have followed your suggestion and waited longer (30+ minutes), but it still returns the same messages. I tried to change the resolution to 640 x 480, but nothing changed.
Here is my recording when I try to get a still image or start streaming from the device URL.

https://github.com/espressif/esp-iot-solution/assets/48994203/b3b422b8-98c7-43b2-8540-f6709c609e56

I also noticed there are two `640x480` resolutions on my log. Is this normal?

[image]

leeebo, 2023-06-27:
@MrNash97 `two 640x480 resolutions` is also OK, some camera descriptors do this in order to ensure compatibility.

Your current configuration is perfect to me, if it does not work, we can only assume that the camera can not work in `MPS=512` (interface 1-2), or not work in Full speed mode.

MrNash97, 2023-07-02:
I did some more testing, but still unable to get the video stream working while the UAC is working without any issues. I also tried a similar example of the [UVC project](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc) from the espressif-idf GitHub. The result is similar to this example, cannot get an image from the camera. When I try using `640x480 @25 fps` it returns an `Unknown Error` message. When I try to reduce the fps to `15 fps` the error is gone, but I am getting this `Negotiating streaming format failed, trying again...` message on an endless loop.

I noticed that the LED status on my camera turned red when I ran that example, instead of blue like when I ran this example project. I think that is not the main issue. I guess it is more like this project able to get the audio stream running hence the LED turning to blue, while the other example only focused on video or images.

After all the testing, I assume that this webcam is not fully compatible with this project since my webcam does not support any `360x240` resolution or `640x480 @15 fps`. If you do not have any more suggestions, I think you can close this issue.

leeebo, 2023-07-02:
@MrNash97 Seems your webcam can not work as normal in fullspeed mode (12Mbps) or in the MPS=512 interface.

So it can not be supported in ESP32Sx platform ...

MrNash97, 2023-07-02:
Okay then. I will close this issue.

closed 2023-07-02 by MrNash97