espressif/esp-iot-solution #280: 使用usb_cdc_4g_module例子，可以识别设备，但是发送AT超时

Szeroy, 2023-06-28:
使用的4G模块是合宙的air780e，看日志是识别出来设备了，但是发送AT显示超时，例子里面没有air780e，所以选了air724ug，不过都是合宙的应该是一样的吧。
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17b8
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x2fec
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (25) boot: compile time 16:34:53
I (25) boot: chip revision: 0
I (26) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : QIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1f088h (127112) map
I (124) esp_image: segment 1: paddr=0002f0b0 vaddr=3fc995c0 size=00f68h (  3944) load
I (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8e170h (582000) map
I (218) esp_image: segment 3: paddr=000be198 vaddr=3fc9a528 size=0341ch ( 13340) load
I (220) esp_image: segment 4: paddr=000c15bc vaddr=40374000 size=155b4h ( 87476) load
I (240) esp_image: segment 5: paddr=000d6b78 vaddr=50000000 size=00010h (    16) load
I (249) boot: Loaded app from partition at offset 0x10000
I (249) boot: Disabling RNG early entropy source...
I (261) cpu_start: Pro cpu up.
I (261) cpu_start: Starting app cpu, entry point is 0x40375620
0x40375620: call_start_cpu1 at D:/esp-idf/esp-idf-4.4v/frameworks/esp-idf-v4.4.3/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (275) cpu_start: Pro cpu start user code
I (275) cpu_start: cpu freq: 240000000
I (275) cpu_start: Application information:
I (277) cpu_start: Project name:     usb_cdc_4g_module
I (283) cpu_start: App version:      34a150e7-dirty
I (289) cpu_start: Compile time:     Jun 28 2023 17:31:53
I (295) cpu_start: ELF file SHA256:  94e85c82921e5b33...
I (301) cpu_start: ESP-IDF:          v4.4.3
I (306) heap_init: Initializing. RAM available for dynamic allocation:
I (313) heap_init: At 3FCA11A8 len 00048568 (289 KiB): D/IRAM
I (319) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (326) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (332) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (339) spi_flash: detected chip: gd
I (343) spi_flash: flash io: qio
W (347) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (360) sleep: Configure to isolate all GPIO pins in sleep state
I (367) sleep: Enable automatic switching of GPIO sleep configuration
I (374) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (389) led_indicator: LED Indicator Version: 0.4.0
I (395) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (404) led_indicator: blink_lists is null, use default blink list
W (411) led_indicator: D:/esp-idf/esp-iot-solution/components/led/led_indicator/src/led_indicator.c:294 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (429) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default
I (442) led_indicator: LED Indicator Version: 0.4.0
I (448) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (457) led_indicator: blink_lists is null, use default blink list
W (464) led_indicator: D:/esp-idf/esp-iot-solution/components/led/led_indicator/src/led_indicator.c:294 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (482) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x11, is_active_level_high:1, blink_lists:default
I (495) led_indicator: LED Indicator Version: 0.4.0
I (500) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[... 186 lines cut ...]
I (9597) USB_HCDC: CDC Device Connected
I (9601) esp-modem: --------- Modem PreDefined Info ------------------
I (9608) esp-modem: Model: AIR724UG-NFM
I (9613) esp-modem: Modem itf: IN Addr:0x84, OUT Addr:0x03
I (9619) esp-modem: ----------------------------------------------------
I (9626) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (9636) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (9646) USB_HCDC: rx0 flush -0 = 0
W (9649) USB_HCDC: Itf:0 ep: 0x84 PIPE_EVENT_ERROR_OVERFLOW
I (9656) modem_board: DTE reconnect, reconnecting ...

I (9656) 4g_main: Modem Board Event: USB connected
I (10661) modem_board: reconnect after 5s...
I (11661) modem_board: reconnect after 4s...
I (12661) modem_board: reconnect after 3s...
I (13661) modem_board: reconnect after 2s...
I (14661) modem_board: reconnect after 1s...
I (14661) modem_board: Modem state STAGE_SYNC, Start
E (16671) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT
I (16686) modem_board: Current timeouts: 1 and errors: 0
E (46686) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
W (46686) esp_modem_dce: esp_modem_dce_generic_command(22): Command:+++ response timeout
E (48691) esp-modem-dte: esp_modem_dte_send_cmd(266): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(22): Command:AT`

leeebo, 2023-06-29:
@Szeroy 即使是来自同一个供应商，不同型号用的芯片平台、SDK 都有区别，导致 AT 接口是不同的。你这里一共有三个接口

1. OUT 0x1, IN 0x82
2. OUT 0x2, IN 0x84
3. OUT 0x3, IN 0x86

你 menuconfig 选择 Customer 模式，手动配置接口，一对儿 一对儿去测试 。

Szeroy, 2023-06-29:
成功了，一直是我结点错了，我一直认为EP 1 OUT和EP 1 IN是一对，现在用了01和82就通上了，感谢

closed 2023-06-29 by Szeroy

Sid9017, 2025-03-01:
@Szeroy 请教一下Air780e怎么手动配置的接口，我这边启动的时候in_ep_desc->bEndpointAddress 报空指针错误

Szeroy, 2025-03-02:
> @Szeroy 请教一下Air780e怎么手动配置的接口，是在menuconfig中的Component config → IoT USB Host Modem → USB CDC interface config 么

好像是的，有一个配置是选择模组型号，没有的话就用用户自定义

Sid9017, 2025-03-03:
@Szeroy 我一个个Description试过来 选0x03好像就好了 thx all the same 👍