espressif/esp-iot-solution #281: Espressif Esp TinyUsb component doesn't match on dependencies.lock

yosafatkesuma, 2023-06-30:
## Environment

- Development Kit:      [ESP32-S3-USB-OTG]
- IDF version : v4.4.5-104-g8b94183c9c
- Build System:         [CMake]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating System:     [Linux/Ubuntu]
- Power Supply:         [USB]

## Problem Description

When i try to set-target to esp32s3, it's got error something like this 
CMake Error at $HOME/esp/esp-idf/tools/cmake/build.cmake:463 (message):
  ERROR: The hash sum of the component "espressif/esp_tinyusb(1.1.0)" does
  not match the one recorded in your dependencies.lock file.  This could be
  due to a potential spoofing of the download server, or your lock file may
  have become corrupted.  Please review the lock file and verify the download
  server's authenticity to ensure the component's security and integrity.

Call Stack (most recent call first):
  $HOME/esp/esp-idf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "$HOME/esp/esp-iot-solution/examples/usb/device/usb_msc_wireless_disk/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1 

### Expected Behavior
I don't know what the expected result, 'cause this is my first time

### Actual Behavior
I don't know what the expected result, 'cause this is my first time

### Steps to repropduce
I follow the step from here https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/device/usb_msc_wireless_disk

## Return Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

[CMakeOutput.log](https://github.com/espressif/esp-iot-solution/files/11914669/CMakeOutput.log)

leeebo, 2023-06-30:
Hi @yosafatkesuma the `add_usb_solutions` branch is achieved now, please try USB examples on the master branch https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_msc_wireless_disk

leeebo, 2023-06-30:
@yosafatkesuma For the error `ERROR: The hash sum of the component "espressif/esp_tinyusb(1.1.0)" does
not match the one recorded in your dependencies.lock file`, delete `dependencies.lock` file under your project then clean the build forlder may help.

yosafatkesuma, 2023-07-01:
> @yosafatkesuma For the error `ERROR: The hash sum of the component "espressif/esp_tinyusb(1.1.0)" does not match the one recorded in your dependencies.lock file`, delete `dependencies.lock` file under your project then clean the build folder may help.

Thank you @leeebo . After i follow your instruction, i got new issue with port

[image]

even i check with the screen and the result is like this
[image]

could you help me with this?

yosafatkesuma, 2023-07-03:
> Thank you @leeebo . After i follow your instruction, i got new issue with port
>
> [image]
>
> even i check with the screen and the result is like this [image]
>
> could you help me with this?

for this issue, run this command 

> sudo chown $USER /dev/ttyUSB0

closed 2023-07-03 by yosafatkesuma