espressif/esp-iot-solution #282: usb_dongle command not recognised, ends up in infinite loop

idrougge, 2023-07-02:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM, ESP32S2F
- IDF version (run ``git describe --tags`` to find it): v4.4.5-104-g8b94183
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. Check out and install esp-idf release v4.4 according to instructions.
2. Check out esp-iot-solution branch `add_usb_solutions`.
3. Go to `esp-iot-solution/example/usb/device/usb_dongle`
4. `idf.py set-target esp32s2`
5. `idf.py menuconfig`
6. In `menuconfig`, go to `Components → TinyUSB Stack`.
7. Enable TinyUSB NET driver.
8. Set RNDIS network communication standard.
9. Enable TinyUSB DFU feature.
10. Enable TinyUSB CDC feature.
11. Enable TinyUSB CDC-ACM feature.
12. Set number of Serial (CDC) Port to 2.
13. Check that `Example Configuration` in `menuconfig` root is disabled.
14. `idf.py build dfu dfu-flash`
15. After flashing is complete, reset device.
16. Connect terminal to `usbmodem1234563`.
17. Type `help`, press return.
18. Serial console return `>cognised.  Enter ›help‹ to view a list of available commands.`
19. Type `help` again.
20. Serial console will now enter an infinite loop outputting `>cognised.  Enter ›help‹ to view a list of available commands.`

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

Refer to https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/device/usb_dongle/README_EN.md
[... 1471 lines cut ...]
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

tswen, 2023-07-05:
Hi @idrougge 
I didn't quite understand what usbmodem1234563 is. When I have all the mentioned classes such as CDC, NET, DFU enabled and I open the serial port tool on /dev/ttyACM0 and enter the "help" command, everything seems to be working fine. Although I tried it with an ESP32-S3.

idrougge, 2023-07-09:
`1234563` is the predefined identifier for the CDC unit.

I have tried this on both a ESP32-WROOM and a ESP32S2F card and they both fail in the same way. Unfortunately I have no S3 units to test on.

yunfusheng33, 2024-07-03:
@idrougge I would like to ask how you communicate between S3 and PC，I have burned the project and only print it, but I cannot input it to S3

Rokachy, 2024-10-10:
I have the same issue with S3 DevKit using SDK5.2.2 version :(
My laptop runs Win10, I run terminal software (TeraTerm) and connect "UART" micro USB connector.
I can see prints coming but cannot send any command, there isn't any prompt. you can see below the log after powerup.
I tried to disable the CDC but it didn't help.  
When I connect to the "USB" connector I can see the RNDIS device at windows.
I didn't change any line of code.  
Power up log:
---------------
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1788
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc5c
load:0x403cc700,len:0x2d18
entry 0x403c98f8
I (27) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (27) boot: compile time Oct 10 2024 15:14:10
W (27) boot: Unicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (114) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=248ech (149740) map
I (154) esp_image: segment 1: paddr=00034914 vaddr=3fc97f00 size=042f4h ( 17140) load
I (158) esp_image: segment 2: paddr=00038c10 vaddr=40374000 size=07408h ( 29704) load
I (166) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=940cch (606412) map
I (277) esp_image: segment 4: paddr=000d40f4 vaddr=4037b408 size=0cab8h ( 51896) load
I (297) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (308) cpu_start: Unicore app
I (309) cpu_start: Single core mode
I (318) cpu_start: Pro cpu start user code
I (318) cpu_start: cpu freq: 240000000 Hz
I (318) cpu_start: Application information:
I (321) cpu_start: Project name:     usb_dongle
I (326) cpu_start: App version:      4730d91d
I (331) cpu_start: Compile time:     Oct 10 2024 15:34:35
I (337) cpu_start: ELF file SHA256:  4dd169349...
I (342) cpu_start: ESP-IDF:          v5.2.2-dirty
I (348) cpu_start: Min chip rev:     v0.0
I (352) cpu_start: Max chip rev:     v0.99
I (357) cpu_start: Chip rev:         v0.1
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 3FCA12C8 len 00048448 (289 KiB): RAM
I (375) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (381) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (388) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (395) spi_flash: detected chip: generic
I (399) spi_flash: flash io: dio
W (402) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (416) sleep: Configure to isolate all GPIO pins in sleep state
I (422) sleep: Enable automatic switching of GPIO sleep configuration
I (430) coexist: coex firmware version: d96c1e51f
I (435) coexist: coexist rom version e7ae62f
I (440) main_task: Started on CPU0
I (444) main_task: Calling app_main()
I (452) USB_Dongle: USB initialization
W (453) tusb_desc: No Device descriptor provided, using default.
W (459) tusb_desc: No FullSpeed configuration descriptor provided, using default.
W (467) tusb_desc: No String descriptors provided, using default.
I (474) tusb_desc:
---------------------------------------------------
x  USB Device Descriptor Summary  x
---------------------------------------------------
xbDeviceClass           0           
---------------------------------------------------
xbDeviceSubClass     0           
---------------------------------------------------
xbDeviceProtocol      0           
---------------------------------------------------
xbMaxPacketSize0     64          
---------------------------------------------------
xidVendor                  0x303a      
---------------------------------------------------
xidProduct                 0x4000      
---------------------------------------------------
xbcdDevice                 0x100       
---------------------------------------------------
xiManufacturer            0x1         
---------------------------------------------------
xiProduct                     0x2         
---------------------------------------------------
xiSerialNumber            0x3         
---------------------------------------------------
xbNumConfigurations  0x1         
---------------------------------------------------
I (642) TinyUSB: TinyUSB Driver installed
I (648) pp: pp rom version: e7ae62f
I (651) net80211: net80211 rom version: e7ae62f
I (660) wifi_init: rx ba win: 6
I (660) wifi_init: tcpip mbox: 32
I (664) wifi_init: udp mbox: 6
I (667) wifi_init: tcp mbox: 6
I (671) wifi_init: tcp tx win: 5760
I (675) wifi_init: tcp rx win: 5760
I (679) wifi_init: tcp mss: 1440
I (683) wifi_init: WiFi IRAM OP enabled
I (688) wifi_init: WiFi RX IRAM OP enabled
I (693) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
I (731) uart: queue free spaces: 20
I (732) UART: uart[2] event:
I (732) UART: uart rx break
I (732) USB_Dongle: USB initialization DONE
I (736) main_task: Returned from app_main()

tswen, 2024-10-11:
cdc will virtualize a serial port /dev/ttyACM0. You need to use the serial port debugging tool to open /dev/ttyACM0 for communication.

Rokachy, 2024-10-29:
I am working with TeraTerm tool on Windows 10.
Can you send me more details how to do it?
I assume that I need to connect to the 'USB' connector and not 'UART'
connector on the ESP32S3 EVB

lijunru-hub, 2024-10-31:
You need to connect the chip's USB port to the computer, then use a serial port tool on the computer to communicate with the CDC port, and send the Wi-Fi information to the chip.