espressif/esp-iot-solution #285: No YUYV support

sxrap1, 2023-07-29:
Camera is YUYV 512 bytes transfer, and yes it will be slow but thats fine.
However the new stream component and examples all force to MJPEG, which is not an option.
Im happy to code in the YUYV support, but really have no idea where to start, just looking for some pointers to bring this functionality in.

Thanks

leeebo, 2023-07-30:
@sxrap1 Yes the [usb_stream ](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/usb_stream) supports MJPEG only due to the limited throughput of USB Fullspeed, while adding support for YUV is possible but quite slow.

**YUV format possible frame rate:**

* For YUV frame size 320*240, on the frame is 320*240*2 = 153,600 Bytes.
* For ESP32S3 Isochronous transfer (MPS = 512), the throughput is 512*1000 = 512,000Bytes/s
* For ESP32S3 Bulk transfer, the throughput is 64*19*1000 = 1216,000 Bytes/s
* The final possible frame rate should be  `8 > frame rate  >= 3` Frame/s

**How to support YUV:**

In `_update_config_from_descriptor` add a case handler for the `VIDEO_CS_ITF_VS_FORMAT_UNCOMPRESSED` to get the detailed info for the YUV format like the [MJEPG format](](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/usb_stream/usb_stream.c#L955) ). In particular for uncompressed format, you need to check the `guidFormat` to judge which format the camera support eg. `YUY2` `NV12`, for details please refer [USB_Video_Payload_Uncompressed_1.5.pdf](https://github.com/espressif/esp-iot-solution/files/12208419/USB_Video_Payload_Uncompressed_1.5.pdf)

For the rest payload handler, the processing process should be the same for both YUV and MJPEG

sxrap1, 2023-08-03:
Thank you, I have got the code running now to the point it starts streaming.
However no matter what settings I try from the format and frame descriptor almost all packet headers have error flag set, 1 in about 30 comes through with 500 bytes with the EOF flag set.
Slowly driving me crazy, but I will keep at it.

leeebo, 2023-08-04:
@sxrap1 `error flag` in headers because the bandwidth is too low to transmit the frame rate you set, which may cause the image to break, so the camera set the bit.

sxrap1, 2023-08-04:
Yes , starting to think the camera may be too fast for little esp32 s3,

Ive tried pretty much everything, but from what I can see from below 4000000 (25 FPS) is as slow as I can get Cam to send.

Frame 37: 386 bytes on wire (3088 bits), 386 bytes captured (3088 bits) on interface \\.\USBPcap2, id 0
    Section number: 1
    Interface id: 0 (\\.\USBPcap2)
        Interface name: \\.\USBPcap2
        Interface description: USBPcap2
    Encapsulation type: USB packets with USBPcap header (152)
    Arrival Time: Aug  4, 2023 09:13:37.404325000 E. Australia Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1691104417.404325000 seconds
    [Time delta from previous captured frame: 0.106572000 seconds]
    [Time delta from previous displayed frame: 0.106572000 seconds]
    [Time since reference or first frame: 6.188138000 seconds]
    Frame Number: 37
    Frame Length: 386 bytes (3088 bits)
    Capture Length: 386 bytes (3088 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo:usbvideo]
USB URB
    [Source: 2.19.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffc48c52b4d010
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
        0000 000. = Reserved: 0x00
        .... ...1 = Direction: PDO -> FDO (0x1)
    URB bus id: 2
    Device address: 19
    Endpoint: 0x80, Direction: IN
        1... .... = Direction: IN (1)
        .... 0000 = Endpoint number: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 358
    [Request in: 36]
    [Time from request: 0.106572000 seconds]
    Control transfer stage: Complete (3)
CONFIGURATION DESCRIPTOR
    bLength: 9
    bDescriptorType: 0x02 (CONFIGURATION)
    wTotalLength: 358
    bNumInterfaces: 2
    bConfigurationValue: 1
    iConfiguration: 4
    Configuration bmAttributes: 0x80  NOT SELF-POWERED  NO REMOTE-WAKEUP
        1... .... = Must be 1: Must be 1 for USB 1.1 and higher
        .0.. .... = Self-Powered: This device is powered from the USB bus
        ..0. .... = Remote Wakeup: This device does NOT support remote wakeup
    bMaxPower: 50  (100mA)
INTERFACE ASSOCIATION DESCRIPTOR
    bLength: 8
    bDescriptorType: 0x0b (INTERFACE ASSOCIATION)
    bFirstInterface: 0
    bInterfaceCount: 2
    bFunctionClass: Video (0x0e)
    bFunctionSubClass: 0x03
    bFunctionProtocol: 0x00
    iFunction: 5
INTERFACE DESCRIPTOR (0.0): class Video
    bLength: 9
    bDescriptorType: 0x04 (INTERFACE)
    bInterfaceNumber: 0
    bAlternateSetting: 0
    bNumEndpoints: 1
    bInterfaceClass: Video (0x0e)
[... 348 lines cut ...]
    bInterval: 1
INTERFACE DESCRIPTOR (1.7): class Video
    bLength: 9
    bDescriptorType: 0x04 (INTERFACE)
    bInterfaceNumber: 1
    bAlternateSetting: 7
    bNumEndpoints: 1
    bInterfaceClass: Video (0x0e)
    bInterfaceSubClass: 0x02
    bInterfaceProtocol: 0x00
    iInterface: 0
ENDPOINT DESCRIPTOR
    bLength: 7
    bDescriptorType: 0x05 (ENDPOINT)
    bEndpointAddress: 0x81  IN  Endpoint:1
        1... .... = Direction: IN Endpoint
        .... 0001 = Endpoint Number: 0x1
    bmAttributes: 0x05
        .... ..01 = Transfertype: Isochronous-Transfer (0x1)
        .... 01.. = Synchronisationtype: Asynchronous (0x1)
        ..00 .... = Behaviourtype: Data-Endpoint (0x0)
    wMaxPacketSize: 5120
        ...1 0... .... .... = Transactions per microframe: 3 (2)
        .... ..00 0000 0000 = Maximum Packet Size: 0
    bInterval: 1

codyprupp, 2023-08-30:
@sxrap1 Would it be possible for you to share how you changed the `_update_config_from_descriptor` function? I'm having the same issue and would like to check if I'm doing it correctly. Thanks!

jeff-getlucky, 2024-03-08:
> @sxrap1 Would it be possible for you to share how you changed the `_update_config_from_descriptor` function? I'm having the same issue and would like to check if I'm doing it correctly. Thanks!

Have you solved this problem?

codyprupp, 2024-03-08:
> Have you solved this problem?

No, I opted for a different video streaming method for my camera instead.

sxrap1, 2024-03-08:
Never really got this going , in the end I switched to a more capable
processor. Cameras with bandwidth under about 3 or 4Mps will work, that
pretty much rules out YUYV at anything more than 5 or 6 frames per second
(assuming the camera can be set down to that).
Im now using a banana pi cm4 running ubuntu and YOLOv5, so way beyond what
ESP could ever do.