espressif/esp-iot-solution #287: "TypeError: Version is not semantic" while building any esp-iot-solution examples

armwdev, 2023-08-09:
## Environment

- Development Kit:      ESP32S3_USB_OTG
- Module or chip used:  ESP32S3
- IDF version: idf-v5.1.0
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 12.2.0_20230208
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

I am not sure if that's a bug or my mistake as I tried on multiple PCs. When I tried to build any examples (lvgl_example) from the esp-iot-solution of the master branch, lvgl (8.3) and esp-idf latest stable release (5.1), it fails to build with error: "TypeError: Version is not semantic". You can see the full logs in the "Debug Logs" at the bottom.

### Expected Behavior

As there is no modifications to the files, it should build the examples normally.

### Actual Behavior

It builds the esp-idf examples fine but not the esp-iot-solution examples.

### Steps to repropduce

1- Downloaded the latest stable ESP-IDF, tools, lvgl and master branch of the esp-iot-solution
2- cd to any example project (lvgl_example), set the target to esp32s3, applied idf.py build
3- It starts to build but at the end it shows the error above and logs at the bottom.

### Code to reproduce this issue:
From main.cpp of the lvgl_example below:
```cpp
/* component includes */
#include <stdio.h>

/* freertos includes */
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "esp_freertos_hooks.h"
#include "iot_board.h"
#include "lvgl_gui.h"
#include "esp_log.h"

#include "lv_examples/src/lv_demo_printer/lv_demo_printer.h"
#include "lv_examples/src/lv_demo_widgets/lv_demo_widgets.h"
#include "lv_examples/src/lv_ex_get_started/lv_ex_get_started.h"
#include "lv_examples/src/lv_demo_benchmark/lv_demo_benchmark.h"
#include "lv_examples/src/lv_demo_keypad_encoder/lv_demo_keypad_encoder.h"
#include "lv_examples/src/lv_demo_stress/lv_demo_stress.h"
#include "lv_examples/src/lv_ex_style/lv_ex_style.h"

static const char *TAG = "example_lvgl";

void app_main()
{
    iot_board_init();
    spi_bus_handle_t spi2_bus = iot_board_get_handle(BOARD_SPI2_ID);

    scr_driver_t lcd_drv;
    touch_panel_driver_t touch_drv;
    scr_interface_spi_config_t spi_lcd_cfg = {
        .spi_bus = spi2_bus,
        .pin_num_cs = BOARD_LCD_SPI_CS_PIN,
        .pin_num_dc = BOARD_LCD_SPI_DC_PIN,
        .clk_freq = BOARD_LCD_SPI_CLOCK_FREQ,
        .swap_data = true,
    };
[... 103 lines cut ...]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\esp\idf-v5.1\tools-v5.1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 529, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
                                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\esp\idf-v5.1\tools-v5.1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 146, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\esp\idf-v5.1\tools-v5.1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 99, in is_solve_required
      component.name, spec='=={}'.format(component.version.semver))
                                         ^^^^^^^^^^^^^^^^^^^^^^^^
    File "c:\esp\idf-v5.1\tools-v5.1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_tools\manifest\manifest.py", line 275, in semver
      raise TypeError('Version is not semantic')

  TypeError: Version is not semantic

Call Stack (most recent call first):
  C:/esp/idf-v5.1/esp-idf/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:16 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp/esp-iot-solution/examples/hmi/lvgl_example/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\esp\esp-iot-solution\examples\hmi\lvgl_example\build -S c:\esp\esp-iot-solution\examples\hmi\lvgl_example" terminated with exit code: 1. 

```

leeebo, 2023-08-10:
@armwdev Hi the examples under [examples/hmi](https://github.com/espressif/esp-iot-solution/tree/master/examples/hmi#the-lvgl-examples-is-temporary-disabled) were temporarily disabled. As we have [removed all submodules from the repo](https://github.com/espressif/esp-iot-solution/commit/9e83a8e6795b05fd0276ab2de521b3a9122ca844), we will merge a new version code (based on [esp component manager](https://components.espressif.com/components/espressif/esp_lvgl_port)) and guide docs about HMI soon.

Currently please refer the examples in {esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples)

armwdev, 2023-08-10:
@leeebo Hi and thanks for the quick response. I was mainly interested in the lvgl_wificonfig example to set the wifi credentials using the lvgl as this seems a bit complicated task. Is there any way so I can make modifications to make the lvgl_wificonfig example to work with the idf 5.1?

As a workaround, I've just modified the manifest.py to return 0 instead of the previous error. But now I receive "fatal error: esp_event.h: No such file or directory" even though I included the "REQUIRES esp_event" in the CMakeLists.txt.

Please let me know if there is a better way or how do I fix this error.

Thanks

leeebo, 2023-08-11:
@armwdev  For `wifi credentials` using LVGL, there is an example [86box_smart_panel](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/86box_smart_panel), we suggest developing based it

armwdev, 2023-08-12:
Thanks for your suggestion. That example seems a bit resource intensive and incompatible with the development kit that I have. I need a compact and simpler example so I hope you can fix the hmi examples soon.

closed 2023-08-12 by armwdev