espressif/esp-iot-solution #291: Usb_camera _wifi transfer not printing the camera descriptor

nadeem268, 2023-08-21:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- DevKitC-1 
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
When build the code , i got an error msg related to the watch dog timer reset, not printing the usbc camera derscriptor 
*** Configuration descriptor ***
wTotalLength 317
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x87   EP 7 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 2
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 89
        bInterfaceClass 0x55 (Unknown)
        bInterfaceSubClass 0x59

E (134522) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (134522) task_wdt:  - IDLE (CPU 1)
E (134522) task_wdt: Tasks currently running:
E (134522) task_wdt: CPU 0: IDLE
E (134522) task_wdt: CPU 1: usb_proc
E (134522) task_wdt: Print CPU 1 backtrace

Backtrace: 0x4037A1CA:0x3FC9AFA0 0x403775C5:0x3FC9AFC0 0x400559DD:0x3FCCB080 |<-CORRUPTED
0x4037a1ca: esp_crosscore_isr at /home/nadeem/esp/esp-idf/components/esp_system/crosscore_int.c:96

0x403775c5: _xt_lowint1 at /home/nadeem/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

  

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
![WhatsApp Image 2023-05-23 at 16 25 36](https://github.com/espressif/esp-iot-solution/assets/79097012/a347acfe-cc38-4478-95bb-318c308f1a82)

// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
I (129506) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
wTotalLength 317
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x87   EP 7 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 2
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 89
        bInterfaceClass 0x55 (Unknown)
        bInterfaceSubClass 0x59
E (134522) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (134522) task_wdt:  - IDLE (CPU 1)
E (134522) task_wdt: Tasks currently running:
E (134522) task_wdt: CPU 0: IDLE
E (134522) task_wdt: CPU 1: usb_proc
E (134522) task_wdt: Print CPU 1 backtrace

Backtrace: 0x4037A1CA:0x3FC9AFA0 0x403775C5:0x3FC9AFC0 0x400559DD:0x3FCCB080 |<-CORRUPTED
0x4037a1ca: esp_crosscore_isr at /home/nadeem/esp/esp-idf/components/esp_system/crosscore_int.c:96

0x403775c5: _xt_lowint1 at /home/nadeem/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

leeebo, 2023-08-23:
@nadeem268 Hi, please set the log level to verbose to print more logs.

And please try the basic [host/usb_host_lib](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) example to print the camera's descriptors.

There may be some compatibility issues in the descriptors processing stage in usb_stream

nadeem268, 2023-08-25:
Hello sir!
Now i got this type's of error from this example code host/usb_host_lib
<https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib>
.When i plug -in my camera
V (470) esp_image: segment 4 map_segment 0 segment_data_offs 0x40fbc
load_addr 0x40375df4
0x40375df4: efuse_ll_get_chip_wafer_version_major at
/home/nadeem/esp/esp-idf/components/hal/esp32s3/include/hal/efuse_ll.h:56
 (inlined by) efuse_hal_get_major_chip_version at
/home/nadeem/esp/esp-idf/components/hal/esp32s3/efuse_hal.c:35

I (479) esp_image: segment 4: paddr=00040fbc vaddr=40375df4 size=0bdf4h (
48628) load
D (487) esp_image: free data page_count 0x000001ff
D (492) bootloader_flash: rodata starts from paddr=0x00040fbc, size=0xbdf4,
will be mapped to vaddr=0x3c000000
V (502) bootloader_flash: after mapping, starting from paddr=0x00040000 and
vaddr=0x3c000000, 0x10000 bytes are mapped
V (524) esp_image: image start 0x00010000 end of last section 0x0004cdb0
D (524) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
D (527) boot: Calculated hash:
95b0cf96cb8afff7cd1687c2e4e5ac1bcd2910710a27ced4ed04be4d7ccd0c82
I (541) boot: Loaded app from partition at offset 0x10000
I (541) boot: Disabling RNG early entropy source...
D (547) boot: Mapping segment 0 as DROM
D (551) boot: Mapping segment 3 as IROM
D (555) boot: calling set_cache_and_start_app
D (559) boot: configure drom and irom and start
V (564) boot: rodata starts from paddr=0x00010020, vaddr=0x3c030020,
size=0xbe7c
V (571) boot: after mapping rodata, starting from paddr=0x00010000 and
vaddr=0x3c030000, 0x10000 bytes are mapped
V (582) boot: text starts from paddr=0x00020020, vaddr=0x42000020,
size=0x20f94
0x42000020: _stext at ??:?

V (589) boot: after mapping text, starting from paddr=0x00020000 and
vaddr=0x42000000, 0x30000 bytes are mapped
D (599) boot: start: 0x40375254
0x40375254: call_start_cpu0 at
/home/nadeem/esp/esp-idf/components/esp_system/port/cpu_start.c:249

I (614) cpu_start: Pro cpu up.
I (614) cpu_start: Starting app cpu, entry point is 0x403751b8
0x403751b8: call_start_cpu1 at
/home/nadeem/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (628) cpu_start: Pro cpu start user code
I (628) cpu_start: cpu freq: 240000000 Hz
I (628) cpu_start: Application information:
I (631) cpu_start: Project name:     usb_host_lib_example
I (637) cpu_start: App version:      1
I (642) cpu_start: Compile time:     Aug 25 2023 15:18:54
I (648) cpu_start: ELF file SHA256:  ef4f9a5cb3bfac37...
I (654) cpu_start: ESP-IDF:          v5.0-541-g885e501d99-dirty
I (660) cpu_start: Min chip rev:     v0.0
I (665) cpu_start: Max chip rev:     v0.99
I (670) cpu_start: Chip rev:         v0.1
I (675) heap_init: Initializing. RAM available for dynamic allocation:
I (682) heap_init: At 3FC949B0 len 00054D60 (339 KiB): D/IRAM
I (688) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (695) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (701) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (708) spi_flash: detected chip: generic
I (712) spi_flash: flash io: dio
W (716) spi_flash: Detected size(8192k) larger than the size in the binary
image header(2048k). Using the size in the binary image header.
I (730) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (750) DAEMON: Installing USB Host Library
I (780) CLASS: Registering Client
E (3980) HUB: Incorrect number of bytes returned: CHECK_FULL_CONFIG_DESC
E (3980) HUB: Stage failed: CHECK_FULL_CONFIG_DESC

Nadeemranm, 2023-08-31:
Hello sir !
please give me any solution

leeebo, 2023-09-01:
@nadeem268 If [host/usb_host_lib](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) can not enum the device's descriptors successfully, maybe the device not support Full-speed.

Nadeemranm, 2023-09-01:
Any other way to print device's descriptors, and which parameter will be change for half -speed support ?

nadeem268, 2023-09-05:
What?