espressif/esp-iot-solution #292: How do I use the button component with the keypad?

ertekinsener, 2023-09-01:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version : [v1]
- Module or chip used:  [ESP32 S3]
- IDF version : v5.0.2
- Build System:         [Make|CMake]
- Compiler version : 11.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

How can I use button components with keypads? I tried to handle it with button_custom_get_key_value . I couldn't do it because the function takes the gpio number as a parameter.

A detailed explanation of the keypads I am trying to use is available at this link: https://lastminuteengineers.com/arduino-keypad-tutorial/

Thanks for your answers

lijunru-hub, 2023-09-11:
The default GPIO detection for buttons is not suitable for a matrix keyboard. You need to use the `button_custom_config_t` in the button component to create a custom button. Set the `priv` field to your current key's custom identifier, and then you need to create 16 button objects (4x4 matrix). The button will automatically invoke the `button_custom_get_key_value` function you've configured and pass `priv` to it. Within that function, you should determine if a key is pressed and return its status.

ertekinsener, 2023-09-13:
Thank you for your reply. I did as you said and managed to get it to work. Do you plan to add keypad support to the library in the future?

Secondly,  **" TEST_CASE("gpio button test", "[button][iot]")**"   What does this do? I couldn't find it in the documentation.

leeebo, 2023-09-13:
@ertekinsener `TEST_CASE("gpio button test", "[button][iot]")` is used for component unit test, you can refer the doc https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/unit-tests.html

lijunru-hub, 2023-10-12:
Components [button](https://components.espressif.com/components/espressif/button) support matrix keypad on version 3.1.0

closed 2023-10-12 by leeebo