espressif/esp-iot-solution #294: GPIO 14 modem hardware control needed?

sirtet, 2023-09-09:
Is GPIO 14  mandatory in all cases?
In the Photo in the readme, only USB seems to be connected...
I am considering using a 4g USB dongle. That would not be possible then, right?

leeebo, 2023-09-11:
@sirtet Sorry, I didn't get your point. Do you mean the `Modem Reset Control` in [usb_cdc_4g_module#hardware-requirement](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module#hardware-requirement)? in this case it's better to let esp32 control modem reset to handle some error case.

sirtet, 2023-09-11:
Yes, that was my point. So, this line is not mandatory, but optimal? If so, the readme should maybe be corrected.
Am I correct that the demo photo&video do not use the reset line?
And as a follow up question: I guess the reset could also be done by Interrupting the USB 5v, if no direct  access to the modem's reset line is available?

leeebo, 2023-09-11:
@sirtet Yes, the demo photo does not use the reset line, but it is recommended in the product.

If the VBUS can be controlled by your ESP32, power-off the power-on has the same effect as reset, In this case reset line is optional

closed 2023-11-02 by leeebo