espressif/esp-iot-solution #296: Will Bluetooth OTA update be supported on ESP32-H2?

johannchangpro, 2023-09-20:
Is OTA over Bluetooth be supported on ESP32-H2? ESP32-H2 does not have Wi-Fi, if Bluetooth OTA is not supported, there's no way we can OTA update it.

lhespress, 2023-09-22:
@johannchangpro Please apply the patch as the attachment.

[ESP32-H2.zip](https://github.com/espressif/esp-iot-solution/files/12700638/ESP32-H2.zip)

johannchangpro, 2023-09-22:
Thank you very much @lhespress, it works.

I actually did that before posting the question. I realized I shouldn't follow the [Nimble OTA section in the README](https://github.com/espressif/esp-iot-solution/tree/master/examples/bluetooth/ble_ota#5-nimble-ota) for ESP32-H2. It won't compile. That's why I posted this issue. I saw many code of Bluetooth for ESP32-H2 only for Nimble, which led me think I had to use Nimble.

closed 2023-09-22 by johannchangpro