espressif/esp-iot-solution #298: USB CDC 4G MODEM - SPEED ISSUE

lovepreethitech, 2023-10-04:
Hi @leeebo 

I am using ESP32S3-WROOM1-N16R8 with my modem, I am facing speed issues. Ping time is in thousands of seconds and taking too much time to open to webpages even.

MODEM ENDPOINTS - 0x85 0x04

Please suggest what could i improve to increase speed.
I Checked my modem in dialup mode with directly connected to PC giving good speed. 

Attached Menuconfig -
[sdkconfig.zip](https://github.com/espressif/esp-iot-solution/files/12802996/sdkconfig.zip)

leeebo, 2023-10-07:
@lovepreethitech Hi, which 4G model are you using?

lovepreethitech, 2023-10-07:
@leeebo I m using Neoway N58 module - https://www.neoway.com/products/wireless-communication/823.html

Also i have tried the same on Quectel EC200u but similar speeds.

previously i was usign ESP32S2 without PSRAM but getting slows speeds then shifted to ESP32S3 N16R8 but same results.

Please suggest if there are any configurations i can tweak.

leeebo, 2023-10-16:
@lovepreethitech Hi would you please paste your download/upload test result? [here](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module#performance-test) is the result from our test.

lovepreethitech, 2023-10-16:
@leeebo Hi, I'm getting speed usually avg about 2.5MBps and upload is always less than 1 MBps.

Here image attached [image]