espressif/esp-iot-solution #299: led_indicator features

espzav, 2023-10-05:
Hello,
please, could you help me understand the LED Indicator component? I thought, that it will be similar like iot_button, but it was false.

1. How can I use LED Indicator component for on/off LED? When I configure LED, it can blink, but I don't know, how can I on of off.
2. Please, is there a plan to add LED Strip features into this component? Something like ADC support in iot_button?

It will help with Generic BSP, which I am making now.

Thank you for help.

lijunru-hub, 2023-10-07:
This is a bug that will be fixed in version `0.4.1`.

If there are any requirements, please feel free to mention them.

In the future, we will support the WS2812 driver and provide the ability to modify settings. However, it's possible that gradual color fading may not be supported initially.

espzav, 2023-10-09:
Hi @lijunru-hub,
thank you for the response. Please, do you know, when the version 0.4.1 will be released?

Support of the WS2812 will be good. I used this component https://components.espressif.com/components/espressif/led_strip now and added it in custom config. It is working, but I haven't got API for change color. For the first, there can be function for set color, and after that on/off, blink and fade all channels of brightness in same values. 

The last think, there will be nice to change default settings here: https://github.com/espressif/esp-iot-solution/blob/master/components/led/led_indicator/src/led_indicator_blink_default.c to something more independent. For example:
1. LED_ON
2. LED_OFF
3. LED_BLINK_SLOW
4. LED_BLINK_FAST
5. LED_BREATHE_SLOW
6. LED_BREATHE_FAST

Thank you very much!

espzav, 2023-10-19:
Hi,
FIY: I updated `led_strip` component for support in IDF4.4, it can help to support RGB inside the `led_indicator`. Here is PR: https://github.com/espressif/idf-extra-components/pull/253

lijunru-hub, 2023-10-19:
@espzav version 0.5.0 will released as soon as possible.

leeebo, 2023-11-01:
Hi @espzav , the feature has been added to [espressif/led_indicator](https://components.espressif.com/components/espressif/led_indicator) through ee5f4cc0e71edb55b45907305bf0c46017641f19. please update to `version: 0.6.*`  to have a try

espzav, 2023-11-01:
Hi @leeebo, Thank you very much! Great job!