espressif/esp-iot-solution #300: [20:39] Savita Pawar  Query regarding cyber security support for ESP32

Abhay-2412, 2023-10-06:
Hi Team,

 

We are using ESP32 in our project and use case is capture the YUV frames and send to Mobile application over Wi-Fi. Here ESP-32 is configured as AP mode. We are facing issue in sending accurate data to mobile application.

 

In our testing, we observed that Wi-Fi is taking more time to send data. Looks like Wi-Fi send data is blocking call and due to this it is taking time.

 

Can you please suggest any solution to make this Wi-Fi call non-blocking and improve transfer time? Or any other suggestion to improve this.

leeebo, 2023-10-07:
Hi @Abhay-2412 , for the camera wi-fi transfer solution, please refer https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/pic_server

Abhay-2412, 2023-10-07:
Hello @leeebo , I checked the link you mentioned, but didn't find anything to do with YUV .

closed 2024-05-21 by leeebo