espressif/esp-iot-solution #301: 关于USB MSC Wireless Disk例程中提到的已知问题，Known Issues

MichaelDu9226, 2023-10-07:
USB MSC Wireless Disk例程中提到：
[Known Issues](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_msc_wireless_disk/README.md#known-issues)
Files uploaded through web can not be aware by host , so Windows files resource manager can not update the files list automatically. Please remount the disk to update the files list (弹出重新加载) . This bug will be fixed later.
Files added or removed through USB disk, sometimes can not be found by web refresh.

关于MSC驱动在widows文件管理器与ESP32 S3同时写文件，会导致冲突的原因，我大致是清楚的。
但是看到你们提到"This bug will be fixed later"，想了解下何时可以解决这个Bug，或者解决这个Bug的思路是怎样的，我可以借鉴你们的思路去尝试，谢谢

leeebo, 2023-10-07:
If you want to implement HTTP and MSC access "at the same time", the logic will be very complex. We should consider adding locks and remounting in some cases.

```mermaid
stateDiagram-v2
power_on: Power on
wifi_init: Wi-Fi AP/STA Init
usb_monitor: Monitor USB connection
http_init: HTTP File Server Init
fatfs_init: DiskIO + FATFS + VFS Init
handle_http: HTTP File Server Handler
handle_usb: USB MSC Diskio Handler
user_http_write_access: User Wireless upload/delete
user_http_read_access: User Wireless download
user_http_idel: Wireless idle
check_user_http_idel: Check Wireless idle
state Same_Check_Logic_as_USB {
    [*] --> ...
}
check_user_http_only_download: Check HTTP Only Download
user_usb_write_access: User USB Write
user_usb_read_access: User USB Read
user_usb_idel: USB idle
[*] --> power_on
power_on --> wifi_init
wifi_init --> fatfs_init
fatfs_init --> http_init
http_init --> handle_http
handle_http --> user_http_write_access
handle_http --> user_http_read_access
handle_http --> user_http_idel
user_http_write_access --> Same_Check_Logic_as_USB
user_http_read_access --> Same_Check_Logic_as_USB
user_http_idel --> usb_monitor
state if_connected <<choice>>
usb_monitor --> if_connected
if_connected --> handle_usb: USB Connected
handle_usb --> user_usb_write_access
handle_usb --> user_usb_read_access
handle_usb --> user_usb_idel
user_usb_write_access --> check_user_http_idel
check_user_http_idel --> Wait_HTTP_Complete: False
Wait_HTTP_Complete --> Check_if_remount_USB
Check_if_remount_USB --> USB_Write: Write
Check_if_remount_USB --> USB_Read: Read
check_user_http_idel --> Check_if_remount_USB: True
USB_Write --> handle_http: Disk Changed
handle_http --> fatfs_init: Disk Changed Re-mount
user_usb_read_access --> check_user_http_only_download
check_user_http_only_download --> Check_if_remount_USB: True
check_user_http_only_download --> Wait_HTTP_Complete: False
Wait_HTTP_Complete --> Check_if_remount_USB
Check_if_remount_USB --> usb_monitor: Re-mount
if_connected --> Disconnected: USB Disconnected
```

leeebo, 2023-10-07:
That's why Android devices choose protocols like MTP and PTP instead MSC, when they act as a storage device

MichaelDu9226, 2023-10-09:
@leeebo 谢谢你的建议
看起来难度有点大，我放弃了。

leeebo, 2023-10-09:
@MichaelDu9226 新版本会处理掉。到时通知你

MichaelDu9226, 2023-11-17:
@leeebo 你好，请问关于USB MSC PC与ESP32同时读写同步的问题，有什么进展吗？谢谢
我们主要是micropython环境，已经实现了msc，但是两者的读写不同步。
另外还有一个问题，USB msc是否支持动态开启关闭? 就是用户需要点击屏幕或者按键确认开启MSC功能，PC端才会出现U盘。这期间重启设备是可以接受的。目前我使用的msc好像需要在sdconfig里开启

imliubo, 2023-11-27:
应该是同一个问题, 文件列表不同步, 网页刷新不起作用
[image]

hy1080, 2024-03-02:
可不可以以通过http来模拟系统的读写来驱动tinyUSB？持续关注中

volca, 2024-09-04:
> @MichaelDu9226 新版本会处理掉。到时通知你

Any update for the issue?

请问这个问题有什么进展吗？

abc8143, 2024-11-21:
> 应该是同一个问题, 文件列表不同步, 网页刷新不起作用 [...]

我按一下面板上那个RST按钮让他重新启动一下就能看见，那wifi好像都不会断，又或者是那wifi连得太快感觉没断吧，当然usb连肯定断了自动重连了，感觉影响不大，反正我就给那页面做个按钮调一下esp_restart()算了，毕竟就传1个文件给机器干活
[image]

truufseir, 2024-12-10:
MTP and PTP hello leeeebo can i implement these instead to communicate