espressif/esp-iot-solution #302: 我用这个示例下载进我的ESP32-S3会出现这些警告导致系统复位，是什么原因？

huangdiajd, 2023-10-09:
V (1146) memory_layout: reserved range is 0x3c0cf36c - 0x3c0cf394
D (1146) memory_layout: Checking 6 reserved memory ranges:
D (1147) memory_layout: Reserved memory range 0x3c000000 - 0x3e000000
D (1147) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc99600
D (1148) memory_layout: Reserved memory range 0x3fc99600 - 0x3fca1838
D (1148) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (1148) memory_layout: Reserved memory range 0x40374000 - 0x40389600
0x40374000: _WindowOverflow4 at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:2013

D (1149) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (1149) memory_layout: Building list of available memory regions:
V (1149) memory_layout: Examining memory region 0x3c000000 - 0x3e000000
V (1150) memory_layout: Region 0x3c000000 - 0x3e000000 inside of reserved 0x3c000000 - 0x3e000000
V (1150) memory_layout: Examining memory region 0x40374000 - 0x40378000
0x40374000: _WindowOverflow4 at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:2013

0x40378000: _uvc_process_payload at C:/Users/Administrator/Desktop/usb_camera_mic_spk/managed_components/espressif__usb_stream/usb_stream.c:1891
 (discriminator 12)

V (1151) memory_layout: Region 0x40374000 - 0x40378000 inside of reserved 0x40374000 - 0x40389600
0x40374000: _WindowOverflow4 at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:2013

0x40378000: _uvc_process_payload at C:/Users/Administrator/Desktop/usb_camera_mic_spk/managed_components/espressif__usb_stream/usb_stream.c:1891
 (discriminator 12)

0x40374000: _WindowOverflow4 at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:2013

V (1151) memory_layout: Examining memory region 0x3fc88000 - 0x3fc90000
V (1152) memory_layout: Region 0x3fc88000 - 0x3fc90000 inside of reserved 0x3fc84000 - 0x3fc99600
V (1152) memory_layout: Examining memory region 0x3fc90000 - 0x3fca0000
V (1152) memory_layout: Start of region 0x3fc90000 - 0x3fca0000 overlaps reserved 0x3fc84000 - 0x3fc99600
V (1153) memory_layout: Region 0x3fc99600 - 0x3fca0000 inside of reserved 0x3fc99600 - 0x3fca1838
V (1154) memory_layout: Examining memory region 0x3fca0000 - 0x3fcb0000
V (1154) memory_layout: Start of region 0x3fca0000 - 0x3fcb0000 overlaps reserved 0x3fc99600 - 0x3fca1838
D (1154) memory_layout: Available memory region 0x3fca1838 - 0x3fcb0000
V (1155) memory_layout: Examining memory region 0x3fcb0000 - 0x3fcc0000
D (1155) memory_layout: Available memory region 0x3fcb0000 - 0x3fcc0000
V (1156) memory_layout: Examining memory region 0x3fcc0000 - 0x3fcd0000
D (1156) memory_layout: Available memory region 0x3fcc0000 - 0x3fcd0000
V (1156) memory_layout: Examining memory region 0x3fcd0000 - 0x3fce0000
D (1157) memory_layout: Available memory region 0x3fcd0000 - 0x3fce0000
V (1157) memory_layout: Examining memory region 0x3fce0000 - 0x3fce9710
D (1158) memory_layout: Available memory region 0x3fce0000 - 0x3fce9710
V (1158) memory_layout: Examining memory region 0x3fce9710 - 0x3fcf0000
V (1158) memory_layout: End of region 0x3fce9710 - 0x3fcf0000 overlaps reserved 0x3fceee34 - 0x3fcf0000
D (1159) memory_layout: Available memory region 0x3fce9710 - 0x3fceee34
V (1159) memory_layout: Examining memory region 0x3fcf0000 - 0x3fcf8000
D (1160) memory_layout: Available memory region 0x3fcf0000 - 0x3fcf8000
V (1160) memory_layout: Examining memory region 0x600fe000 - 0x60100000
V (1160) memory_layout: Start of region 0x600fe000 - 0x60100000 overlaps reserved 0x600fe000 - 0x600fe010
D (1161) memory_layout: Available memory region 0x600fe010 - 0x60100000
I (1161) heap_init: Initializing. RAM available for dynamic allocation:
D (1162) heap_init: New heap initialised at 0x3fca1838
I (1162) heap_init: At 3FCA1838 len 00047ED8 (287 KiB): DRAM
I (1162) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
D (1163) heap_init: New heap initialised at 0x3fcf0000
I (1163) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
D (1164) heap_init: New heap initialised at 0x600fe010
I (1164) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1164) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
V (1165) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1165) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1166) intr_alloc: Connected src 39 to int 2 (cpu 0)
V (1166) memspi: raw_chip_id: 1840C8

V (1166) memspi: chip_id: C84018

V (1166) memspi: raw_chip_id: 1840C8

V (1166) memspi: chip_id: C84018
[... 80 lines cut ...]
I (1207) wifi_init: WiFi RX IRAM OP enabled
I (1210) camera wifi: wifi_init_softap finished.SSID:ESP32S3-UVC password:
I (1211) phy_init: phy_version 601,fe52df4,May 10 2023,17:26:54
I (1244) wifi:mode : softAP (48:27:e2:67:c3:91)
I (1245) wifi:Total power save buffer number: 16
I (1246) wifi:Init max length of beacon: 752/752
I (1246) wifi:Init max length of beacon: 752/752
I (1247) camera_httpd: Starting web server on port: '80'
I (1248) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1251) camera_httpd: Starting stream server on port: '81'

assert failed: app_main main.c:240 (frame_buffer != NULL)

Backtrace: 0x40375cde:0x3fca52e0 0x4037ebad:0x3fca5300 0x40385dd5:0x3fca5320 0x4200acb6:0x3fca5440 0x42094037:0x3fca54d0 0x403815a9:0x3fca5500
0x40375cde: panic_abort at C:/esp-idf/esp-idf_v5.1/esp-idf/components/esp_system/panic.c:452

0x4037ebad: esp_system_abort at C:/esp-idf/esp-idf_v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40385dd5: __assert_func at C:/esp-idf/esp-idf_v5.1/esp-idf/components/newlib/assert.c:81

0x4200acb6: app_main at C:/Users/Administrator/Desktop/usb_camera_mic_spk/main/main.c:240 (discriminator 1)

0x42094037: main_task at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x403815a9: vPortTaskWrapper at C:/esp-idf/esp-idf_v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

huangdiajd, 2023-10-09:
用的示例是 USB_CAMERA_MIC_SPK

leeebo, 2023-10-09:
@huangdiajd The example works for me both in IDF4.4 and IDF5.0, have you changed the buffer size?

The error `assert failed: app_main main.c:240 (frame_buffer != NULL)` caused by malloc failed, please try to malloc a smaller buffer [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/host/usb_camera_mic_spk/main/main.c#L56), change the frame buffer size from 55KB to 45KB

huangdiajd, 2023-10-09:
> @huangdiajd The example works for me both in IDF4.4 and IDF5.0, have you changed the buffer size?
>
> The error `assert failed: app_main main.c:240 (frame_buffer != NULL)` caused by malloc failed, [...]

OK，I have changed the buffer from 55 to 45 and the program will not restart, but when I connect the USB camera to the ESP32 S3, I get the following error. May I ask how to solve this, or is there any good way to test my USB camera?

E (173951) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2862 (_uvc_uac_device_enum):urb cleared: enum abort
E (173952) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Failed
W (173952) USB_STREAM: USB enum failed, retrying in 200 ms...
W (174174) USB_STREAM: USB enum failed, retry now
I (174174) USB_STREAM: ENUM Stage START, Succeed
I (174176) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
W (174177) USB_STREAM: Pipe(default): ERROR_XFER
I (174177) USB_STREAM: Action: ACTION_PIPE_DFLT_RECOVER
E (174177) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2862 (_uvc_uac_device_enum):urb cleared: enum abort
E (174178) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Failed
E (174178) USB_STREAM: USB enum failed, no more retry
W (178491) USB_STREAM: line 156 HCD_PORT_EVENT_DISCONNECTION

huangdiajd, 2023-10-10:
> @huangdiajd The camera Enum failed. Maybe your camera does not support USB Full-speed, please try the [basic host example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) to confirm it
> 
> If you want to get started quickly, there is a camera on sale in [espressif-online store](https://m.tb.cn/h.5gv01kw?tk=xhDbWbbAhPc).
> 
> BTW, you can also make your own USB Camera Device based on esp32s3 also, please refer [usb/device/usb_webcam](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_webcam)

Hello, I have no problem with this lens, I can identify the camera, and then it is these errors, what is the reason.

*** Configuration descriptor ***
wTotalLength 137
bNumInterfaces 1
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 64
                bInterval 6
W (3781) USB_STREAM: VS Interface(MPS <= 512) NOT found
W (3781) USB_STREAM: Try with first alt-interface config
E (3781) USB_STREAM: MJPEG format NOT found
W (3782) USB_STREAM: MJPEG width*heigh: 65535*65535, NOT found
W (3782) USB_STREAM: Audio control interface NOT found
E (3782) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2965 (_uvc_uac_device_enum):Descriptor Parse Result: No matching co
nfigurations
E (3783) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Failed
W (3783) USB_STREAM: USB enum failed, retrying in 200 ms...
W (4005) USB_STREAM: USB enum failed, retry now
I (4005) USB_STREAM: ENUM Stage START, Succeed
I (4007) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
I (4009) USB_STREAM: Default pipe endpoint MPS update to 64
I (4009) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (4010) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (4021) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (4022) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed

leeebo, 2023-10-10:
The camera not support Fullspeed, so we can not find the correct interface and MJPEG format

```
W (3781) USB_STREAM: VS Interface(MPS <= 512) NOT found
W (3781) USB_STREAM: Try with first alt-interface config
E (3781) USB_STREAM: MJPEG format NOT found
W (3782) USB_STREAM: MJPEG widthheigh: 6553565535, NOT found
W (3782) USB_STREAM: Audio control interface NOT found
E (3782) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2965 (_uvc_uac_device_enum):Descriptor Parse Result: No matching configurations
```

closed 2023-11-02 by leeebo