espressif/esp-iot-solution #303: qspi原始例子不能正确编译

2339212597, 2023-10-14:
## Environment
Error when compiling the example

## Problem Description
open **examples\display\lcd\qspi_with_ram** Direct compilation is error，log below

## Debug Logs

```
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
Processing 9 dependencies:
[1/9] esp_lcd_gc9b71 (0.0.2)
[2/9] esp_lcd_sh8601 (0.0.2)
[3/9] esp_lcd_spd2010 (0.0.2)
[4/9] espressif/cmake_utilities (0.5.2)
[5/9] espressif/esp_lcd_touch (1.1.0)        
[6/9] espressif/esp_lcd_touch_cst816s (1.0.3)
[7/9] espressif/esp_lcd_touch_spd2010 (0.0.1)
[8/9] idf (5.1.1)
[9/9] lvgl/lvgl (8.3.10)
-- Project sdkconfig file C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf

-- App "example_qspi_with_ram" version: 834dd6ce-dirty
-- Adding linker script C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/idf_path/idf/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- ESP_LCD_GC9B71: 0.0.2
-- ESP_LCD_SH8601: 0.0.2
-- ESP_LCD_SPD2010: 0.0.2
-- ESP_LCD_TOUCH_SPD2010: 0.0.1
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_lcd_gc9b71 esp_lcd_sh8601 esp_lcd_spd2010 esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cmake_utilities espressif__esp_lcd_touch espressif__esp_lcd_touch_cst816s espressif__esp_lcd_touch_spd2010 esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lvgl__lvgl lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/idf_path/idf/esp-idf/components/app_trace D:/idf_path/idf/esp-idf/components/app_update D:/idf_path/idf/esp-idf/components/bootloader D:/idf_path/idf/esp-idf/components/bootloader_support D:/idf_path/idf/esp-idf/components/bt D:/idf_path/idf/esp-idf/components/cmock D:/idf_path/idf/esp-idf/components/console D:/idf_path/idf/esp-idf/components/cxx D:/idf_path/idf/esp-idf/components/driver 
D:/idf_path/idf/esp-idf/components/efuse D:/idf_path/idf/esp-idf/components/esp-tls D:/idf_path/idf/esp-idf/components/esp_adc D:/idf_path/idf/esp-idf/components/esp_app_format D:/idf_path/idf/esp-idf/components/esp_coex D:/idf_path/idf/esp-idf/components/esp_common D:/idf_path/idf/esp-idf/components/esp_eth D:/idf_path/idf/esp-idf/components/esp_event D:/idf_path/idf/esp-idf/components/esp_gdbstub D:/idf_path/idf/esp-idf/components/esp_hid D:/idf_path/idf/esp-idf/components/esp_http_client D:/idf_path/idf/esp-idf/components/esp_http_server D:/idf_path/idf/esp-idf/components/esp_https_ota D:/idf_path/idf/esp-idf/components/esp_https_server D:/idf_path/idf/esp-idf/components/esp_hw_support D:/idf_path/idf/esp-idf/components/esp_lcd C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_gc9b71 C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_sh8601 C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_spd2010 D:/idf_path/idf/esp-idf/components/esp_local_ctrl D:/idf_path/idf/esp-idf/components/esp_mm 
D:/idf_path/idf/esp-idf/components/esp_netif D:/idf_path/idf/esp-idf/components/esp_netif_stack D:/idf_path/idf/esp-idf/components/esp_partition D:/idf_path/idf/esp-idf/components/esp_phy D:/idf_path/idf/esp-idf/components/esp_pm D:/idf_path/idf/esp-idf/components/esp_psram D:/idf_path/idf/esp-idf/components/esp_ringbuf D:/idf_path/idf/esp-idf/components/esp_rom D:/idf_path/idf/esp-idf/components/esp_system D:/idf_path/idf/esp-idf/components/esp_timer D:/idf_path/idf/esp-idf/components/esp_wifi D:/idf_path/idf/esp-idf/components/espcoredump C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/managed_components/espressif__cmake_utilities C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/managed_components/espressif__esp_lcd_touch C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/managed_components/espressif__esp_lcd_touch_cst816s C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/managed_components/espressif__esp_lcd_touch_spd2010 D:/idf_path/idf/esp-idf/components/esptool_py D:/idf_path/idf/esp-idf/components/fatfs D:/idf_path/idf/esp-idf/components/freertos D:/idf_path/idf/esp-idf/components/hal D:/idf_path/idf/esp-idf/components/heap D:/idf_path/idf/esp-idf/components/http_parser D:/idf_path/idf/esp-idf/components/idf_test D:/idf_path/idf/esp-idf/components/ieee802154 D:/idf_path/idf/esp-idf/components/json D:/idf_path/idf/esp-idf/components/log C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/managed_components/lvgl__lvgl D:/idf_path/idf/esp-idf/components/lwip C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/main D:/idf_path/idf/esp-idf/components/mbedtls D:/idf_path/idf/esp-idf/components/mqtt D:/idf_path/idf/esp-idf/components/newlib D:/idf_path/idf/esp-idf/components/nvs_flash D:/idf_path/idf/esp-idf/components/openthread D:/idf_path/idf/esp-idf/components/partition_table D:/idf_path/idf/esp-idf/components/perfmon D:/idf_path/idf/esp-idf/components/protobuf-c D:/idf_path/idf/esp-idf/components/protocomm D:/idf_path/idf/esp-idf/components/pthread 
D:/idf_path/idf/esp-idf/components/sdmmc D:/idf_path/idf/esp-idf/components/soc D:/idf_path/idf/esp-idf/components/spi_flash D:/idf_path/idf/esp-idf/components/spiffs D:/idf_path/idf/esp-idf/components/tcp_transport D:/idf_path/idf/esp-idf/components/touch_element D:/idf_path/idf/esp-idf/components/ulp D:/idf_path/idf/esp-idf/components/unity D:/idf_path/idf/esp-idf/components/usb D:/idf_path/idf/esp-idf/components/vfs D:/idf_path/idf/esp-idf/components/wear_levelling D:/idf_path/idf/esp-idf/components/wifi_provisioning D:/idf_path/idf/esp-idf/components/wpa_supplicant D:/idf_path/idf/esp-idf/components/xtensa    
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build
[2/462] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,12M,
*******************************************************************************
[9/462] Building C object esp-idf/esp_lcd_gc9b71/CMakeFiles/__idf_esp_lcd_gc9b71.dir/esp_lcd_gc9b71.c.obj
FAILED: esp-idf/esp_lcd_gc9b71/CMakeFiles/__idf_esp_lcd_gc9b71.dir/esp_lcd_gc9b71.c.obj 
ccache D:\idf_path\5.2tool\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/AMD/Desktop/esp_iot/examples/display/lcd/qspi_with_ram/build/config -IC:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_gc9b71/include -ID:/idf_path/idf/esp-idf/components/newlib/platform_include -ID:/idf_path/idf/esp-idf/components/freertos/FreeRTOS-Kernel/include -ID:/idf_path/idf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/include/freertos -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/include -ID:/idf_path/idf/esp-idf/components/freertos/esp_additions/arch/xtensa/include -ID:/idf_path/idf/esp-idf/components/esp_hw_support/include -ID:/idf_path/idf/esp-idf/compon [... cut ...]
C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_sh8601/esp_lcd_sh8601.c:74:10: note: each undeclared identifier is reported only once for each function it appears in
C:/Users/AMD/Desktop/esp_iot/components/display/lcd/esp_lcd_sh8601/esp_lcd_sh8601.c:77:10: error: 'LCD_RGB_ELEMENT_ORDER_BGR' undeclared (first use in this function)
   77 |     case LCD_RGB_ELEMENT_ORDER_BGR:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
[22/460] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sdl/lv_draw_sdl_composite.c.obj
ninja: build stopped: subcommand failed.
```

## Other items if possible

won‘t upload files

Lzw655, 2023-10-16:
Hi @skyoffcx, please update your IDF. Because the QSPI LCD feature has been updated recently.

skyoffcx, 2023-10-16:
thanks for your reply，Line 12 in my compilation log is mine idf version ，Maybe my current ability may not be able to get any closer to solving the problem，Can you give some suggestions on configuration files or configuration methods？

Lzw655, 2023-10-17:
Hi @skyoffcx, to install the up-to-date release/v5.1 ESP-IDF, please use [online installer](https://dl.espressif.com/dl/esp-idf/?idf=4.4) and chose the release/v5.1(release branch) like below:
[image]

closed 2023-11-02 by leeebo