espressif/esp-iot-solution #305: iot_button not suitable for low-power applications

cyberhuman, 2023-10-25:
Hello,

I was debugging a high power consumption of my custom ESP32-C3-based board in automatic light sleep mode, and I figured out the culprit is the iot_button. When `iot_button_create` is not called, my board consumes ~600uA, with Bluetooth advertising it's ~1.5mA. But creating a GPIO button with default settings adds up another ~2mA. I took a look into the code and it creates a timer with the configured period (5ms by default) and there is a state machine controlled by the GPIO input level.
I would expect GPIO buttons to support activation by external interrupt so the timer doesn't wake up the processor 200 times a second uselessly. Are there any plans to add this functionality? Thank you!

lijunru-hub, 2023-10-26:
In the latest version of iot_button, we've introduced the iot_button_stop function for halting the timer, and you can restart it using iot_button_resume. This can be particularly useful for stopping the timer when entering sleep mode and resuming it upon exit. If your button is of the GPIO type, you can also register an external interrupt outside the iot_button framework. This registration won't conflict with iot_button (make sure to register iot_button first and then the external interrupt). Subsequently, you can restart the iot_button within the external interrupt handler.

Developing a button component triggered by an external interrupt is not a straightforward task.

cyberhuman, 2023-10-26:
Thank you for the information!
Unfortunately it's not straightforward to stop the button before sleeping because the sleep mode is entered automatically.
I'll switch to a manual solution for now.

closed 2023-10-26 by cyberhuman