espressif/esp-iot-solution #306: esp_lcd_st7701 编译失败

fanyg01, 2023-10-27:
使用版本：0.0.2
编译失败信息：

../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'esp_lcd_new_panel_st7701':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:76:29: error: 'esp_lcd_panel_dev_config_t' has no member named 'rgb_ele_order'
   76 |     switch (panel_dev_config->rgb_ele_order) {
      |                             ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:77:10: error: 'LCD_RGB_ELEMENT_ORDER_RGB' undeclared (first use in this function)
   77 |     case LCD_RGB_ELEMENT_ORDER_RGB:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:77:10: note: each undeclared identifier is reported only once for each function it appears in
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:80:10: error: 'LCD_RGB_ELEMENT_ORDER_BGR' undeclared (first use in this function)
   80 |     case LCD_RGB_ELEMENT_ORDER_BGR:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:151:17: error: 'struct esp_lcd_panel_t' has no member named 'user_data'
  151 |     (*ret_panel)->user_data = st7701;
      |                 ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'panel_st7701_init':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:267:53: error: 'esp_lcd_panel_t' has no member named 'user_data'
  267 |     st7701_panel_t *st7701 = (st7701_panel_t *)panel->user_data;
      |                                                     ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'panel_st7701_del':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:280:53: error: 'esp_lcd_panel_t' has no member named 'user_data'
  280 |     st7701_panel_t *st7701 = (st7701_panel_t *)panel->user_data;
      |                                                     ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'panel_st7701_reset':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:294:53: error: 'esp_lcd_panel_t' has no member named 'user_data'
  294 |     st7701_panel_t *st7701 = (st7701_panel_t *)panel->user_data;
      |                                                     ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'panel_st7701_mirror':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:315:53: error: 'esp_lcd_panel_t' has no member named 'user_data'
  315 |     st7701_panel_t *st7701 = (st7701_panel_t *)panel->user_data;
      |                                                     ^~
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c: In function 'panel_st7701_disp_on_off':
../managed_components/espressif__esp_lcd_st7701/esp_lcd_st7701.c:347:53: error: 'esp_lcd_panel_t' has no member named 'user_data'
  347 |     st7701_panel_t *st7701 = (st7701_panel_t *)panel->user_data;
      |                                                     ^~
[31/520] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_rgb.c.objninja: build stopped: subcommand failed.

leeebo, 2023-10-30:
参考：https://github.com/espressif/esp-iot-solution/issues/303

fanyg01, 2023-10-31:
> 参考：#303

感谢您的回复，#303解决了我提问题的前半部分，但是esp_lcd_panel_handle_t类型中没有user_data定义的问题还存在

fanyg01, 2023-10-31:
更换IDF版本可以解决问题。

closed 2023-10-31 by fanyg01