espressif/esp-iot-solution #307: bootloader-support-plus using LittleFS instead of storage partition

kohait00, 2023-10-31:
Just a general RFC..

I want to get rid of spiffs and tbh from the additional ota partitions by trying to implement Filesystem access to the bootloader.
The benefit is:

* only one dedicated OTA partition for the real fw, one bootloader
* update is not only OTA but a simple access to the fs, no matter OTA, via HTTPd UI or whatever..
* less overhead for the separate partitions everywhere
* revert of update up until reset is possible (deleting file)
* uniformity for update, other partitions can be updated in bootloader or application, too

* BL size  < 0xC000 will be a challenge, but probably worth a shot

So just let me hear your thought

WangYuxin-esp, 2023-11-03:
This does bring some benefits, making the entire system look more 'Linux'. However, to my knowledge, free flash access operations cannot be implemented in bootloader, making it more difficult to implement a file system. This requires IDF to provide the prerequisites for implementing this solution.

kohait00, 2023-11-03:
doesnt the normal bootloader component already have flash access when performing the factory erase (you can specify the partitions to erase in sdkconfig).
The biggest challenge I currently see is the size. How much does littleFs  weigh in typically?