espressif/esp-iot-solution #309: Failed to resolve component 'esp_adc_cal'.

stellna, 2023-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32 WoverKit 3.3

### Steps to reproduce.

https://github.com/espressif/esp-iot-solution/issues/229

I'm Thinking that I have a same problem like the person above.

### Build Logs.

```plain
-- Found Git: C:/Users/whgPd/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- Component directory C:/Users/whgPd/esp/esp-idf/components/esp32-csi-tool does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/whgPd/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/whgPd/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/whgPd/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 3 dependencies:
[1/3] espressif/esp-dsp (1.4.8)
[2/3] espressif/mdns (1.2.2)
[3/3] idf (5.1.1)
CMake Error at C:/Users/whgPd/esp/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  C:/Users/whgPd/esp/esp-idf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  C:/Users/whgPd/esp/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  C:/Users/whgPd/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/whgPd/OneDrive/Desktop/esp-test/camera_DL_TEST/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\whgPd\OneDrive\Desktop\esp-test\camera_DL_TEST\build -S c:\Users\whgPd\OneDrive\Desktop\esp-test\camera_DL_TEST" terminated with exit code: 1.
```

### More Information.

I have looked at the website and I understand that because my ESP-IDF version is 5.x that I should use ESP_ADC. And I have able to check that I have download ESP-IDF properly and it is stored in my Esp-Idf folder. Here is the way to get in that folder.
C:\Users\whgPd\esp\esp-idf\components\esp_adc

The thing is that I don't know what kinds of action should I take in order to remove this error. Can anyone please tell me precisely how to solve this problem?

Thank you.

leeebo, 2023-11-09:
From IDF5.0, the component `esp_adc_cal` has been renamed with `esp_adc`, if you want to make your project work both on IDF4.* and IDF5.*, you can prepare your Cmakelists like this https://github.com/espressif/esp-iot-solution/blob/master/examples/common_components/boards/CMakeLists.txt#L9

stellna, 2023-11-09:
Sorry to bother you.
I have fix my Cmakelist as following code in the main folder.

if("${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}" VERSION_GREATER_EQUAL "5.0")
    list(APPEND PRIVREQ esp_adc)
else()
    list(APPEND PRIVREQ esp_adc_cal)
endif()

idf_component_register(SRCS "main.c"
                       INCLUDE_DIRS ".")

And it still showing the same error. I don't know what is wrong in my code. 

[image]

leeebo, 2023-11-09:
@stellna 

https://github.com/espressif/esp-iot-solution/blob/master/examples/common_components/boards/CMakeLists.txt#L35

stellna, 2023-11-09:
After making 2 files of ESP_ADC and ESP_ADC_CAL it has been solved. thank you for your support!

closed 2023-11-09 by stellna

robertojguerra, 2024-07-15:
> After making 2 files of ESP_ADC and ESP_ADC_CAL it has been solved. thank you for your support!

Can you be more specific about what you did to solve this roadblock?
Thank you!

dobodu, 2024-08-29:
> After making 2 files of ESP_ADC and ESP_ADC_CAL it has been solved. thank you for your support!

Hi stellna,
Could you please give us a bit more information on how you got rid of this ESPèADC issue. 
Thancks