espressif/esp-iot-solution #311: SH8601屏幕驱动文件里有些结构没有定义

y1x2h, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
[image]

y1x2h, 2023-11-15:
请问有大佬不修改组件注册表里下载的SH8601组件编译成功的的吗？

Lzw655, 2023-11-15:
@y1x2h 请检查SH8601组件的版本。建议使用 [v1.0.0 版本](https://components.espressif.com/components/espressif/esp_lcd_sh8601)，该版本需要 ESP-IDF master 分支（v5.2）。

Lzw655, 2023-11-15:
> 请问有大佬不修改组件注册表里下载的SH8601组件编译成功的的吗？

该组件依赖了近期更新的新特性，因此指定版本为 ESP-IDF >5.0.4,!=5.1.1，即目前只能在 master 分支使用，或者未来释放的 v5.0.5/v5.1.2 版本。

y1x2h, 2023-11-22:
谢谢，使用了v5.2版本的IDF编译没问题了

closed 2023-11-22 by y1x2h