espressif/esp-iot-solution #312: spd2010 idf component problem

Windsssss, 2023-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-277-gc8243465e4

### Espressif SoC revision.

ESP32-S3 (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

lvgl demo work well

### What is the actual behavior?

the screen lit up for a few seconds and then went out

### Steps to reproduce.

1. download esp idf master (v5.3-dev-277-gc8243465e4)
2. create-project-from-example "espressif/esp_lcd_spd2010^1.0.0:qspi_with_ram"
3. set target esp32s3, set qio spi flash 80mhz ,set octal psram 80mhz, set lcd spd2010
4. build and download program

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x3010
entry 0x403c993c
I (27) boot: ESP-IDF v5.3-dev-277-gc8243465e4 2nd stage bootloader
I (27) boot: compile time Nov 17 2023 13:23:49
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.1
I (35) qio_mode: Enabling default flash chip QIO
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00300000
I (93) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=77968h (489832) map
I (180) esp_image: segment 1: paddr=00087990 vaddr=3fc95100 size=03468h ( 13416) load
I (183) esp_image: segment 2: paddr=0008ae00 vaddr=40374000 size=05218h ( 21016) load
I (190) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=4f498h (324760) map
I (244) esp_image: segment 4: paddr=000df4c0 vaddr=40379218 size=0be30h ( 48688) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (272) cpu_start: Multicore app
I (272) octal_psram: vendor id    : 0x0d (AP)
I (272) octal_psram: dev id       : 0x02 (generation 3)
I (275) octal_psram: density      : 0x03 (64 Mbit)
I (281) octal_psram: good-die     : 0x01 (Pass)
I (286) octal_psram: Latency      : 0x01 (Fixed)
I (291) octal_psram: VCC          : 0x01 (3V)
I (296) octal_psram: SRF          : 0x01 (Fast Refresh)
I (302) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (308) octal_psram: BurstLen     : 0x01 (32 Byte)
I (314) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (320) octal_psram: DriveStrength: 0x00 (1/1)
I (326) MSPI Timing: PSRAM timing tuning index: 5
I (330) esp_psram: Found 8MB PSRAM device
I (335) esp_psram: Speed: 80MHz
I (624) esp_psram: SPI SRAM memory test OK
I (633) cpu_start: Pro cpu start user code
I (633) cpu_start: cpu freq: 240000000 Hz
I (634) cpu_start: Application information:
I (636) cpu_start: Project name:     example_qspi_with_ram
I (643) cpu_start: App version:      v5.3-dev-277-gc8243465e4
I (649) cpu_start: Compile time:     Nov 17 2023 13:23:34
I (655) cpu_start: ELF file SHA256:  b73841d09...
I (661) cpu_start: ESP-IDF:          v5.3-dev-277-gc8243465e4
I (667) cpu_start: Min chip rev:     v0.0
I (672) cpu_start: Max chip rev:     v0.99
I (677) cpu_start: Chip rev:         v0.1
I (681) heap_init: Initializing. RAM available for dynamic allocation:
I (688) heap_init: At 3FC995F0 len 00050120 (320 KiB): RAM
I (695) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (701) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (707) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (713) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (721) spi_flash: detected chip: generic
I (725) spi_flash: flash io: qio
W (729) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (742) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (753) sleep: Configure to isolate all GPIO pins in sleep state
I (760) sleep: Enable automatic switching of GPIO sleep configuration
I (767) main_task: Started on CPU0
I (771) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (780) main_task: Calling app_main()
I (784) example: Turn off LCD backlight
I (789) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (798) example: Initialize SPI bus
I (802) example: Install panel IO
I (806) example: Install SPD2010 panel driver
I (811) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (820) spd2010: LCD panel create success, version: 1.0.0
I (1088) example: Initialize I2C bus
I (1088) example: Initialize touch controller SPD2010
I (1088) gpio: GPIO[47]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
I (1096) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1130) SPD2010: Touch panel create success, version: 0.0.1
I (1130) example: Turn on LCD backlight
I (1130) example: Initialize LVGL library
I (1135) example: Register display driver to LVGL
I (1142) example: Install LVGL tick timer
I (1145) example: Starting LVGL task
I (1189) example: Display LVGL demos
I (1388) main_task: Returned from app_main()
```

Windsssss, 2023-11-17:
Uploading a7a56e03b20b8bf3bc318d32b8689c38.mp4…

Lzw655, 2023-11-17:
Hi @Windsssss, according to your logs, the program seems to work well. So I tried to run this example with the same LCD, and it also worked well for a long time. (I can't upload the video because of the poor network...)

So I think there might be some problems with your hardware. This LCD module has a so weak FPC that I already broke one. You also can try to connect its "RS" pin to GND or VCC although the datasheet said it's NC.

Windsssss, 2023-11-17:
> Hi @Windsssss, according to your logs, the program seems to work well. So I tried to run this example with the same LCD, and it also worked well for a long time. (I can't upload the video because of [...]

hello, @Lzw655 , i connect rs pin to GND or VCC and still no effect, i attach my project below.
[qspi_with_ram.zip](https://github.com/espressif/esp-iot-solution/files/13388759/qspi_with_ram.zip)

Lzw655, 2023-11-17:
@Windsssss I checked your project and directly flashed it to my board. Looks like everything is fine...

Windsssss, 2023-11-18:
@Lzw655 Hi, I test my board, it's a cable problem, poor contact, thank you！

Lzw655, 2023-11-21:
@Windsssss You're welcome! If there are no more problems, please close this issue. Thank you!

closed 2023-11-21 by Windsssss