espressif/esp-iot-solution #313: ST77903 support still under developing?

qiqiji, 2023-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我可以成功的使用 @Lzw655 在某个帖子里发布的这个压缩包编译成功
https://github.com/espressif/esp-dev-kits/files/12249442/ST77903_QSPI_1_6_400_400.zip
但由于它需要使用 master 分支并退回到指定的版本，所以目前只能用于测试用途。因为之前使用 st7789屏幕在 arduino 开发的程序，现在屏幕需要更换成 st77903 的屏，整体移植到esp-idf工作量非常大。似乎正式版的 idf 是允许使用 arduino 的库？因为这个zip包中驱动所要求的 idf 版本号已经到 5.2，我尝试在 idf 中添加 arduino 支持，提示版本不能大于5.1.99？有没有任何可能这个驱动移植到 arduino 会相对简单些？

另外，esp-iot-solution 下的版本似乎没有上传到 Espressif's component service 就像ST7701等其他屏幕一样，不知道是否仍在开发中？我尝试编译也没有成功，不知应该采用什么版本的 idf？

抱歉我使用中文，英语水平有限，实在无法有效的描述以上问题。。。

Lzw655, 2023-11-27:
@qiqiji 你好，使用中文描述完全没有问题，无需担心！

由于 ST77903 中的 QSPI 驱动依赖 IDF 还未合入的新特性（SPI 部分，正在支持中），如你所知，需要基于特定的 commit 版本并打上补丁，因此现有的 arduino-esp32 版本（即使是 3.0.0-alpha2）是无法支持的。

据我所知，可以对 arduino-esp32 进行[重新编译](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html)，这种方法目前只适用于 IDF release/v5.1，但是也可以尝试一下修改 `build.sh` 来使用 v5.2 的 master。除此之外就是你提到的将 arduino 作为 IDF 的一个组件，我不清楚这种方法的限制，但是据你的描述也存在版本问题。

总结来说，目前 ST77903 的驱动确实难以与 arduino 结合使用，IDF 的正式支持应该会在 v5.3 的版本中。

qiqiji, 2023-12-07:
非常感谢您的时间，由于这个屏幕的特殊性（没有GRAM），我们暂时放弃使用，准备改用 ST77916

closed 2023-12-07 by qiqiji