espressif/esp-iot-solution #317: usb_camera_mic_spk 上電啟動時usb Camera連接錯誤，USB拔除重新連接後可以正常運行

8888alex, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2-dirty 2nd stage bootloader

### Espressif SoC revision.

ESP-ROM:esp32s2-rc4-20191025

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s2-devkitm-1-v1

### Power Supply used.

USB

### What is the expected behavior?

下面為硬體重新連接後的日誌

W (458349) USB_STREAM: line 156 HCD_PORT_EVENT_DISCONNECTION
I (458349) USB_STREAM: Recover Stream Task
I (458399) USB_STREAM: Action: ACTION_PIPE_DFLT_DISABLE
I (458399) USB_STREAM: Action: ACTION_PORT_RECOVER
I (458399) USB_STREAM: Action: ACTION_DEVICE_DISCONN
I (458405) USB_STREAM: Waiting USB Connection
I (460156) USB_STREAM: line 152 HCD_PORT_EVENT_CONNECTION
I (460156) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (460206) USB_STREAM: Resetting Port
I (460266) USB_STREAM: Setting Port FIFO, 1
I (460266) USB_STREAM: USB Speed: full-speed
I (460266) USB_STREAM: ENUM Stage START, Succeed
I (460271) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
I (460277) USB_STREAM: Default pipe endpoint MPS update to 64
I (460283) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (460290) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (460306) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (460307) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x46d
idProduct 0x991
bNumConfigurations 1
I (460324) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (460332) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
I (460339) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Succeed
I (460346) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
[... 572 lines cut ...]
W (4089) USB_STREAM: Pipe(default): ERROR_XFER
I (4094) USB_STREAM: Action: ACTION_PIPE_DFLT_RECOVER
I (4100) USB_STREAM: Action: ACTION_PIPE_DFLT_CLEAR
E (4106) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2876 (_uvc_uac_device_enum):urb cleared: enum abort
E (4118) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Failed
W (4124) USB_STREAM: USB enum failed, retrying in 200 ms...
W (4352) USB_STREAM: USB enum failed, retry now
I (4352) USB_STREAM: ENUM Stage START, Succeed
I (4353) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
W (4357) USB_STREAM: Pipe(default): ERROR_XFER
I (4362) USB_STREAM: Action: ACTION_PIPE_DFLT_RECOVER
I (4368) USB_STREAM: Action: ACTION_PIPE_DFLT_CLEAR
E (4374) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2876 (_uvc_uac_device_enum):urb cleared: enum abort
E (4386) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Failed
W (4392) USB_STREAM: USB enum failed, retrying in 200 ms...
W (4619) USB_STREAM: USB enum failed, retry now
I (4619) USB_STREAM: ENUM Stage START, Succeed
I (4620) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
W (4624) USB_STREAM: Pipe(default): ERROR_XFER
I (4629) USB_STREAM: Action: ACTION_PIPE_DFLT_RECOVER
I (4635) USB_STREAM: Action: ACTION_PIPE_DFLT_CLEAR
E (4641) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2876 (_uvc_uac_device_enum):urb cleared: enum abort
E (4653) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Failed
E (4659) USB_STREAM: USB enum failed, no more retry
```

lijunru-hub, 2023-12-18:
你的这颗摄像头和 `AEGHB-489` 中是同一颗摄像头吗？你的摄像头是自供电设备吗？

8888alex, 2023-12-18:
> 你的这颗摄像头和 `AEGHB-489` 中是同一颗摄像头吗？你的摄像头是自供电设备吗？

Logitech PC Laptop USB Webcam QuickCam V-UBU48 2MP Carl Zeiss 2.0/3.7 Autofocus
[image]

leeebo, 2023-12-18:
```
I (461116) USB_STREAM: Set frequence endpoint 0x86: (16000) Hz
E (462116) USB_STREAM: Control Transfer Timeout
I (462117) USB_STREAM: Action: ACTION_PIPE_DFLT_RECOVER
I (462117) USB_STREAM: Action: ACTION_PIPE_DFLT_CLEAR
```

Seems the Mic interface cannot work, but the UVC interface should works. please retry with UVC only (set `ENABLE_UAC_MIC_SPK_FUNCTION` to `0` in [usb_camera_mic_spk/main/main.c#L21](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/host/usb_camera_mic_spk/main/main.c#L21)).

8888alex, 2023-12-19:
在 ENABLE_UAC_MIC_SPK_FUNCTION to 0 之後。

開始monitor 連接失敗
直接reset 依舊失敗
斷開相機重新連接 成功

### LOG
```
PS C:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk> set IDF_PATH=c:\Users\alex\esp\esp-idf-v5.0.2
PS C:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.11_env\Scripts\python.exe c:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk\build\usb_camera_mic_spk.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (21) boot: compile time 19:27:22
I (21) boot: chip revision: v1.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=2cb88h (183176) map
I (127) esp_image: segment 1: paddr=0003cbb0 vaddr=3ffcacc0 size=032fch ( 13052) load
I (130) esp_image: segment 2: paddr=0003feb4 vaddr=40024000 size=00164h (   356) load
I (132) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=8ae94h (568980) map
I (254) esp_image: segment 4: paddr=000caebc vaddr=40024164 size=16b5ch ( 93020) load
I (289) boot: Loaded app from partition at offset 0x10000
I (289) boot: Disabling RNG early entropy source...
I (301) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (301) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (304) esp_psram: Found 2MB PSRAM device
I (309) esp_psram: Speed: 40MHz
V (313) mmu: after coalescing, 4 regions are left
V (317) mmu: found laddr is 0x1800000
V (321) esp_psram: 8bit-aligned-region: actual_mapped_len is 0x200000 bytes
V (328) esp_psram: 8bit-aligned-range: 0x200000 B, starting from: 0x3f800000
I (335) cpu_start: Pro cpu up.
I (749) esp_psram: SPI SRAM memory test OK
D (754) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 4 bit
D (755) efuse: In EFUSE_BLK2__DATA4_REG is used 4 bits starting with 0 bit
D (759) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 16 bit
D (774) clk: RTC_SLOW_CLK calibration value: 6483354
I (783) cpu_start: Pro cpu start user code
I (783) cpu_start: cpu freq: 240000000 Hz
I (783) cpu_start: Application information:
I (788) cpu_start: Project name:     usb_camera_mic_spk
I (794) cpu_start: App version:      d875ea86-dirty
I (799) cpu_start: Compile time:     Dec 15 2023 20:00:19
I (805) cpu_start: ELF file SHA256:  07572d9cb4cf0c04...
I (811) cpu_start: ESP-IDF:          v5.0.2-dirty
I (817) cpu_start: Min chip rev:     v0.0
I (821) cpu_start: Max chip rev:     v1.99 
I (826) cpu_start: Chip rev:         v1.0
V (831) memory_layout: reserved range is 0x3f02cb70 - 0x3f02cb98
D (837) memory_layout: Checking 5 reserved memory ranges:
[... 534 lines cut ...]
I (13397) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 2, width = 160, height = 120, length = 5179, ptr = 0
I (13465) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 3, width = 160, height = 120, length = 5172, ptr = 0
I (13533) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 4, width = 160, height = 120, length = 5174, ptr = 0
I (13601) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 5, width = 160, height = 120, length = 5132, ptr = 0
I (13669) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6, width = 160, height = 120, length = 5166, ptr = 0
I (13737) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 7, width = 160, height = 120, length = 5188, ptr = 0
I (13805) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 8, width = 160, height = 120, length = 5255, ptr = 0
I (13873) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 9, width = 160, height = 120, length = 5304, ptr = 0
I (13941) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 10, width = 160, height = 120, length = 5389, ptr = 0
I (14008) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 11, width = 160, height = 120, length = 5420, ptr = 0
I (14073) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 12, width = 160, height = 120, length = 5452, ptr = 0
W (14113) USB_STREAM: line 156 HCD_PORT_EVENT_DISCONNECTION
I (14114) USB_STREAM: Recover Stream Task
I (14114) USB_STREAM: usb stream task wait reset
I (14119) USB_STREAM: usb stream task reset, reason: device recover
I (14126) USB_STREAM: Resetting UVC pipe
I (14130) USB_STREAM: sample processing stop
I (14135) USB_STREAM: Sample processing task deleted
I (14191) uvc_mic_spk_demo: Device disconnected
I (14192) USB_STREAM: Action: ACTION_PIPE_DFLT_DISABLE
I (14192) USB_STREAM: Action: ACTION_PORT_RECOVER
I (14196) USB_STREAM: Action: ACTION_DEVICE_DISCONN
I (14202) USB_STREAM: Waiting USB Connection

```

leeebo, 2023-12-20:
> 開始monitor 連接失敗
直接reset 依舊失敗
斷開相機重新連接 成功

@8888alex 請再重試多次，每次都需要斷開重連才能成功嗎？

8888alex, 2023-12-20:
反覆測試的結果應該是 VS CODE 的monitor功能有問題。
開始monitor後會先是連接失敗，但重新上電便可恢復正常。
同時或分別上電，不同先後順序也都可以正常連接。
謝謝您的時間。

closed 2023-12-20 by 8888alex