espressif/esp-iot-solution #318: usb_camera_mic_spk 運行無法找OV2659摄像头模块

8888alex, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2-dirty 2nd stage bootloader

### Espressif SoC revision.

ESP-ROM:esp32s2-rc4-20191025

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s2-devkitm-1-v1

### Power Supply used.

USB

### What is the expected behavior?

USB_STREAM: USB Device Connected

### What is the actual behavior?

E (2990) USB_STREAM: MJPEG format NOT found
W (2995) USB_STREAM: MJPEG width*heigh: 65535*65535, NOT found
W (3001) USB_STREAM: Audio control interface NOT found
E (3007) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2979 (_uvc_uac_device_enum):Descriptor Parse Result: No matching configurationsE (3021) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Failed

### Steps to reproduce.

連接usb Camera [OV2659摄像头模块](https://item.taobao.com/item.htm?id=668222676610&_u=g2eeairgbf39)
上電運行

### Debug Logs.

```plain
PS C:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk> set IDF_PATH=c:\Users\alex\esp\esp-idf-v5.0.2
PS C:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.11_env\Scripts\python.exe c:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\esp\esp-iot-solution\examples\usb\host\usb_camera_mic_spk\build\usb_camera_mic_spk.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1     
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc 
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (21) boot: compile time 19:27:22
[... 401 lines cut ...]
I (3261) USB_STREAM: Default pipe endpoint MPS update to 64
I (3266) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (3274) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (3289) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (3290) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x58f
idProduct 0x5608
bNumConfigurations 1
I (3308) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (3315) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
W (3321) USB_STREAM: line 156 HCD_PORT_EVENT_DISCONNECTION
E (3327) USB_STREAM: ./managed_components/espressif__usb_stream/usb_stream.c:2855 (_uvc_uac_device_enum):Enum abort
E (3338) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Failed
I (3344) USB_STREAM: Recover Stream Task
I (3350) USB_STREAM: Action: ACTION_PIPE_DFLT_DISABLE
I (3355) USB_STREAM: Action: ACTION_PORT_RECOVER
I (3360) USB_STREAM: Action: ACTION_DEVICE_DISCONN
I (3365) USB_STREAM: Waiting USB Connection
```

leeebo, 2023-12-18:
@8888alex From the descriptor, this Camera does not support MJPEG format in Full-speed mode.

Please keep in mind, for most Camera device, in different speed modes, there may be different descriptors, which is why the computer (high-speed) works, but the ESP32-S3 (full-speed) not work.

8888alex, 2023-12-19:
好的。
請問我該如何調整代碼()以使相機可以工作?

leeebo, 2023-12-20:
> 好的。 請問我該如何調整代碼()以使相機可以工作?

因爲相機本身不支持在 Full speed 下工作，只能調整相機端的固件。ESP32-S3 主機端沒有辦法讓它正常工作

8888alex, 2023-12-20:
請原諒我對高速與全速這些規格規範的不理解。
我的理解是
full-speed = 全部速度(我想應該是比較高級的)
high-speed = 僅有高速模式
low-speed .... 不知道有沒有這規格
既然ESP32可以full-speed
所以在設備僅能以高速運作或低速運作的情況下
能否使ESP32-S2(我是用這個)配合降速
降速會掉幀率是必然的，但我的需求也僅是要擷取靜態影像。
可能這兩個設備(Logitech PC Laptop USB Webcam QuickCam V-UBU48 2MP Carl Zeiss 2.0/3.7 Autofocus 和 OV2659摄像头模块)
在電腦上都是免驅動隨插即用，所以讓我把事情想簡單了。:D
不知道您這是否有可行的方案?

lijunru-hub, 2023-12-20:
high-speed 比 full-speed 速度更快，你這個攝像頭本身就是不支持在 full-speed 下輸出 mjpeg 格式的圖像的。所以只能更換攝像頭或者更改攝像頭固件了。

8888alex, 2023-12-21:
所以是esp32-s2的硬體效能無法提供high-speed？

leeebo, 2023-12-21:
@8888alex esp32-s2/s3 硬件不支持 USB High Speed。如果做低成本解决方案，可以要求摄像头供应商修改固件，或提供 Fullspeed 可用的摄像头

closed 2024-02-04 by leeebo