espressif/esp-iot-solution #319: iot_usbh_modem 4G优化建议

XuGang1986, 2023-12-20:
在实际产品使用过程中可能需要将更多的  ``AT操作接口 `` 释放出来，目前   ``usbh_modem_board.h `` 只有：
```cpp
esp_err_t modem_board_get_signal_quality(int *rssi, int *ber);
esp_err_t modem_board_get_sim_cart_state(int *if_ready);
esp_err_t modem_board_get_operator_state(char *buf, size_t buf_size);
esp_err_t modem_board_set_apn(const char *new_apn, bool force_enable);
```
举几个例子：
1. 很多4G类物联网产品可能更倾向于使用模组的  ``IMEI号码 `` 作为唯一标识；
2. 很多4G类物联网产品批量后，会面临统一管理 ``SIM卡资费`` 事宜，因此需要每台设备自动获取 ``ICCID号码``，``IMSI号码``是不合适的；
等等......

可以在``usbh_modem_board.h ``中多释放几个接口，也可以将
```cpp
esp_modem_dce_common_commands.h
usbh_modem_board.h
```
直接释放出来或者在 ``usbh_modem_board `` 中再封装一次方便用户自定义

XuGang1986, 2023-12-20:
esp_modem_dce_common_commands.h

XuGang1986, 2023-12-20:
为了更高的自由度可以将以下：
esp_modem_dce.h
esp_modem_dce_common_commands.h
释放出来，或者把 AT 操作相关接口封装下通过usbh_modem_board.h释放出来

hyzgit, 2024-04-03:
但是是拨号上网后，就无法使用AT命令了

leeebo, 2024-04-03:
@hyzgit 如果想在联网的同时使用 AT 指令：
1. 选择支持 PPP + AT 双接口的模组，这样一个接口走 PPP 网络，另一个接口（Secondary AT Port）可以同时走 AT, 互不干扰。https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module
2. 将上网状态临时切换回 AT 模式，AT 响应以后，重新回到网络状态 （参考 API https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/include/usbh_modem_board.h#L135）